echavet/MitsubishiCN105ESPHome #196: Fan mode is not valid

flybrys, 2024-12-07:
Recently updated the esphome code and now this error is showing in the HA logs:

Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/commands.py:287
integration: Home Assistant WebSocket API ([documentation](https://www.home-assistant.io/integrations/websocket_api), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+websocket_api%22))
First occurred: 6:31:35 AM (2 occurrences)
Last logged: 6:32:09 AM

[140610517122016] Fan mode is not valid. Valid fan modes are: auto, low, medium, high, middle, quiet

Nothing specifically says its from this integration but its the only device I have that has those modes.

Functionally, I can't see anything wrong (yet)

jamer007, 2024-12-28:
I had those errors at some point too when trying to change the fan mode (STAGE) into HA, my issue was I was using my smartphone and my keyboard auto-capitalized my text, I was ending with Middle instead of middle, High instead of high. The mode text is case-sensitive in HA. I hope this helps you.

phidauex, 2025-01-03:
Hi @flybrys - Did the suggestion to confirm capitalization help? Since we haven't heard back on this one I'm going to close for now. If you've done additional troubleshooting and continue to get the error, feel free to add more information and reopen.

flybrys, 2025-01-04:
Hi @phidauex and @jamer007 sorry I've been away for Christmas and couldn't check.

Where exactly am I looking for capitalisation? I haven't changed anything that required me to type in the fan modes manually so I'm not quite sure. The error just started appearing one day.

phidauex, 2025-01-04:
Can you post your `climate:` section of your configuration YAML? Do you have `fan_mode:` specified anywhere in there?

flybrys, 2025-01-04:
I don't have` fan_mode:` anywhere in the file. Here is the `climate:` section:

```
climate:
  - platform: cn105
    id: hp
    name: "${friendly_name}"
    icon: mdi:heat-pump
    update_interval: 2s        # update interval can be ajusted after a first run and logs monitoring* 
    debounce_delay: 500ms
    visual:
      min_temperature: 15
      max_temperature: 31
      temperature_step:
        target_temperature: 1
        current_temperature: 0.1
    remote_temperature_timeout: 30min
    outside_air_temperature_sensor:
      name: Outside Air Temp
    compressor_frequency_sensor:
      name: ${friendly_name} Compressor Frequency
    vertical_vane_select:
      name: ${friendly_name} Vertical Vane
    horizontal_vane_select:
      name: ${friendly_name} Horizontal Vane
    stage_sensor:
      name: Stage
      entity_category: diagnostic
    sub_mode_sensor:
      name: Sub Mode
      entity_category: diagnostic
    auto_sub_mode_sensor:
      name: Auto Sub Mode
      entity_category: diagnostic
    input_power_sensor:
      name: Input Power
    kwh_sensor:
      name: Energy Usage
    runtime_hours_sensor:
      name: Runtime Hours
      entity_category: diagnostic
      disabled_by_default: true 
```

flybrys, 2025-01-05:
Could we please reopen this?
Recompiled from scratch and it still pops up. I can't be too sure if it is coincidental but it appeared a bit later this time and the only thing I did was change the fan mode to auto

jamer007, 2025-01-23:
Hi again,
Here a good example in the HA ui:
[image]

And now a bad one:
[image]

flybrys, 2025-01-23:
> Hi again, Here a good example in the HA ui

Thanks for the reply, but I'm not using any automations (yet). The error shows up with just the device installed

echavet, 2025-03-11:
Hi,
I know it sucks but I need more info to be able to diagnose the issue.
You can set the logging level to DEBUG for cn105.

```yaml
logger:
  baud_rate: 0
  level: DEBUG
  logs:
    ...
    CN105: DEBUG
    ...
```

try to follow the logs from the "ESPHome Builder" addon and to provide me more info on the pb you're facing.
thanks

phidauex, 2025-06-04:
Hi @flybrys - Any update on this issue? Are you still experiencing the problem?

flybrys, 2025-06-23:
Still getting the issue.
I can't see anything in the debug log relevant but here is a snippet.
The error only shows up on Home Assistant start up. If I clear the HA logs, and then restart the ESP, it doesn't immediately show up.

```
INFO ESPHome 2025.6.1
INFO Reading configuration /config/esphome/air-conditioner.yaml...
INFO Detected timezone 'Australia/Sydney'
INFO Starting log output from 172.16.11.40 using esphome API
INFO Successfully resolved air-conditioner @ 172.16.11.40 in 0.000s
INFO Successfully connected to air-conditioner @ 172.16.11.40 in 0.003s
INFO Successful handshake with air-conditioner @ 172.16.11.40 in 0.112s
[08:48:32][I][app:137]: ESPHome version 2025.6.1 compiled on Jun 24 2025, 08:41:43
[08:48:32][C][wifi:613]: WiFi:
[08:48:32][C][wifi:434]:   Local MAC: 94:3C:C6:33:E4:C0
[08:48:32][C][wifi:439]:   SSID: 'IoT'[redacted]
[08:48:32][C][wifi:442]:   IP Address: 172.16.11.40
[08:48:32][C][wifi:446]:   BSSID: 66:22:32:2F:84:BA[redacted]
[08:48:32][C][wifi:446]:   Hostname: 'air-conditioner'
[08:48:32][C][wifi:446]:   Signal strength: -41 dB ▂▄▆█
[08:48:32][C][wifi:455]:   Channel: 11
[08:48:32][C][wifi:455]:   Subnet: 255.255.255.0
[08:48:32][C][wifi:455]:   Gateway: 172.16.11.254
[08:48:32][C][wifi:455]:   DNS1: 172.16.11.252
[08:48:32][C][wifi:455]:   DNS2: 0.0.0.0
[08:48:32][C][logger:211]: Logger:
[08:48:32][C][logger:211]:   Max Level: DEBUG
[08:48:32][C][logger:211]:   Initial Level: DEBUG
[08:48:32][C][logger:217]:   Log Baud Rate: 0
[08:48:32][C][logger:217]:   Hardware UART: UART0
[08:48:32][C][logger:224]:   Task Log Buffer Size: 768
[08:48:32][C][logger:229]:   Level for 'CN105': DEBUG
[08:48:32][C][uart.idf:159]: UART Bus 0:
[08:48:32][C][uart.idf:160]:   TX Pin: GPIO1
[08:48:32][C][uart.idf:161]:   RX Pin: GPIO3
[08:48:32][C][uart.idf:163]:   RX Buffer Size: 256
[08:48:32][C][uart.idf:165]:   Baud Rate: 2400 baud
[08:48:32][C][uart.idf:165]:   Data Bits: 8
[08:48:32][C][uart.idf:165]:   Parity: EVEN
[08:48:32][C][uart.idf:165]:   Stop bits: 1
[08:48:32][C][uptime.sensor:033]: Uptime Sensor 'Air Conditioner Uptime'
[08:48:32][C][uptime.sensor:033]:   State Class: 'total_increasing'
[08:48:32][C][uptime.sensor:033]:   Unit of Measurement: 's'
[08:48:32][C][uptime.sensor:033]:   Accuracy Decimals: 0
[08:48:32][C][uptime.sensor:033]:   Device Class: 'duration'
[08:48:32][C][uptime.sensor:033]:   Icon: 'mdi:timer-outline'
[08:48:32][C][homeassistant.time:010]: Home Assistant Time:
[08:48:32][C][homeassistant.time:010]:   Timezone: 'AEST-10AEDT,M10.1.0,M4.1.0/3'
[08:48:32][C][restart.button:017]: Restart Button 'Restart Air Conditioner'
[08:48:32][C][restart.button:017]:   Icon: 'mdi:restart'
[08:48:32][C][esp32_ble:518]: ESP32 BLE:
[08:48:32][C][esp32_ble:518]:   MAC address: 94:3C:C6:33:E4:C2
[08:48:32][C][esp32_ble:518]:   IO Capability: none
[08:48:32][C][esp32_ble_tracker:763]: BLE Tracker:
[08:48:32][C][esp32_ble_tracker:764]:   Scan Duration: 300 s
[08:48:32][C][esp32_ble_tracker:764]:   Scan Interval: 1100.0 ms
[08:48:32][C][esp32_ble_tracker:764]:   Scan Window: 1100.0 ms
[08:48:32][C][esp32_ble_tracker:764]:   Scan Type: ACTIVE
[08:48:32][C][esp32_ble_tracker:764]:   Continuous Scanning: YES
[08:48:32][C][esp32_ble_tracker:780]:   Scanner State: RUNNING
[08:48:32][C][esp32_ble_tracker:792]:   Connecting: 0, discovered: 0, searching: 0, disconnecting: 0
[08:48:32][C][bluetooth_proxy.connection:017]: BLE Connection:
[08:48:32][C][esp32_ble_client:048]:   Address: 
[08:48:32][C][esp32_ble_client:048]:   Auto-Connect: FALSE
[08:48:32][C][esp32_ble_client:085]:   State: IDLE
[08:48:32][C][bluetooth_proxy.connection:017]: BLE Connection:
[08:48:32][C][esp32_ble_client:048]:   Address: 
[08:48:32][C][esp32_ble_client:048]:   Auto-Connect: FALSE
[08:48:32][C][esp32_ble_client:085]:   State: IDLE
[08:48:32][C][bluetooth_proxy.connection:017]: BLE Connection:
[... 342 lines cut ...]
[08:48:46][D][EVT_SETS:062]: [wanted]-> [power: -, target °C: -1.0, mode: -, fan: -, vane: -, wvane: -, hasChanged ? ->  NO, hasBeenSent ? ->  NO]
[08:48:46][D][SETTINGS:089]: [climate]-> [mode: HEAT, target °C: 21.0, fan: AUTO, swing: BOTH]
[08:48:46][D][CYCLE:359]: 3a: Sending room °C request (0x03)
[08:48:47][D][Header:095]: command: (62) data length: [10]<-- header
[08:48:47][D][READ:172]: FC 62 01 30 10 03 00 00 0D 00 01 AE B2 FE 42 00 01 49 B6 00 00 AC 
[08:48:47][D][chkSum:081]: OK-> AC=AC 
[08:48:47][D][CYCLE:365]: 3b: Receiving room °C response
[08:48:47][D][Decoder:293]: [Room °C: 23.000000]
[08:48:47][D][Decoder:294]: [OAT  °C: nan]
[08:48:47][I][STATUS:130]: [received]-> [room C°: 23.0, operating: NO , compressor freq: 0.0 Hz]
[08:48:47][I][STATUS:130]: [current]-> [room C°: 21.0, operating: NO , compressor freq: 0.0 Hz]
[08:48:47][D][CN105:270]: Setting action to 4 (effective_operating: false, use_stage_fallback: no, current_stage: LOW)
[08:48:47][D][CN105:348]: Climate mode is: 3
[08:48:47][D][CN105:349]: Climate action is: 4
[08:48:47][D][climate:396]: 'Air Conditioner' - Sending state:
[08:48:47][D][climate:399]:   Mode: HEAT
[08:48:47][D][climate:401]:   Action: IDLE
[08:48:47][D][climate:404]:   Fan Mode: AUTO
[08:48:47][D][climate:416]:   Swing Mode: BOTH
[08:48:47][D][climate:419]:   Current Temperature: 23.00°C
[08:48:47][D][Header:095]: command: (62) data length: [10]<-- header
[08:48:47][D][READ:172]: FC 62 01 30 10 06 00 00 00 00 00 04 23 CC 00 00 42 00 00 00 00 22 
[08:48:47][D][chkSum:081]: OK-> 22=22 

```