echavet/MitsubishiCN105ESPHome #201: Heat pump heating beyond target

CopperSubstrate, 2024-12-15:
Our home has many Mitsubishi heat pumps. I've installed the integration on all of them, replacing the prior integration. 
	
Recently as it has become cold I've noticed an issue. Our largest unit, a Mitsubishi MSZ-GL18NA-U1 (the only unit which has vertical and horizontal swing modes), will sometimes continue to heat past the heat target set in the integration, and the integration will report that it's not heating. Logs show the heat climbing past the target but do not show the unit as actively heating.

phidauex, 2024-12-15:
Hmm, that is odd. It could be intended operation, I've noticed they do have some hysteresis and will sometimes overshoot a bit in order to run a longer cycle. It could also be bleedover if you have multiple heads on the same outdoor unit - if other units are still heating, then hot refrigerant is still run through the "off" head, in order to keep oil from pooling. This can cause them to put out some heat even if the unit is trying to be off.

What do the LEDs on the front say when it is doing this?

[image]

CopperSubstrate, 2024-12-16:
> Hmm, that is odd. It could be intended operation, I've noticed they do have some hysteresis and will sometimes overshoot a bit in order to run a longer cycle. It could also be bleedover if you have multiple heads on the same outdoor unit - if other units are still heating, then hot refrigerant is still run through the "off" head, in order to keep oil from pooling. This can cause them to put out some heat even if the unit is trying to be off.
> 
> What do the LEDs on the front say when it is doing this?
> 
> <img alt="image" width="371" src="https://private-user-images.githubusercontent.com/49047/395882175-3a4ac529-eee0-43f1-bfe8-e90caecf3d3e.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQzMTE0MDYsIm5iZiI6MTczNDMxMTEwNiwicGF0aCI6Ii80OTA0Ny8zOTU4ODIxNzUtM2E0YWM1MjktZWVlMC00M2YxLWJmZTgtZTkwY2FlY2YzZDNlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjE2VDAxMDUwNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTVmODAwNWFkZTI3MDJlNWQwMTNlNGY5OWFjZTMyOWFmMDY5YTQwOGRkODU2NjQ5NmUyZGY3NTJiODA4ZjkyYzgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.63btSHjgwF85N2DbpvRZCnrS8ecOufO5uv_vP0-OAPo">

It was the usual light pattern - only the top light on and solid.

CopperSubstrate, 2024-12-22:
I'm seeing this issue again with a different unit. This one definitely has both lights illuminated.

CopperSubstrate, 2025-01-08:
I've now had this issue with two of my Mitsubishi units of different models:

- Temperature is below target. Unit is turned on by the integration.
- Temperature target is reached. The unit should stop heating and is marked as not heating in the logs, but it still heats.
- We discover later that a room set to something like 66 degrees F is actually at 72 degrees. We turn the unit off manually.

While this intergration has been superior in every other way to the gysmo38 integration, I did not have this issue with that integration despite using the same units and same ESP8266s. It definitely wasn't an issue when using the remotes.

CopperSubstrate, 2025-01-08:
> Hmm, that is odd. It could be intended operation, I've noticed they do have some hysteresis and will sometimes overshoot a bit in order to run a longer cycle. It could also be bleedover if you have [...]

I do not  think this is the intended operation as the variance between the set temp and actual temp has been as high as seven degrees.

I think this qualifies as a bug rather than a question.

CopperSubstrate, 2025-01-10:
Sure. Here's one of the graphs of such an incident. As you can see, the Nursery is set to 65 degrees. The room was unoccupied and so we didn't know anything was amiss. It was around 7 PM that we noticed that the heat was on and that the temperature in the room had raised way beyond the target - at that point I shifted the unit from "heat" to "off."
[image]

The YAML for this unit is here: [Nursery YAML.txt](https://github.com/user-attachments/files/18370317/Nursery.YAML.txt)

phidauex, 2025-01-10:
Interesting - the remote temperature is being updated correctly, and the unit is reporting the correct setpoint temperature. Is there another indoor unit connected to this same outdoor unit? If so, can you show what it was doing during this time as well? Any chance of a power meter reading?

CopperSubstrate, 2025-01-11:
I have an Emporia Vue hooked up to the breaker box downstairs, but the measurements I get are only by condenser - so I do have measurements but only for the entire floor - that consists of three pumps - one other was on during the period and the other was off. Picture attached - I'm not sure I can make much sense of the significance here myself, except that it 

[image]

CopperSubstrate, 2025-01-11:
[image]
Here's a much messier diagram showing both units' data with the power use.

phidauex, 2025-01-11:
Are Nursery and Bedroom on the same outdoor unit (Condensor 1)? Something definitely seems off because Bedroom is calling for heat nearly all day, but its reported current temperature is almost dead flat. Then at 12:15 and again at 7:00, the measured temperature in that room spikes, the unit turns off, and it rapidly cools as expected, but then goes flat again.

I'm also noticing that while the Bedroom unit claims to be calling for heat, the outdoor unit is off quite a lot, you'd expect to see the outdoor unit running and modulating as long as the indoor unit is calling for heat.

What remote temperature sensor is Bedroom looking at, and does it show clean output?

echavet, 2025-01-13:
Hey,
Are you un this same situation?

https://hvac-talk.com/vbb/threads/2263095-Mitsubishi-MSZ-FS-Heating-Temperature-Overshoot

CopperSubstrate, 2025-01-13:
> Are Nursery and Bedroom on the same outdoor unit (Condensor 1)? Something definitely seems off because Bedroom is calling for heat nearly all day, but its reported current temperature is almost dead [...]

They're on the same outdoor unit. Interesting point about the bedroom. I assume it was on but the fan mode was on "quiet" and the doors were open so heating was minimal, but so far I don't see a way to see historical fan settings. 

The bedroom sensor is a Centralite 3310-G and is accurate, albeit it's on the far side of the room from the heat pump. 

This did make me think of something as I noticed this same issue today with the Attic into, which shares an outdoor unit with the Guest pump. While the Attic unit was listed as off (on heat, actual temp > target temp), the guest unit *was* on and calling for heat. The Attic heat seemed to stop when I turned the Guest unit off. 

This is just one example so I'll have to keep testing, but you've got me thinking about how this could be related to interactions between pumps that share a unit.

CopperSubstrate, 2025-01-13:
> Hey, Are you un this same situation?
> 
> https://hvac-talk.com/vbb/threads/2263095-Mitsubishi-MSZ-FS-Heating-Temperature-Overshoot

Opposite issue in some ways. Cooling is fine but the unit's own temperature readings are usually wildy higher than ambient temperatures, so you need to crank up the units to 76+ just to get the units to activate and heat to 66. 

The issue impacted different units to different extents. I was never able to fix it directly, so the main reason I jumped to this integration was that it allowed me to use remote temperature sensors (secondary reason: this integration is much more reactive than the old one).

phidauex, 2025-01-16:
To see historical fan settings, you can enable the "Stage" setting, which seems to work on most indoor units. It shows the current actual fan speed of the unit, which can show if the fan is still running even if the unit is otherwise "off".

[image]

In the case of Guest and Attic being on the same outdoor unit, what can happen is that if Guest is calling for heat, the outdoor unit will circulate hot refrigerant. To keep oil from pooling in the system, it will circulate hot coolant to Attic as well. Different outdoor units can throttle this excess coolant down more or less (units with branch boxes and Smart Multi can throttle further down), but they all let some through. If Guest and Attic have very different heat loads, then it can mean that when Guest is crankin', Attic is getting overheated. This happens more in colder conditions because Guest's runtime will be proportionally longer.

One thing that can make this worse, but that maybe could be addressed through automation, is the fan speed. If Attic is set to "heat", then the fan will never turn off, it will stay on Gentle in order to circulate air through the temp sensors so it knows if it should turn back on. This means the evaporator is getting heat, and the fan is running, so the heat is transferring to the room. People usually notice that if they turn the overheating unit to "off", then the louvers close, the fan shuts off, and the overshoot is mitigated. It might be possible to automate that behavior - say, if actual temp is 2 degrees over the setpoint, then turn the unit to "off". If actual temp is less than 2 degrees over setpoint, turn back to "heat".

CopperSubstrate, 2025-01-16:
I'll have to enable the stage setting today and see if that adds more detail and see what hapened next time.

CopperSubstrate, 2025-01-23:
I believe you're on to something here. The bedroom unit's assigned remote temperature sensor was mounted on a relatively cool wall - a poorly insulated area is behind it that has some plumbing. I moved it elsewhere and the bedroom's use now looks much "healthier" - it isn't always on. 

[image]

I've been testing and while I have still experienced this overshoot issue, it seems linked to another pump within the system actively heating - I've been able to sometimes stop the issue  in the attic by turning the guest unit off (its compressor "buddy.")

CopperSubstrate, 2025-02-01:
[image]

An insight here: I enabled the stage mode and it shows the fan mode switching to low and off nonstop while this issue is active.

echavet, 2025-03-11:
hi guys,
Is this issue still considered a BUG, or is it a regulation issue to be attributed to the side effects of multisplit systems?

CopperSubstrate, 2025-03-11:
Honestly I'm not sure - the Mitsubishi support line didn't give me much insight. So for now I think the answer is "probably?" but I'm not sure.

closed 2025-04-05 by CopperSubstrate