echavet/MitsubishiCN105ESPHome #204: Remote current Temperature overwritten by local current temperature

victorgolfecho, 2024-12-21:
I setup a remote temperature sensor which is correctly updates the temperature based on the HA entity id.  However, the Current Temperate is regularly overwritten by the room temperature being provided by unit.  Note: this behavior is happening on only 1 of my 2 heat pumps (they are different models) configured with a remote thermostat.

I enabled DEBUG log and captured when it's happening.  In this example, the remote temp is reading 19.5C and the unit is reading 11C.

> [13:30:27][D][CN105:308]: sending a request for settings packet (0x02)
[13:30:27][I][CYCLE:039]: 1: Cycle start
[13:30:27][D][CYCLE:311]: 2a: Sending settings request (0x02)
[13:30:27][D][CN105:323]: creating Info packet
[13:30:27][D][CN105:084]: writing packet...
[13:30:27][D][WRITE:170]: FC 42 01 30 10 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7B 
[13:30:27][D][Header:091]: command: (62) data length: [10]<-- header
[13:30:27][D][READ:170]: FC 62 01 30 10 02 00 00 01 01 0F 02 05 00 00 00 A0 00 00 00 00 A3 
[13:30:27][D][chkSum:077]: OK-> A3=A3 
[13:30:27][D][CYCLE:315]: 2b: Receiving settings response
[13:30:27][D][Decoder:173]: [0x02 is settings]
[13:30:27][D][Decoder:181]: [Power : ON]
[13:30:27][D][Decoder:182]: [iSee  : 0]
[13:30:27][D][Decoder:183]: [Mode  : HEAT]
[13:30:27][D][Decoder:195]: [Temp °C: 16.000000]
[13:30:27][D][Decoder:198]: [Fan: 1]
[13:30:27][D][Decoder:201]: [Vane: ↓↓]
[13:30:27][W][lookup:204]: wideVane reading caution: value 0 not found, returning value at index 0
[13:30:27][D][Decoder:206]: [wideVane: ←← (adj:0)]
[13:30:27][D][SETTINGS:113]: [current        ]-> [power: ON , target °C: 16.0, mode: HEAT  , fan: 1     , vane: ↓↓, wvane: ←←]
[13:30:27][D][SETTINGS:113]: [received       ]-> [power: ON , target °C: 16.0, mode: HEAT  , fan: 1     , vane: ↓↓, wvane: ←←]
[13:30:27][D][EVT_SETS:072]: [wanted         ]-> [power: -  , target °C: -1.0, mode: -     , fan: -     , vane: -     , wvane: -     , hasChanged ? ->  NO, hasBeenSent ? ->  NO]
[13:30:27][D][SETTINGS:087]: [climate]-> [mode: HEAT, target °C: 16.0, fan: LOW, swing: OFF]
[13:30:27][D][CN105:218]: setting action to -> 4
[13:30:27][D][CN105:267]: Climate mode is: 3
[13:30:27][D][CN105:268]: Climate action is: 4
[13:30:27][D][climate:396]: 'Heat Pump - Living Room' - Sending state:
[13:30:27][D][climate:399]:   Mode: HEAT
[13:30:27][D][climate:401]:   Action: IDLE
[13:30:27][D][climate:404]:   Fan Mode: LOW
[13:30:27][D][climate:416]:   Swing Mode: OFF
[13:30:27][D][climate:419]:   Current Temperature: 19.50°C
[13:30:27][D][climate:425]:   Target Temperature: 16.00°C
[13:30:27][D][CYCLE:318]: 3a: Sending room °C request (0x03)
[13:30:27][D][CN105:323]: creating Info packet
[13:30:27][D][CN105:084]: writing packet...
[13:30:27][D][WRITE:170]: FC 42 01 30 10 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7A 
[13:30:27][W][component:237]: Component esphome.coroutine took a long time for an operation (75 ms).
[13:30:27][W][component:238]: Components should block for at most 30 ms.
[13:30:27][D][Header:091]: command: (62) data length: [10]<-- header
[13:30:27][D][READ:170]: FC 62 01 30 10 03 00 00 01 00 92 96 00 00 00 00 00 00 00 00 00 31 
[13:30:27][D][chkSum:077]: OK-> 31=31 
[13:30:27][D][CYCLE:324]: 3b: Receiving room °C response
[13:30:27][D][Decoder:253]: [Room °C: 11.000000]
[13:30:27][D][Decoder:254]: [OAT  °C: 9.000000]
[13:30:27][I][STATUS:134]: [received       ]-> [room C°: 11.0, operating: NO , compressor freq: 0.0 Hz]
[13:30:27][I][STATUS:134]: [current        ]-> [room C°: 19.5, operating: NO , compressor freq: 0.0 Hz]
[13:30:27][D][CN105:218]: setting action to -> 4
[13:30:27][D][CN105:267]: Climate mode is: 3
[13:30:27][D][CN105:268]: Climate action is: 4
[13:30:27][D][climate:396]: 'Heat Pump - Living Room' - Sending state:
[13:30:27][D][climate:399]:   Mode: HEAT
[13:30:27][D][climate:401]:   Action: IDLE
[13:30:27][D][climate:404]:   Fan Mode: LOW
[13:30:27][D][climate:416]:   Swing Mode: OFF
[13:30:27][D][climate:419]:   Current Temperature: 11.00°C
[13:30:27][D][climate:425]:   Target Temperature: 16.00°C
[13:30:27][D][CYCLE:327]: 4a: Sending status request (0x06)
[13:30:27][D][CN105:323]: creating Info packet
[13:30:27][D][CN105:084]: writing packet...
[13:30:27][D][WRITE:170]: FC 42 01 30 10 06 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 77 
[13:30:27][W][component:237]: Component esphome.coroutine took a long time for an operation (56 ms).
[13:30:27][W][component:238]: Components should block for at most 30 ms.
[13:30:27][D][Header:091]: command: (62) data length: [10]<-- header

You can see this line:
>[13:30:27][I][STATUS:134]: [received       ]-> [room C°: 11.0, operating: NO , compressor freq: 0.0 Hz]
[13:30:27][I][STATUS:134]: [current        ]-> [room C°: 19.5, operating: NO , compressor freq: 0.0 Hz]

is from here: https://github.com/echavet/MitsubishiCN105ESPHome/blob/89e525a51288245a5df82120980044d5ed53eb12/components/cn105/hp_readings.cpp#L432

which sets room temp on line 444.

Here is the relevant parts of the config yaml:

> sensor:
>   - platform: homeassistant
>     name: "Temperature Sensor From Home Assistant"
>     entity_id: ${remote_temp_sensor}    
>     internal: false
>     device_class: temperature
>     state_class: measurement
>     unit_of_measurement: "°C"
>     filters:
>       - lambda: 'return (x - 32) * (5.0/9.0);'
>       - clamp: # Limits values to range accepted by Mitsubishi units
>           min_value: 1
>           max_value: 40
>           ignore_out_of_range: true
>       - throttle: 30s
>     on_value:
>       then:
>         - logger.log:
>             level: WARN
>             format: "Remote temperature received from HA: %.1f C"
>             args: [ 'x' ]
>         - lambda: 'id(hp).set_remote_temperature(x);'

> climate:
>   - platform: cn105
>     name: ${friendly_name}
>     id: hp
>     remote_temperature_timeout: 30min
>     update_interval: 4s
>     debounce_delay : 100ms

Lastly, this is easily reproducible on my side.  It oscillates between the two temp readings every few seconds, happy to provide any more info if it'd be helpful.

CopperSubstrate, 2025-01-08:
I've had this issue as well.

victorgolfecho, 2025-01-08:
I have a change into the geoffdavis version of this project that appears to be working for me: https://github.com/geoffdavis/esphome-mitsubishiheatpump/pull/173

I have not had a chance to make the changes to this version.

phidauex, 2025-01-08:
@victorgolfecho What indoor and outdoor unit models do you have? I wonder how the MHK and Kumo systems deal with the issue if something about the indoor unit overwrites the remote temperature value on its own (perhaps they do what your proposed solution does, and simply republish the remote temperature frequently).

echavet, 2025-01-15:
> I have the same issue, it is updating from the remote sensor, but it's mixing up for some reason.
> 
> I also noticed my unit is not stopping when at target temperature, there seems to be a built-in offset of at least 1 degree.
> 
> 
> 
> @echavet any idea in which direction a solution can be found for this issue?

No. I need to dig!

echavet, 2025-01-15:
Hi @victorgolfecho,
I'm very surprised by the magnitude of the difference between the two temperature probes.
How do you explain this discrepancy?
The bug is indeed visible in the logs, but I'd like to better understand the context in which it occurs.

echavet, 2025-01-15:
hey everyone, 
could you provide your unit ref (which is facing the issue of course)?

echavet, 2025-01-15:
> I have a change into the geoffdavis version of this project that appears to be working for me: [geoffdavis/esphome-mitsubishiheatpump#173](https://github.com/geoffdavis/esphome-mitsubishiheatpump/pull/173)
> 
> I have not had a chance to make the changes to this version.

@victorgolfecho 
it is very confusing for me.
The geoffrey version is very different. Do you mean you've had the same issue with geoff Davis's version, but you've got it fixed with this commit?

jan-dewit, 2025-01-15:
> hey everyone, could you provide your unit ref (which is facing the issue of course)?

I'm using a MSZ-HR35VF.

CopperSubstrate, 2025-01-16:
Two units seem impacted: a MSZ-GL18NA-U1 and a MSZ-GL06NA-U1

youknowjack0, 2025-02-26:
I faced this issue with a SEZ-KD unit - i.e. it "forgets" remote temperature after about a minute. I resolved it simply by adding a 15s "heartbeat" filter to the remote sensor to keep republishing the remote temp, something like:

```
sensor:
- platform: homeassistant
  [...]
  filters:
    - heartbeat: 15s        
  [...]
  on_value:
      - lambda: 'id(hp).set_remote_temperature(x);'
```

victorgolfecho, 2025-02-28:
@youknowjack0 heartbeat fix definitely works.  I validated it a few days ago and it's been perfect for the last 36 hours.  This is wonderful!

tfasz, 2025-03-08:
I ran into this issue as well (my units are ducted so the "room" temp is the un-insulated attic/basement temp). I've applied the same work-around but it's a bit of a concern that it's just masking an issue/race condition.

echavet, 2025-03-11:
I would like to move forward with resolving this issue, but I will need some help.  

Several hypotheses:  
- some air conditioning models might have a specific behavior regarding the update of the external temperature.  
- it might be a peculiarity of my implementation(It would be necessary to test with GeoffDavis' implementation to confirm this).  
- It is possible that the cycle duration (which is logged when the `CYCLE: DEBUG` logger is enabled) exceeds that of `update_interval`. The stress caused by sending packets to the air conditioner might make it skip processing the external temperature packet. You can try temporarily increasing `update_interval` to verify this hypothesis.  

I have added specific logs in the next firmware update (issue #245) to more accurately trace the calls to `set_remote_temperature()`.  

```yaml
logger:
  #hardware_uart: USB_CDC
  baud_rate: 0
  level: DEBUG
  logs:
    ACK : DEBUG
    REMOTE_TEMP : DEBUG
```
with this set, it is expected to read something like that:

```
[10:26:45][D][REMOTE_TEMP:301]: Sending remote temperature...
[10:26:45][D][REMOTE_TEMP:373]: Sending remote temperature packet... -> 20.799999
[10:26:45][I][CYCLE:052]: 6: Cycle ended in 1.0 seconds (with timeout?:  NO)
[10:26:45][D][Update-ACK:170]: FC 61 01 30 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 5E 
[10:26:45][D][ACK:407]: Last heatpump data update successful!
```

If you have any additional information, I’m interested.  

In any case, please note that the external temperature sensor value is always sent at the end of a cycle.  

```cpp
void CN105Climate::terminateCycle() {
    if (this->shouldSendExternalTemperature_) {
        // We will receive an ACK packet for this.
        // Sending WantedSettings must be delayed in this case (lastSend timestamp updated).
        this->sendRemoteTemperature();
    }

    this->loopCycle.cycleEnded();

    if (this->hp_uptime_connection_sensor_ != nullptr) {
        // If the uptime connection sensor is configured,
        // we trigger a manual update at the end of a cycle.
        this->hp_uptime_connection_sensor_->update();
    }

    this->nbCompleteCycles_++;
}
```

Dale-Muccignat, 2025-03-14:
Unsure if this is specific to this issue, but I observed a similar behaviour with a MSZ-AP25VGD2 where it would revert back to the internal temp after 30 minutes (seemingly precisely) of no change to the remote sensor value. Red being the remote sensor and blue being the current temp reported from the aircon

[image]

[image]

anoopr, 2025-05-09:
This is happening to me as well on both a MVZ-A30AA7 and SVZ-KP18NA. They are both ducted air handlers. Heartbeat workaround fixed the issue on both heat pumps.

I'll test out the `update_interval` hypothesis and report back.

xeoMe015, 2025-05-09:
> This is happening to me as well on both a MVZ-A30AA7 and SVZ-KP18NA. They are both ducted air handlers. Heartbeat workaround fixed the issue on both heat pumps.
> 
> I'll test out the `update_interval` hypothesis and report back.

Could you share a daily chart of your temperature history? Maybe one before the fix and one after the fix?
My temperatures go up and down and i'm not exactly sure if this is expected behaviour.

luiggi78, 2025-05-23:
> Two units seem impacted: a MSZ-GL18NA-U1 and a MSZ-GL06NA-U1

Did you get it to work? I also have MSZ unit.

ams2990, 2025-06-14:
It took me a few tries to set 102 to 3, but I was able to get it eventually.

jimmy-zhening-luo, 2025-08-28:
> Also curious if using the heartbeat approach effectively defeats the `remote_temperature_timeout` setting? I used to use that approach with the GeoffDavis implementation, but not sure it's a good idea if my remote sensors fail (i.e. will heartbeat keep synthetically updating the temperature?)

I was thinking about this. @messnerdev , what do you think about something like this?

- Intuitively, this would refuse to emit either `nan` or out of range values.
- I also switched to using a priority throttle also prioritize emitting a `nan`.
- Using a timeout instead of a heartbeat seems more functionally correct -- keep blasting every 15 seconds but allow a real value to interrupt and reset the clock.

That said, logically if your unit is forgetting its remote temperature if not emitted again in 45s, that's already your effective (very bad) remote temperature timeout. All my filter changes do is better ensure that no values are emitted in the nan case for when your sensor is actually down. And, when your sensor is back, it will start emitting correctly again. It also survives things like Home Assistant restarts that happen in fewer than 45 seconds typically. That said, this is only suitable if your sensor is generally stable. And if any reader is seeing this and you have a very noisy sensor than you have the opposite problem so this solution (and this bug) don't apply to you most likely. Anyway here's the filter: 

```yaml
    filters:
      - throttle_with_priority:
          timeout: 1s
          value: [nan]
      - timeout:
          timeout: 15s
          value: last
      - lambda: return (x - 32.0) * (5.0/9.0); # if using °F
      - clamp:
          min_value: 1
          max_value: 40
          ignore_out_of_range: true
      - filter_out: nan
```

jimmy-zhening-luo, 2025-08-30:
> > I have the same issue, it is updating from the remote sensor, but it's mixing up for some reason. I also noticed my unit is not stopping when at target temperature, there seems to be a built-in offset of at least 1 degree.
> > [@echavet](https://github.com/echavet) any idea in which direction a solution can be found for this issue?
> 
> I have the same my AC has a offset of 1 degree and the current_temperature is set to 0.1 and the target_temperature is set to 0.5 however whenever my remote temperature measures 20.3 it mentions current temperature 20.5 so it’s using steps of 0.5 even though its configured as 0.1
> 
> Setting the AC to set temperature 19 degrees my remote temperature sensor bounces between 19.7 and 20.3. 20.3 makes it 20.5 so the unit stops and 19.7 makes it 19.5 so the unit starts. Not sure if this is normal behaviour?
> 
> I also have the heartbeat configured with 30s and one unit on 15s but same results for both.
> 
> Each few seconds i see cycle start, cycle ended in 0.8 seconds is this expected?
> 
> Outdoor unit: MXZ-3F54
> 
> Indoor units: MSZ-AY42 and MSZ-AY25

I am guessing this is not related to the remote temperature bug. Because I observe this behavior with or without remote temperature. If someone else sees this, does ___anybody___ have units that successfully use currently room temperature deltas smaller than 0.5°C?

jimmy-zhening-luo, 2025-08-30:
Incidentally, @ams2990 you can probably help with this too since your code also touched this last: I see that both your code (which in Fahrenheit mode uses your mapping) and the original took the final Celsius float and divided it by 2, yielding those half Celsius fluctuations we observe above. Can we therefore assume that the Mitsubishi units indeed can only understand even current room temperature in 0.5°C steps? Remote or not.

I will save my choice words for the genius engineers at Mitsubishi who designed one of the best, most efficient continuous heat pumps known to humankind, capable of maintaining a room temperature perfectly if it wanted to, yet decided to make it worthless by continuously wildly oscillate between two 0.5°C steps.

ams2990, 2025-08-30:
> I'm digging into all this, I'm also confused why this remote temperature update code was removed from a seemingly unrelated patch: [0e5e805](https://github.com/echavet/MitsubishiCN105ESPHome/commit/0e5e805b7b7cb56640373b44abd7437f6dcca2f7)
> 
> [@ams2990](https://github.com/ams2990) I think you mentioned there was some buggy code you removed. Since you've touched it, I'm wondering if you have any ideas off the top of your head about why remote temperature seems to not stick past 60 seconds?

It wasn't removed, the function just moved from componentEntries.cpp to climateControls.cpp, so it could use the `mapCelsiusForConversionFromFahrenheit` function I added to that file.

I don't have any insight into why the value doesn't stick. I observed this on my own units and worked around it with a 15s heartbeat.

> Incidentally, [@ams2990](https://github.com/ams2990) you can probably help with this too since your code also touched this last: I see that both your code (which in Fahrenheit mode uses your [...]

That's correct, if you look at the protocol, the 0.5°C steps are converted into integers when passed over the wire, so there's no room for a value in between.

> I will save my choice words for the genius engineers at Mitsubishi who designed one of the best, most efficient continuous heat pumps known to humankind, capable of maintaining a room temperature [...]

I also find this frustrating because my thermostats can measure in 0.5°F, but my air handler can do at best 1.0°F. To avoid the oscillation problem, I created a "lagging" sensor that ignores half-degree changes. This is a little different than rounding, because if the sensor moves from 1.0 to 1.5, it stays at 1, or if it moves from 2.0 to 1.5, it stays at 2. This helps reduce oscillation by requiring the temperature to move by a full degree. Here's what that looks like, if you're interested:
```jinja
{% set new =
     states('sensor.2nd_floor_temperature') | float %}
{% if new | round == new %}
  {{ new }}
{% else %}
  {% set old =
       states('sensor.2nd_floor_temperature_lagging') %}
  {% set rounding_mode = 
    'common'
    if old in ('unavailable', 'unknown') 
    else 'ceil'
    if new < old | int 
    else 'floor' %}
  {{ new | round(method=rounding_mode) }}
{% endif %}
```

jimmy-zhening-luo, 2025-08-30:
I am still reading / processing, but that filter you shared is brilliant and seems very promising to me. Thank you for sharing and for your fast response and help.

jimmy-zhening-luo, 2025-09-01:
Still wrapping my head around your template sensor; in the meantime I've implemented  it and am watching it, without feeding it to the heat pump yet (brain not ready for those second-order dynamics). For other people interested in quickly testing, here's the full template config version of ams2990's rounded temperature sensor:

```yaml
name: Climate Guest Temperature Rounded
unique_id: guest_temperature_rounded
state_class: measurement
device_class: temperature
unit_of_measurement: °F
availability: "{{ has_value('sensor.climate_guest_temperature') }}"
state: >-
  {% set new = states('sensor.climate_guest_temperature') | float %}
  {{
  new if new == new | round
  else new | round(
    0,
    'common' if not has_value(this.entity_id)
    else
      'ceil' if states(this.entity_id) | int > new
      else 'floor'
  )
  }}

```

ams2990, 2025-09-04:
> When heating (room temperature below 21.5 °C), round down the temperature => heating will not be terminated prematurely by rounding up; when cooling, round down and do not terminate cooling prematurely.

I used to use a similar strategy, but I observed two problems: 1) The oscillating. A small change in the ambient temperature could move my rounded sensor by a full degree F. The lagging sensor above is a lot more stable, because it requires a larger change in the input to produce a change in the output. 2) It was overcooling. I observed my unit to overshoot with "always round", to around 1.5°F, whereas it stays much closer with my approach above.

xeoMe015, 2025-09-04:
I think the devices just have a present build-in to cool or heat above the set temperature.
From the debug logs you can also see target -1.0

[14:43:16][D][SETTINGS:105]: [current]-> [power: OFF, target °C: 24.0, mode: HEAT, fan: AUTO, vane: ↑, wvane: ←]
[14:43:16][D][SETTINGS:105]: [received]-> [power: OFF, target °C: 24.0, mode: HEAT, fan: AUTO, vane: ↑, wvane: ←]
[14:43:16][D][EVT_SETS:062]: [wanted]-> [power: -, **target °C: -1.0,** mode: -, fan: -, vane: -, wvane: -, hasChanged ? ->  NO, hasBeenSent ? ->  NO]
[14:43:16][D][SETTINGS:089]: [climate]-> [mode: OFF, target °C: 24.0, fan: AUTO, swing: OFF]

My unit allows 0.5 degrees steps in target temperature via this MitsubishiCN105ESPHome project and also via MELCloud however the remote only allows 1 degree steps.
Also oversized units might go to their lowest available kwh ouput but if this is still to much the room overshoots in heating/cooling where it then stops/idles above the target if crossed by a specific range.

I use a thottle_average and with this when cooling is set to 21 my units stops when the remote temperature sensor has reached 19.7 (which then gets rounded by the heatpump to 19.5 and the unit continues the cooling when the remote temperature sensor goes to 20.3 gets arounded again to 20.5

disruptivepatternmaterial, 2025-09-05:
I think if you look at the code I shared in the discussion about a dual set point you will find (if I understand!) that the meta climate accounts for all this. You can set headbands, overruns, max/min run times, etc... 

https://github.com/echavet/MitsubishiCN105ESPHome/discussions/384

I have a couple more change to put in it, like some changes to the mapping of the modes, but other than that it VERY much made a difference in how these operate that I love. It is worth checking out...

ams2990, 2025-09-06:
@disruptivepatternmaterial wow, that is excellent! I've got a ton of automation in HA this could conceivably replace. I also like that I could write the automation in C++, which is far more enjoyable and maintainable than HN templates. I'll have to look into all of those settings you're configuring to understand how they work and can be applied to my own setup.

echavet, 2025-11-22:
> I setup a remote temperature sensor which is correctly updates the temperature based on the HA entity id.  However, the Current Temperate is regularly overwritten by the room temperature being provided by unit.  Note: this behavior is happening on only 1 of my 2 heat pumps (they are different models) configured with a remote thermostat.
> 
> 
> 
> I enabled DEBUG log and captured when it's happening.  In this example, the remote temp is reading 19.5C and the unit is reading 11C.
> 
> 
> 
> ```
> 
> [13:30:27][D][CN105:308]: sending a request for settings packet (0x02)
> 
> [13:30:27][I][CYCLE:039]: 1: Cycle start
> 
> [13:30:27][D][CYCLE:311]: 2a: Sending settings request (0x02)
> 
> [13:30:27][D][CN105:323]: creating Info packet
> 
> [13:30:27][D][CN105:084]: writing packet...
> 
> [13:30:27][D][WRITE:170]: FC 42 01 30 10 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7B 
> 
> [13:30:27][D][Header:091]: command: (62) data length: [10]<-- header
> 
> [13:30:27][D][READ:170]: FC 62 01 30 10 02 00 00 01 01 0F 02 05 00 00 00 A0 00 00 00 00 A3 
> 
> [13:30:27][D][chkSum:077]: OK-> A3=A3 
> 
> [13:30:27][D][CYCLE:315]: 2b: Receiving settings response
> 
> [13:30:27][D][Decoder:173]: [0x02 is settings]
> 
> [13:30:27][D][Decoder:181]: [Power : ON]
> 
> [13:30:27][D][Decoder:182]: [iSee  : 0]
> 
> [13:30:27][D][Decoder:183]: [Mode  : HEAT]
> 
> [13:30:27][D][Decoder:195]: [Temp °C: 16.000000]
> 
> [13:30:27][D][Decoder:198]: [Fan: 1]
> 
> [13:30:27][D][Decoder:201]: [Vane: ↓↓]
> 
> [13:30:27][W][lookup:204]: wideVane reading caution: value 0 not found, returning value at index 0
> 
> [13:30:27][D][Decoder:206]: [wideVane: ←← (adj:0)]
> 
> [13:30:27][D][SETTINGS:113]: [current        ]-> [power: ON , target °C: 16.0, mode: HEAT  , fan: 1     , vane: ↓↓, wvane: ←←]
> 
> [13:30:27][D][SETTINGS:113]: [received       ]-> [power: ON , target °C: 16.0, mode: HEAT  , fan: 1     , vane: ↓↓, wvane: ←←]
> 
> [13:30:27][D][EVT_SETS:072]: [wanted         ]-> [power: -  , target °C: -1.0, mode: -     , fan: -     , vane: -     , wvane: -     , hasChanged ? ->  NO, hasBeenSent ? ->  NO]
> 
> [13:30:27][D][SETTINGS:087]: [climate]-> [mode: HEAT, target °C: 16.0, fan: LOW, swing: OFF]
> 
> [13:30:27][D][CN105:218]: setting action to -> 4
> 
> [13:30:27][D][CN105:267]: Climate mode is: 3
> 
> [13:30:27][D][CN105:268]: Climate action is: 4
> 
> [13:30:27][D][climate:396]: 'Heat Pump - Living Room' - Sending state:
> 
> [13:30:27][D][climate:399]:   Mode: HEAT
> 
> [13:30:27][D][climate:401]:   Action: IDLE
> 
> [13:30:27][D][climate:404]:   Fan Mode: LOW
> 
[... 119 lines cut ...]
> 
>  climate:
> 
>    - platform: cn105
> 
>      name: ${friendly_name}
> 
>      id: hp
> 
>      remote_temperature_timeout: 30min
> 
>      update_interval: 4s
> 
>      debounce_delay : 100ms
> 
> ```
> 
> 
> 
> Lastly, this is easily reproducible on my side.  It oscillates between the two temp readings every few seconds, happy to provide any more info if it'd be helpful.

Hi,
Did you find a cause for your issue?
If not, can you tell me more about the sensor you're using as ${remote_temp_sensor}.
I have faced the same issue yesterday with a zigbee thermometer which was very lazy with sending sensor updates. And it looks like my Mitsubishi unit ends up toggling to internal sensor when no external temperature is sent.

echavet, 2025-12-06:
> I had similar issues when relying on the update frequency of zigbee sensors. The controller seems to like frequent updates, even if the value does not change.

Yes I have exactly the same issue.
I has to set the timeout to 1 hour and to change the sonoff SNZB-02D reports frequency.

Max Rep Interval and Min Rep Change parameters can be set from zigbee2MQTT report section for the device. 
You have to press the back button on the device to allow it to listen to z2mqtt change requests. 

"Min Rep change" must be set to a smaller value : 10 or 20 for 0.1 or 0.2 range.

kchen, 2025-12-25:
I don't know if this is the case for anyone else here, but when I initially encountered this problem, I had connected my ESP32 to my PAC-USWHS002-WF-2 (and removed the WF-2's remote sensor in the Mitsubishi Comfort app), and the WF-2 was then connected to my indoor unit. I tried several of the workarounds in this thread, and they failed to fix the problem.

However, once I removed the WF-2 and connected the ESP32 directly to the indoor unit's CN105 port, I believe the remote temperature sensor defined in the ESP32 started working properly for me.

I assume that once I removed the WF-2's remote sensor, the WF-2 may have started instructing the unit to use its internal sensor, so even if I attempted to use the "Set Functions" workaround in this ticket, the WF-2 may have still been overwriting that setting.

ams2990, 2025-12-25:
Not the case for me -- I had this problem with direct connection to the indoor unit.

raiseandfall, 2026-01-19:
@rudizl you can add the following config to your `climate`: https://github.com/echavet/MitsubishiCN105ESPHome/issues/204#issuecomment-2973215545

01awolf, 2026-05-02:
Hi @echavet I have also the same sensors SNZB-02P.
My Max rep interval is set to 3600 so I changed my remote_temperature_timeout to 65 mins.
Let's see if it's correcting the issue.

BTW great work ! Cocorico !

echavet, 2026-05-02:
In addition to ignoring the maximum reporting interval, the SNZB-02D sensor also fails to respect the Min rep change (Minimum Reporting Change) configuration in Zigbee2MQTT. This parameter is designed to let users define the exact temperature or humidity variation (e.g., 0.1°C or 0.2°C) required to trigger a new data transmission. However, the device typically relies on its own hardcoded factory thresholds to maximize battery life. Because it ignores these custom sensitivity settings, the sensor remains completely silent during periods of slow temperature variation or perfect environmental stability, creating problematic data gaps.

echavet, 2026-05-03:
You need to press the button on the device so as it records the new values

echavet, 2026-09-15:
## Maintainer update / roadmap

Still **P1**: remote temperature must not be overwritten by the indoor unit’s local reading when remote temp is active.

We will treat this carefully (multi-model): any fix should be gated on “remote temperature mode engaged”, with clear keepalive/timeout behaviour — not a blanket ignore of all room-temp frames for every install.

If you can still reproduce on current `main`, a short log with remote temp enabled showing the overwrite (Decoder room °C vs remote) would help prioritise the PR.