echavet/MitsubishiCN105ESPHome #206: Don't send updates if value doesn't change

melyux, 2024-12-29:
I noticed this component sends an update on every topic at every `update_interval`. Since it's retained, is there any reason to send updates for a topic if its value hasn't changed?

phidauex, 2025-01-03:
Hi, could you clarify the concern a bit? If you mean sending updates to the indoor unit, then that I believe is mandatory - each update cycle to the indoor unit needs to include all parameters, even if they didn't change.

If you mean updating HA entities through the HA api, then I suppose it could be considered redundant, but there is also a benefit in knowing on each update that, for instance, the status is still "heating", even if that was the same status in the last update, because at least you know the status of the point is fresh.

melyux, 2025-01-03:
I do mean updating the HA entities. I would expect

melyux, 2025-01-03:
I do mean updating the HA entities, or more like the MQTT topics. Since we're retaining all of the MQTT messages, we don't need to update every topic unless it's changed. The "status" topic already tells us if the ESP is online, so we need not fear that any topic is stale

phidauex, 2025-01-03:
Ah, I see what you mean. The intended use of this component for most people would be to connect to HomeAssistant primarily, and if that is your use case, then the recommended way is to use the HA native API, which is considerably faster and more network efficient (https://esphome.io/components/api.html), so I don't think many folks are using MQTT except for testing.

If you need MQTT to talk to some other system, then I'd look more into the ESPHome MQTT configuration - this component doesn't implement anything unique, it just uses the default capabilities. The `idf_send_async` parameter could be helpful here? You could also slow down the update rate - 1s is quite fast, I find 2 to 4s to still provide good latency, and there isn't much value in sending temperature updates faster than that.

Beyond that, PRs would be welcome, but reworking the mqtt update logic would probably not be a high priority given some of the other enhancements ahead in line, and supporting additional units.

melyux, 2025-01-03:
I see in the code that this line:

https://github.com/echavet/MitsubishiCN105ESPHome/blob/89e525a51288245a5df82120980044d5ed53eb12/components/cn105/hp_readings.cpp#L434-L436

still goes into the if block even though the logs show no change between the `status` and `currentStatus`:

```
[15:20:36][I][STATUS:134]: [received       ]-> [room C°: 21.0, operating: NO , compressor freq: 0.0 Hz]
[15:20:36][I][STATUS:134]: [current        ]-> [room C°: 21.0, operating: NO , compressor freq: 0.0 Hz]
```

Wonder if that's a possible cause.

melyux, 2025-01-04:
I was on the right track there, but the settings and stage/submode settings needed further work. PR out in #210

phidauex, 2025-01-04:
Thanks for rising to the topic and submitting a PR! I'll test soon, and @echavet should probably take a look since most of this code is his.

PR #210 (2025-01-05): Publish updates only on changes
There was already a check to only publish new status values when the status changed, but it had a bug: when comparing two `nan` values for `outsideAirTemperature`, the equality check would fail because `nan == nan` always fails in C++.

There was also already a check to only publish new setting values when the settings changed, but the line wasn't included in the otherwise empty if block (maybe by mistake).

Stage, sub mode, and auto sub mode statuses needed new checks, since it appears they were excluded from the settings by commenting out `// this->heatpumpUpd [... cut]
closed 2025-01-05 by echavet