echavet/MitsubishiCN105ESPHome #208: Feature request: Night mode

siljelb, 2024-12-30:
First of all, thank you for this excellent project!

My MSZ-LN35VG2W is working great with the basic functionality, but it also has a "night mode" which makes the outdoor unit quieter. I'd love to be able to set this automatically from HA instead of having to do it manually. Is this feasible to implement? Whether as a fan mode or maybe as a separate switch?

echavet, 2024-12-31:
Sure it's possible through an automation:
It would look something like this:

'''
action: climate.set_fan_mode
data:
  fan_mode: QUIET
target:
  entity_id: climate.esp32_hp_clim_sejour
'''

echavet, 2024-12-31:
But I did not test

siljelb, 2024-12-31:
The "night mode" apparently isn't a fan mode as such, but a separate parameter of some sort. I've just tested specifically, and none of the states or parameters currently visible in HA change when I turn on and off night mode:
## Night mode off
[image]
```yaml
hvac_modes:
  - "off"
  - cool
  - heat
  - fan_only
  - dry
  - auto
min_temp: 10
max_temp: 31
target_temp_step: 1
fan_modes:
  - auto
  - low
  - medium
  - high
  - middle
  - quiet
swing_modes:
  - "off"
  - both
  - vertical
  - horizontal
icon: mdi:heat-pump
friendly_name: Varmepumpe
supported_features: 425
current_temperature: 21.5
temperature: 21
fan_mode: auto
hvac_action: heating
swing_mode: "off"
```

## Night mode on
[image]
```yaml
hvac_modes:
  - "off"
  - cool
  - heat
  - fan_only
  - dry
  - auto
min_temp: 10
max_temp: 31
target_temp_step: 1
fan_modes:
  - auto
  - low
  - medium
  - high
  - middle
  - quiet
swing_modes:
  - "off"
  - both
  - vertical
  - horizontal
icon: mdi:heat-pump
friendly_name: Varmepumpe
supported_features: 425
current_temperature: 21.5
temperature: 21
fan_mode: auto
hvac_action: heating
swing_mode: "off"
```

siljelb, 2024-12-31:
Closed in error.

phidauex, 2024-12-31:
Is this just set from an IR remote control? Or is this a standalone thermostat that connects using the CN105 port on the indoor unit? If it is set from an external thermostat, then it may be possible to determine the serial command associated with it, but since this all has to be reverse engineered from scratch, someone would have to setup to do the logging while the commands are sent. Some examples of this type of debugging are in this PR: https://github.com/echavet/MitsubishiCN105ESPHome/pull/166

siljelb, 2024-12-31:
It's set from the IR remote; apart from the ESP32 I don't have anything to connect to the CN105. I tried something I read in a different PR, setting READ, DECODER and LOGS to DEBUG while sending the command, but I don't know whether this achieves anything like what you're describing? In any case, here are the debug logs where I turned "night mode" on and off again three times immediately after the "19:36:11" log entry.
[debug_log_nightmode.txt](https://github.com/user-attachments/files/18283219/debug_log_nightmode.txt)

melyux, 2025-01-05:
It seems like a preset that's not actually a distinct mode on the unit. Mine has "econocool" like this, where it just sets the temperature up 2 degrees and turns on swing. So it might be something you handle at the Home Assistant level, and/or as a preset for the climate entity.

siljelb, 2025-01-05:
Additionally the outdoor unit will have a lower noise level, which is my main reason for wanting to control the night mode feature. As far as I can understand, it does this without changing the set temperature.

I agree the "circulator" mode is more conceptually similar to the "econocool" mode that other units have, and I guess it could easily be emulated in HA by an automation switching between "heating" and "fan only" based on the set temp vs measured temp.

robdraaijer, 2025-01-06:
I would also be interested in night mode support. Based on the manual i would believe that it would limit the maximum compressor speed at night, which might also improve the COP of the system.

echavet, 2025-03-11:
hi all,

first : sorry for my first reply, I haven't noticed it was about the external unit.
then: I'd love to be able to implement this feature but I think I can't! Correct me if I'm wrong, but this special feature is surely a new paquet type.

So I need 2 conditions to be able to enhance the projets:
- having myself a unit that supports the night mode
- having the melcloud or other options to "reverse-engine" the protocol or to sniff the melcloud api (like I've done with my hitachi yutampo )

But anybody who feels confident in making it possible is welcome. I 'd be happy to help.

lrehmann, 2025-04-06:
Wanted to explore if there is any similarity between the data sent via the IR remote and the serial commands. I've used my flipper Zero to record the enable/disable of night mode, labeled "Sleep" on my MXZ-series remote

In this sample, my remote was set to heat at 68ºF, Auto Fan, Full Range Vane. Upon pressing Night [Sleep] mode, the remote seems to default to 72ºF.

Night Mode Enable:
`13 4D 92 00 00 10 41 A0 00 78 00 00 00 00 08 00 03 43`

Night Mode Disable:
`13 4D 92 00 00 10 40 80 00 78 00 00 00 00 00 00 02 77`

Temp change to 66ºF:
`13 4D 92 00 00 10 41 20 00 78 00 00 00 00 00 00 03 8F`

frustrated with the similarity but lack of clarity i gained when comparing to the wiki data, I recorded 4 more, this time in cool mode:

Cool, 69ºF
`13 4D 92 00 00 10 62 81 80 78 00 00 00 00 00 00 00 2F`

Cool, Night Mode 83ºF
`13 4D 92 00 00 10 60 C1 80 78 00 00 00 00 08 00 03 A3`

Cool, Night Mode 82ºF
`13 4D 92 00 00 10 63 61 80 78 00 00 00 00 08 00 01 93`

Cool, 68ºF
`13 4D 92 00 00 10 60 81 80 78 00 00 00 00 00 00 03 CF`

Across the heat/cool modes, it appears the 15th byte enables/disables sleep/night mode on the IR remote.

I would think that some part of the IR receiver component is translating these values into serial commands, but maybe they're interpreted further down the line. 

Hopefully my AI-assisted IR decoding didn't just spit out a bunch of garbage. Either way, it was a fun rabbit hole. Don't think it is going to provide any clarity on what serial commands to use for sleep mode.

Similar reading: [reverse engineering ir](https://www.analysir.com/blog/2015/01/06/reverse-engineering-mitsubishi-ac-infrared-protocol/). 

python code
[mitsubishi-ir-to-serial-hex.py.txt](https://github.com/user-attachments/files/19620131/mitsubishi-ir-to-serial-hex.py.txt)

IR Signal Capture (Heat)
[Mitsubishi-Remote.ir.txt](https://github.com/user-attachments/files/19620172/Mitsubishi-Remote.ir.txt)
(Cool)
[Cool69.ir.txt](https://github.com/user-attachments/files/19620322/Cool69.ir.txt)

lrehmann, 2025-04-13:
Digging into the I.R receiver (REC / Receive P.C. Board) a bit more, it's setup on port CN301/CN125.
It appears to be a dumb component that just passes on IR commands, don't think there'd be anywhere to MITM. 

[image]
[image]
[image]
[image]

siljelb, 2025-08-12:
Looks like it's working great! Amazing work! 🙌