echavet/MitsubishiCN105ESPHome #211: Project thinks my cooling-only unit is "heating" when it's drying in auto mode

melyux, 2025-01-05:
This is due to this logic here that assumes that if the unit set to auto, the temperature is below the setpoint, and the unit is active that it's heating. For my unit, it's actually drying in this case.

https://github.com/echavet/MitsubishiCN105ESPHome/blob/89e525a51288245a5df82120980044d5ed53eb12/components/cn105/climateControls.cpp#L249-L254

echavet, 2025-01-13:
Hi @melyux,
thanks for your report
Does the "auto mode" make sense with your cooling-only unit ?
What would you expect your unit to do ?

disruptivepatternmaterial, 2025-01-13:
Oh yes, I always knew this was not right...just worked as long as it wasn't one of the cool-only units. I am not sure what exactly the solution is - I have never encountered a cool-only unit, so I couldn't test it. Sorry about that... if you give us some sense of what the behavior is, I can also ponder a solution. It might be as simple as looking for the available modes and, if it is only COOL and OFF, to do something different.

melyux, 2025-01-16:
@somlefant It's a model NAYWST24A112AA unit. The manual says the "auto" mode is actually the "I FEEL" mode you mentioned. Auto sub mode doesn't seem to be supported on this, it never changes from the default. I'll check what happens in the logs in a little bit. Here's the tidbit from the manual on how "I FEEL" works on this thing:

[image]

@echavet @disruptivepatternmaterial It seems like it implements the "I FEEL" mode when in auto, check out this above part from the manual explaining the function

melyux, 2025-02-21:
I have the logs for when AUTO mode is on (but the unit is always idle and can't run right now because it's too cold inside for it to operate). Let me know if logs when it's running are required.

```
[01:21:13][D][CN105:308]: sending a request for settings packet (0x02)
[01:21:13][I][CYCLE:039]: 1: Cycle start
[01:21:13][D][CYCLE:311]: 2a: Sending settings request (0x02)
[01:21:13][D][CN105:323]: creating Info packet
[01:21:13][D][CN105:084]: writing packet...
[01:21:13][D][WRITE:170]: FC 42 01 30 10 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7B 
[01:21:13][D][Header:091]: command: (62) data length: [10]<-- header
[01:21:13][D][READ:170]: FC 62 01 30 10 02 00 00 01 08 09 00 00 00 00 04 AC 00 00 00 00 99 
[01:21:13][D][chkSum:077]: OK-> 99=99 
[01:21:13][D][CYCLE:318]: 2b: Receiving settings response
[01:21:13][D][Decoder:176]: [0x02 is settings]
[01:21:13][D][Decoder:184]: [Power : ON]
[01:21:13][D][Decoder:185]: [iSee  : 0]
[01:21:13][D][Decoder:186]: [Mode  : AUTO]
[01:21:13][D][Decoder:198]: [Temp °C: 22.000000]
[01:21:13][D][Decoder:201]: [Fan: AUTO]
[01:21:13][D][Decoder:204]: [Vane: AUTO]
[01:21:13][D][Decoder:209]: [wideVane: → (adj:0)]
[01:21:13][D][SETTINGS:113]: [current        ]-> [power: ON , target °C: 22.0, mode: AUTO  , fan: AUTO  , vane: AUTO  , wvane: →   ]
[01:21:13][D][SETTINGS:113]: [received       ]-> [power: ON , target °C: 22.0, mode: AUTO  , fan: AUTO  , vane: AUTO  , wvane: →   ]
[01:21:13][D][EVT_SETS:072]: [wanted         ]-> [power: -  , target °C: -1.0, mode: -     , fan: -     , vane: -     , wvane: -     , hasChanged ? ->  NO, hasBeenSent ? ->  NO]
[01:21:13][D][SETTINGS:087]: [climate]-> [mode: AUTO, target °C: 22.0, fan: AUTO, swing: OFF]
[01:21:13][D][CYCLE:321]: 3a: Sending room °C request (0x03)
[01:21:13][D][CN105:323]: creating Info packet
[01:21:13][D][CN105:084]: writing packet...
[01:21:13][D][WRITE:170]: FC 42 01 30 10 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7A 
[01:21:14][D][Header:091]: command: (62) data length: [10]<-- header
[01:21:14][D][READ:170]: FC 62 01 30 10 03 00 00 03 00 00 9A 00 00 00 00 00 00 00 00 00 BD 
[01:21:14][D][chkSum:077]: OK-> BD=BD 
[01:21:14][D][CYCLE:327]: 3b: Receiving room °C response
[01:21:14][D][Decoder:256]: [Room °C: 13.000000]
[01:21:14][D][Decoder:257]: [OAT  °C: nan]
[01:21:14][D][CYCLE:330]: 4a: Sending status request (0x06)
[01:21:14][D][CN105:323]: creating Info packet
[01:21:14][D][CN105:084]: writing packet...
[01:21:14][D][WRITE:170]: FC 42 01 30 10 06 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 77 
[01:21:14][D][Header:091]: command: (62) data length: [10]<-- header
[01:21:14][D][READ:170]: FC 62 01 30 10 06 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 57 
[01:21:14][D][chkSum:077]: OK-> 57=57 
[01:21:14][D][CYCLE:348]: 4b: Receiving status response
[01:21:14][D][Decoder:280]: [0x06 is status]
[01:21:14][D][CYCLE:352]: 5a: Sending power request (0x09)
[01:21:14][D][CN105:323]: creating Info packet
[01:21:14][D][CN105:084]: writing packet...
[01:21:14][D][WRITE:170]: FC 42 01 30 10 09 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 74 
[01:21:14][D][Header:091]: command: (62) data length: [10]<-- header
[01:21:14][D][READ:170]: FC 62 01 30 10 09 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 53 
[01:21:14][D][chkSum:077]: OK-> 53=53 
[01:21:14][D][CYCLE:367]: 5b: Receiving Power/Standby response
[01:21:14][D][Decoder:148]: [0x09 is sub modes]
[01:21:14][D][Decoder:155]: [Stage : LOW]
[01:21:14][D][Decoder:156]: [Sub Mode  : NORMAL]
[01:21:14][D][Decoder:157]: [Auto Mode Sub Mode  : AUTO_OFF]
[01:21:14][I][CYCLE:052]: 6: Cycle ended in 0.9 seconds (with timeout?:  NO)
```

echavet, 2025-03-11:
hey @melyux,

could you try to use this version?

```yaml
external_components:
  - source: github://echavet/MitsubishiCN105ESPHome@echavet/issue211
     refresh: 0s
```

I did not choose to fall back to dry but to FAN_ONLY. I'm not sure this is a relevent choice. Tell me please...

commit da7038d7 (2025-04-13): Project thinks my cooling-only unit is "heating" when it's drying in auto mode
Fixes #211
commit ed7447ac (2025-04-13): Project thinks my cooling-only unit is "heating" when it's drying in auto mode
Fixes #211
closed 2025-04-13 by echavet