echavet/MitsubishiCN105ESPHome #221: Feature request: Add the "Smart Set" function (heat 10 degrees setpoint)

jfdrolet11, 2025-01-17:
My unit has the “Smart Set” function. This function allows the unit to be set in heat mode at 10 degrees. This is handy when you're away from home for a long time. 

I've tried to do a script in Home Assistant, but it doesn't work. It doesn't take the 10 degrees setpoint, it sets the heating to 16. 

Is-it possible to add a button for this function in ESPHome?

phidauex, 2025-02-02:
This would be handy, but I think we'd need to capture the serial command that enables the feature. A few people have gone about doing that work, but I'm not the expert on it.

rkboni, 2025-02-07:
AFAIR based on discussions in https://github.com/geoffdavis/esphome-mitsubishiheatpump, there really isn't a serial command that does this, the remote just sends the individual mode / temp / ... requests as a bundle.  I've faked this on my devices by just setting the temp range to start at 10 degC and manually adjusting it to that point:

```
climate:
  - platform: cn105
    name: "${friendly_name}"
    id: hp
    icon: mdi:heat-pump
    visual:
      min_temperature: 10
      max_temperature: 31
      temperature_step:
        target_temperature: 1
        current_temperature: 0.1
```

phidauex, 2025-02-07:
Interesting, I didn't realize you could just set a min temperature that low and it would accept the value. Does it accept intermediate values like 12C? Or just 10C and then jumping up to the 15C normal minimum?

rkboni, 2025-02-07:
I haven't tested values between 10C and 15C in any sense, though nothing
blew up when I manually decreased the temp from 15C to 10C ;)

rkboni, 2025-02-08:
> I haven't tested values between 10C and 15C in any sense, though nothing
> blew up when I manually _stepped_ the temp from 15C to 10C ;)

It does look like those intermediate values work -- just set my gym minisplit to 52 F (from the previous 50) and it is heating... will report if it stops at 52 F or not 😁

jfdrolet11, 2025-02-09:
Thanks @rkboni for the workaround! It's working perfectly. I will close my feature request. I have added a button in the ESPHome to active it more easily. 

`button:
  - platform: template
    name: "Smart Set"
    icon: mdi:heat-wave
    on_press:
     - climate.control:
         id: hp
         mode: "HEAT"
         target_temperature: 10
         swing_mode: "OFF"
         fan_mode: "AUTO"`

closed 2025-02-09 by jfdrolet11

phidauex, 2025-02-10:
I'll drop the min temp setting in the documentation defaults to 10C, and add a note about SmartSet features to make it easier for others to find.