echavet/MitsubishiCN105ESPHome #222: Current air temp too high

tdm-fr, 2025-01-19:
Hello,

D'abord un grand merci pour ce projet, c'est juste formidable de pouvoir domotiser aussi facilement ses clims !
(for non french readers, I just thanks the author 😁)

I finished building up and installing in the AC the ESP modules yesterday, in terms of control everything is working perfectly.
But in terms of gathered data I noticed something strange regarding the "current air temp" : it's always between 1.5 and 2.5°C above real temp.

For example : 
[image]

The night was cold : 3°C when I woke up at 8am, now it's 10°C.
It's impossible that the "Chambre" that's off (and was off since yesterday 11pm) can read 22°C
Also, the 2 AC of the "Salon" are set to 17°C since yesterday 11pm, my thermostat reads 18°C (and I feel it !), so the temp is not 20°C or 19°C.

These temps do update frequently, gaining or losing 0.5°C, but they still are above the real temp.

Any idea on why ?

tdm-fr

jamer007, 2025-01-23:
Bonjour tdm-fr.
I have experienced the same behavior when my esphome configuration was not using a remote sensor for the current temperature but was still using the internal minisplit thermometer. Possibly because the heat source and the height where is the minisplit, the minisplit temperature reading is higher than other thermometers in the room.
If it is not the case, I strongly suggest you to configure the remote sensor, more comfort and better energy usage (less short cycle).

Cheers.

tdm-fr, 2025-01-23:
Salut jamer007,

I thought that this temp value was not gathered/read/interpreted correctly by the ESP or HA : how come a famous HVAC vendor failed to locate the temp sensor in a more reliable spot in the unit ?!
And now I understand better why the unit in my room was making shorts cycle (5-10 seconds) the whole night, it was driving me crazy and I ended with turning off the heat when I sleep.
My main goal with the ESP was to have control of my HVAC in HA, but having the possibility to enhance them with the external sensor is just phenomenal, I just have to buy some zigbee sensor to do that and uncomment some lines in the code, easy peasy !

Thanks ! 😊

echavet, 2025-01-23:
hey guys,
thanks for your messages.
Could you provide your yaml configuration files? I wonder wether you configured your esp firmware to use external or internal temperature sensor.
Could you provide a homeassistant graph showing an entire 24h period of set, and room temperature?

For info my 2 units are also always heating the room 0.5 to 1.5°C above the defined set point. This is an interesting observation!

tdm-fr, 2025-01-23:
Hello,

You are totally right, I didn't pay attention to this part. I used the full example you gave, so I have : 
`sensor:
  - platform: homeassistant
    name: "Remote Temperature Sensor"
    entity_id: ${remote_temp_sensor} # Replace with your HomeAssistant remote sensor entity id, or include in substitutions
    internal: false
    disabled_by_default: true
    device_class: temperature
    state_class: measurement
    unit_of_measurement: "°C"`

What should I change (except the internal boolean to true I guess) to use the internal temp sensor ?

phidauex, 2025-01-24:
To use a remote temperature sensor, you'd include this in your config:

```yaml
sensor:
  - platform: homeassistant
    name: "Remote Temperature Sensor"
    entity_id: ${remote_temp_sensor} # Replace with your HomeAssistant remote sensor entity id, or include in substitutions
    internal: false
    disabled_by_default: true
    device_class: temperature
    state_class: measurement
    unit_of_measurement: "°C"
    filters:
    # Uncomment the lambda line to convert F to C on incoming temperature
    #  - lambda: return (x - 32) * (5.0/9.0);
      - clamp: # Limits values to range accepted by Mitsubishi units
          min_value: 1
          max_value: 40
          ignore_out_of_range: true
      - throttle: 30s
    on_value:
      then:
        - logger.log:
            level: INFO
            format: "Remote temperature received from HA: %.1f C"
            args: [ 'x' ]
        - lambda: 'id(hp).set_remote_temperature(x);'
```

You would then change `${remote_temp_sensor}` to the entity ID of your Home Assistant remote temperature sensor (your Zigbee sensor or whatever you get). What this section means is "Read the sensor from Home Assistant with this entity_id, define it as a temperature. When the value changes, send the sensor value to the function set_remote_temperature."

So if you have a temperature sensor you can feed it from HA, then whenever that updates in HA, the ESPHome device will update the temperature value in the heatpump. If it stops updating, then the heatpump will revert to the internal sensor.

So to use the internal sensor, you can just comment out this entire section! When you have a remote sensor to use, uncomment it and change `entity_id` to your sensor, IE, `sensor.living_room_temperature` or whatever.

`internal: false` doesn't mean internal temp sensor in this case - it is an ESPHome value saying "is this sensor fully internal to the ESPHome device? Or should I show it to HomeAssistant?" TRUE would mean that the device won't send the value to HomeAssistant. FALSE means it sends the value to HomeAssistant, but the entity is disabled by default. This is just there to make people's lives a bit easier - if they want to verify the remote sensor value the unit is seeing from within HA, then they can enable the entity and look at it. Hope that helps.

tdm-fr, 2025-01-24:
Wow thanks @phidauex for the detailled explanation !
And yes last evening I commented this code section in my 5 ESP and nothing changed : there must be a fallback to revert to the internal sensor if there is no external sensor available, pretty clean.
And actually the screenshot I gave is not the most representative, in a smaller bedroom it's way worse.

I'm browsing through external sensors and I think I will try the Sonoff ones.

jamer007, 2025-01-25:
Something interesting that you can do with Home Assistant is to create a "virtual" sensor, combining the data from nultiple thermometer, an issue i had with only a simple zigbee sensor was the update frequency was very slow and trigger the esphome to fallback tonthe internal sensor. 
If you want to use the remote sensor feature, use as temperature source something that refresh as fast as possible.

tdm-fr, 2025-01-25:
I read about that actually. If the sensor doesn't refresh the data for a while the connection appears to be lost, even though the sensor is still connected.
That can happens on Sonoff sensors because of their battery optimisation. The temperature can be updated every 5s if there is a change of 0.2°C. If no significant change for 15 minutes the temperature is update nevertheless. 
In this case the only solution is the virtual sensor : to keep track of the value and never "loose connection".

closed 2025-01-25 by tdm-fr