echavet/MitsubishiCN105ESPHome #233: Installer Unit Functions Codes and Installer Setup Options

buryboi, 2025-02-10:
First of all thank you for developing this project! Controlling 4 units in home assistant has been great! 

I see the configuration to get and set values for heatpumpfunctions.    How might one go about setting the "mode" and "setting" or ISU and associated options? For example the MHK2 instructions indicate that Mode 2, Setting 3 enables remote temperature.

I've been reconnecting the MHK2 to adjust things like fahrenheit/Celsius, remote temp, Optional humidifier, optional ERV etc. Once they're set, they probably don't need messed with, but it would be cool to have this functionality so I can get rid of the MHK2's

Just to share the info. Indoor units are SVZ-KP12's and SVZ-KP24 Outdoors are MXZ-4C36. They do not register compressor frequency, ISEE Sensor, Energy Usage, Input Power.  I get swing errors (but assume that's due to being ducted units without horizontal it vertical swing).

Thanks again!

echavet, 2025-02-19:
Hi @buryboi 
sorry I have copied and pasted the heatpumpfunctions part of the original SwiCago library but I actually did not test anything about these features.
I don't own any MHK2. Sure it would be great to implements these features but the protocole is not known and this work is not planned at all. 
But thank you for all the infos.

closed 2025-03-08 by echavet

avwuff, 2025-04-10:
FYI I had a need for same and have an implementation you can use in #265