echavet/MitsubishiCN105ESPHome #236: Error compiling on esphome 2025.2.0

Mense1978, 2025-02-19:
I got those errors:

INFO ESPHome 2025.2.0
INFO Reading configuration /config/esphome/aa-mitsubishi-dormitorioprincipal.yaml...
INFO Detected timezone 'Europe/Madrid'
INFO Generating C++ source...
INFO Compiling app...
Processing aa-mitsubishi-dormprincipal (board: d1_mini; framework: arduino; platform: platformio/espressif8266@4.2.1)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- ESPAsyncTCP-esphome @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 1.1.1
|-- ESP8266WiFi @ 1.0
|-- ESP8266mDNS @ 1.2
|-- noise-c @ 0.1.6
Compiling .pioenvs/aa-mitsubishi-dormprincipal/src/esphome/components/cn105/climateControls.cpp.o
Compiling .pioenvs/aa-mitsubishi-dormprincipal/src/esphome/components/cn105/cn105.cpp.o
Compiling .pioenvs/aa-mitsubishi-dormprincipal/src/esphome/components/cn105/componentEntries.cpp.o
Compiling .pioenvs/aa-mitsubishi-dormprincipal/src/esphome/components/cn105/cycle_management.cpp.o
In file included from src/esphome.h:39,
                 from src/esphome/components/cn105/Globals.h:2,
                 from src/esphome/components/cn105/cn105.h:2,
                 from src/esphome/components/cn105/climateControls.cpp:1:
src/esphome/components/cn105/uptime_connection_sensor.h:2:10: fatal error: esphome/components/uptime/uptime_seconds_sensor.h: No such file or directory
    2 | #include "esphome/components/uptime/uptime_seconds_sensor.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/aa-mitsubishi-dormprincipal/src/esphome/components/cn105/climateControls.cpp.o] Error 1
In file included from src/esphome.h:39,
                 from src/esphome/components/cn105/Globals.h:2,
                 from src/esphome/components/cn105/cn105.h:2,
                 from src/esphome/components/cn105/cn105.cpp:2:
src/esphome/components/cn105/uptime_connection_sensor.h:2:10: fatal error: esphome/components/uptime/uptime_seconds_sensor.h: No such file or directory
    2 | #include "esphome/components/uptime/uptime_seconds_sensor.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/aa-mitsubishi-dormprincipal/src/esphome/components/cn105/cn105.cpp.o] Error 1
In file included from src/esphome.h:39,
                 from src/esphome/components/cn105/Globals.h:2,
                 from src/esphome/components/cn105/cn105.h:2,
                 from src/esphome/components/cn105/componentEntries.cpp:1:
src/esphome/components/cn105/uptime_connection_sensor.h:2:10: fatal error: esphome/components/uptime/uptime_seconds_sensor.h: No such file or directory
    2 | #include "esphome/components/uptime/uptime_seconds_sensor.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/aa-mitsubishi-dormprincipal/src/esphome/components/cn105/componentEntries.cpp.o] Error 1
In file included from src/esphome.h:39,
                 from src/esphome/components/cn105/Globals.h:2,
                 from src/esphome/components/cn105/cn105.h:2,
                 from src/esphome/components/cn105/cycle_management.cpp:2:
src/esphome/components/cn105/uptime_connection_sensor.h:2:10: fatal error: esphome/components/uptime/uptime_seconds_sensor.h: No such file or directory
    2 | #include "esphome/components/uptime/uptime_seconds_sensor.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/aa-mitsubishi-dormprincipal/src/esphome/components/cn105/cycle_management.cpp.o] Error 1
========================== [FAILED] Took 1.09 seconds ==========================

For your info, do not update esphome

echavet, 2025-02-19:
@Mense1978 thanks

Mense1978, 2025-02-19:
For your info
Still gettiong errors.
Compiling .pioenvs/aa-mitsubishi-porchemesa/src/esphome/components/cn105/climateControls.cpp.o
In file included from src/esphome.h:39,
                 from src/esphome/components/cn105/Globals.h:2,
                 from src/esphome/components/cn105/cn105.h:2,
                 from src/esphome/components/cn105/climateControls.cpp:1:
src/esphome/components/cn105/uptime_connection_sensor.h:2:10: fatal error: esphome/components/uptime/uptime_seconds_sensor.h: No such file or directory
    2 | #include "esphome/components/uptime/uptime_seconds_sensor.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/aa-mitsubishi-porchemesa/src/esphome/components/cn105/climateControls.cpp.o] Error 1
========================== [FAILED] Took 7.97 seconds ==========================

Thanks

echavet, 2025-02-19:
Could you try ?

`esphome clean YAMLfile`

and also check your repository clone is up to date?

maybe:

```yaml
external_components:
  - source: github://echavet/MitsubishiCN105ESPHome
    refresh: 0s
```

echavet, 2025-02-19:
ok, here it is.
for a reason I don't yet understand you must have this sensor defined in your yaml. I will correct this as soon as I understand the pb.

```yaml
sensor:
  # Uptime sensor.
  - platform: uptime
    name: ${name} Uptime
```

xbmcnut, 2025-02-20:
I've added the above sensor and it nearly compiles but at the end I get this error:
```
Reading CMake configuration...
No dependencies
Linking .pioenvs/cn105/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/cn105/src/main.cpp.o:(.literal._Z5setupv+0xb0): undefined reference to `_ZTVN7esphome6uptime19UptimeSecondsSensorE'
/data/cache/platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: .pioenvs/cn105/src/main.cpp.o:(.literal._Z5setupv+0xb4): undefined reference to `_ZTVN7esphome6uptime19UptimeSecondsSensorE'
collect2: error: ld returned 1 exit status
*** [.pioenvs/cn105/firmware.elf] Error 1
========================== [FAILED] Took 5.32 seconds ==========================
```

echavet, 2025-03-01:
Issue reported to esphome but marked closed "Not planned"
https://github.com/esphome/issues/issues/6782

PR #237 (2025-02-19): Error compiling on esphome 2025.2.0
Fixes #236
commit 2b73a61b (2025-02-19): Error compiling on esphome 2025.2.0
Fixes #236