echavet/MitsubishiCN105ESPHome #245: Constant freezes or restarts

echavet, 2025-03-08:
### Discussed in https://github.com/echavet/MitsubishiCN105ESPHome/discussions/244

<div type='discussions-op-text'>

<sup>Originally posted by **tiagoalves** March  6, 2025</sup>
I've been running this project for two months now and it's great. Thank you! I followed this guide: https://github.com/echavet/MitsubishiCN105ESPHome/discussions/83

Unfortunately, I've been having constant freezes or restarts — every few hours or days. I know it freezes (as opposed to losing network connection) because I have a blinking LED on the ESP32 and it stops blinking (stays on or off) when the ESP32 becomes unavailable. I don't see anything relevant in the logs when it freezes.

My setup:

* 1 outdoor and 4 indoor units, brand new
    * Outdoor unit: MXZ-4F83VF
    * Indoor units: MSZ-LN VG2 (MSZ-LN50VG and MSZ-LN25VG)
* ESP32: AtomS3 Lite x 4
* Cable: Grove to PAP-05V-S 24 AWG. 
* Latest ESPHome: 2025.2.0.

What I tried, to no avail:

* Changed from framework `arduino` to `esp-idf`.
* Bought new PAP-05V-S cables with 22 AWG.
* Tried a different, older ESP32 board (Wemos LoLIN32 Lite) and this one doesn't freeze as often but it restarts every few hours. This leads me to believe that it's not the ATOM S3 Lite that have an issue.

Here's the uptime chart of one of the ESP32s that don't freeze but rather restarts. I'm assuming it restarts because the uptime resets to 0 every few hours.

[image]

Here's my config:

[suite.yaml.txt](https://github.com/user-attachments/files/19099177/suite.yaml.txt)
[base.yaml.txt](https://github.com/user-attachments/files/19099266/base.yaml.txt)

I'm at a loss and would appreciate any help fixing the constant freezes and restarts.
</div>

echavet, 2025-03-08:
Hi @tiagoalves,
do you confirm "freezes" rather than restarts ? Do you mean you have to manually reboot the ESP ?

I see my ESP are also restarting very more often than I thought. Particularly my atomS3.

tiagoalves, 2025-03-08:
Yes, exactly. My Atom S3 Lite _mostly_ freeze and I have to manually reboot them by pressing the physical button on the device. Sometimes, rather rarely, they seem to restart by themselves, at least going by the frequently resetting Uptime values.

echavet, 2025-03-09:
Well I'm sorry for that.
I must find where it comes from. Mines are rebooting regularly but never freeze.
I might take a long time to find out what is causing this. I hate these random situations... 
I'm trying to disable functions like remotetemp... to isolate the pb... 
If you want to help, you can also try to use only basic features and see if it's better. If you find something, let me know.

echavet, 2025-03-10:
Looking closer at my 2 ESPS3 behaviors I can say that:
Actually, only my atomS3 is showing a strange uptime reinitialization. But what is wierd is that it does not seem the reboot (because I don't have any breaking event in the logs).
The other S3 is not an atom. It works very well.

Still digging...

echavet, 2025-03-10:
Could you please just try to setup a minimal version of the firmware?
You can start by just commenting :

```yaml
#captive_portal:
#web_server:
#  port: 80
```

phidauex, 2025-03-10:
For what it's worth, I'm getting normal/stable behavior without resets or restarts, and consistent uptimes, on a Seeed Xiao ESP32-S3 with a maximal configuration including BT proxy and temperature sensors, as well as some older Wemos D1 Pro ESP8266 devices. Neither has any unique wiring to the CN105, voltage regulator circuits, etc. Both are now updated to 2025.2.2. Board type is set to `esp32-s3-devkitc-1` for the Xiao ESP32S3.

echavet, 2025-03-11:
[image]

tiagoalves, 2025-03-11:
Thanks for looking into this. I'm away this week so I can't update my devices and help much until next week.

Since I last updated and rebooted my devices and used a simpler yaml config, I'm seeing much more stability. After a day of regular Uptime resets, it has been stable for 5 days.

[image]

This is my yaml file. I removed the remote temperature sensor and other irrelevant things such as the reboot button and my auto-reboot timer at 6:30 AM.

[base_basic.yaml.txt](https://github.com/user-attachments/files/19184538/base_basic.yaml.txt)

However, my other ESP32 devices that used the previous, full config (from my initial report) have also been quite stable. The Uptime resets we can see are because of the auto-reboot timer at 6:30 AM. So, no constant restarts nor freezes which is quite surprising. :/

[image]

tiagoalves, 2025-03-14:
@phidauex I probably don't need the reboot timer. I just added it when I was trying to find a workaround for the constant freezes, when I still thought it might just be connectivity issues. In the meantime I just left the timer in.

Appreciate the fix, @echavet! Will try out the latest release. 🤞

Just for reference, leaving here my uptime graph for the basic yaml 
config shared in https://github.com/echavet/MitsubishiCN105ESPHome/issues/245#issuecomment-2714362514. This is with a release before the fix submitted in this issue thread. Weird that it was stable for 5 days and then went back to restarting every few minutes or hours.

[image]

tiagoalves, 2025-03-17:
I wanted to follow up with my latest findings:

1. In some of my devices, I haven't had a freeze in a long time, even before reporting this issue here. Thinking better about this, I suspect it coincided with the moment where I upgraded ESPHome from `2024.12.0` to `2025.2.0`. Those freezes probably turned into self-healing restarts. Since upgrading all devices to the latest version, I haven't had a freeze in any device.
2. The restart issue still persists, even after updating to the latest version of this project, after this issue's fix.

 I've tried to debug the restart issue and these are my findings:

* I added `esp_reset_reason_t reason = esp_reset_reason();` to the code and after an unexpected restart, it returns 6 which I think equals to `ESP_RST_TASK_WDT` which is apparently a reset due to task watchdog.
* I was able to log the restart with log level `VERY_VERBOSE`. The log was done through wifi which is not ideal but maybe it helps. The last log message is `4a: Sending status request`. Here are the last few seconds before the restart:

<details>

<summary>Here are the last few seconds before the restart</summary>

```log
[14:31:13][I][CYCLE:052]: 6: Cycle ended in 1.8 seconds (with timeout?:  NO)
[14:31:13][W][component:237]: Component esphome.coroutine took a long time for an operation (166 ms).
[14:31:13][W][component:238]: Components should block for at most 30 ms.
[14:31:13][VV][scheduler:225]: Running interval '' with interval=10000 last_execution=62191880 (now=62201881)
[14:31:13][VV][scheduler:225]: Running interval 'update' with interval=1000 last_execution=62200891 (now=62201895)
[14:31:13][VV][scheduler:225]: Running interval 'update' with interval=1000 last_execution=62201267 (now=62202268)
[14:31:14][VV][scheduler:225]: Running interval 'update' with interval=1000 last_execution=62201891 (now=62202891)
[14:31:14][VV][scheduler:225]: Running interval 'update' with interval=1000 last_execution=62202267 (now=62203267)
[14:31:15][VV][scheduler:225]: Running interval 'update' with interval=1000 last_execution=62202891 (now=62203891)
[14:31:15][VV][scheduler:225]: Running interval 'update' with interval=1000 last_execution=62203267 (now=62204272)
[14:31:16][VV][scheduler:225]: Running interval 'update' with interval=1000 last_execution=62203891 (now=62204894)
[14:31:16][VV][scheduler:225]: Running interval 'update' with interval=1000 last_execution=62204267 (now=62205267)
[14:31:17][V][UPDT_ITVL:306]: triggering infopacket because of update interval tick
[14:31:17][V][CONTROL_WANTED_SETTINGS:307]: hasChanged is false
[14:31:17][D][CN105:308]: sending a request for settings packet (0x02)
[14:31:17][I][CYCLE:039]: 1: Cycle start
[14:31:17][D][CYCLE:311]: 2a: Sending settings request (0x02)
[14:31:17][D][CN105:323]: creating Info packet
[14:31:17][D][CN105:084]: writing packet...
[14:31:17][D][WRITE:170]: FC 42 01 30 10 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7B 
[14:31:17][VV][scheduler:225]: Running interval 'update' with interval=1000 last_execution=62204891 (now=62205897)
[14:31:17][V][Decoder:028]: --> FC [nb: 0]
[14:31:17][V][Decoder:028]: --> 62 [nb: 1]
[14:31:17][V][Decoder:052]: data length toujours pas connu
[14:31:17][V][Decoder:028]: --> 01 [nb: 2]
[14:31:17][V][Decoder:052]: data length toujours pas connu
[14:31:17][V][Decoder:028]: --> 30 [nb: 3]
[14:31:17][V][Decoder:052]: data length toujours pas connu
[14:31:17][V][Decoder:028]: --> 10 [nb: 4]
[14:31:17][V][Header:089]: header matches HEADER
[14:31:17][V][Header:090]: [FC] (62) 01 30 [10]<-- header
[14:31:17][D][Header:091]: command: (62) data length: [10]<-- header
[14:31:17][V][Decoder:028]: --> 02 [nb: 5]
[14:31:17][V][Decoder:028]: --> 00 [nb: 6]
[14:31:17][V][Decoder:028]: --> 00 [nb: 7]
[14:31:17][V][Decoder:028]: --> 01 [nb: 8]
[14:31:17][V][Decoder:028]: --> 01 [nb: 9]
[14:31:17][V][Decoder:028]: --> 09 [nb: 10]
[14:31:17][V][Decoder:028]: --> 01 [nb: 11]
[14:31:17][V][Decoder:028]: --> 00 [nb: 12]
[14:31:17][V][Decoder:028]: --> 00 [nb: 13]
[14:31:17][V][Decoder:028]: --> 00 [nb: 14]
[14:31:17][V][Decoder:028]: --> 83 [nb: 15]
[14:31:17][V][Decoder:028]: --> AC [nb: 16]
[14:31:17][V][Decoder:028]: --> 28 [nb: 17]
[14:31:17][V][Decoder:028]: --> 00 [nb: 18]
[14:31:17][V][Decoder:028]: --> 00 [nb: 19]
[14:31:17][V][Decoder:028]: --> 00 [nb: 20]
[14:31:17][V][Decoder:028]: --> F8 [nb: 21]
[14:31:17][V][CN105:113]: processing data packet...
[14:31:17][D][READ:170]: FC 62 01 30 10 02 00 00 01 01 09 01 00 00 00 83 AC 28 00 00 00 F8 
[14:31:17][V][chkSum:067]: controling chkSum should be: F8 
[14:31:17][V][chkSum:070]: adding FC to 000 --> FC
[14:31:17][V][chkSum:070]: adding 62 to 0FC --> 15E
[14:31:17][V][chkSum:070]: adding 01 to 05E --> 5F
[... 367 lines cut ...]
[14:31:23][V][chkSum:070]: adding 0D to 0A2 --> AF
[14:31:23][V][chkSum:070]: adding 00 to 0AF --> AF
[14:31:23][V][chkSum:070]: adding 98 to 0AF --> 147
[14:31:23][V][chkSum:070]: adding AE to 047 --> F5
[14:31:23][V][chkSum:070]: adding AE to 0F5 --> 1A3
[14:31:23][V][chkSum:070]: adding FE to 0A3 --> 1A1
[14:31:23][V][chkSum:070]: adding 42 to 0A1 --> E3
[14:31:23][V][chkSum:070]: adding 00 to 0E3 --> E3
[14:31:23][V][chkSum:070]: adding 00 to 0E3 --> E3
[14:31:23][V][chkSum:070]: adding 79 to 0E3 --> 15C
[14:31:23][V][chkSum:070]: adding CC to 05C --> 128
[14:31:23][V][chkSum:070]: adding 00 to 028 --> 28
[14:31:23][D][CYCLE:328]: 3b: Receiving room °C response
[14:31:24][D][Decoder:256]: [Room °C: 23.000000]
[14:31:24][D][Decoder:257]: [OAT  °C: 12.000000]
[14:31:24][D][CYCLE:331]: 4a: Sending status request (0x06)
WARNING ac-suite @ 192.168.41.223: Connection error occurred: [Errno 54] Connection reset by peer
INFO Processing unexpected disconnect from ESPHome API for ac-suite @ 192.168.41.223
WARNING Disconnected from API
INFO Successfully connected to ac-suite @ 192.168.41.223 in 0.372s
INFO Successful handshake with ac-suite @ 192.168.41.223 in 0.097s
[14:31:46](...connection is back up...)
```

</details>

ScarletRave, 2025-03-28:
Ive been having the same constantly freezing issue at least once a day often more, I was having to manually restart the esp as it was fully locking up(ducted heat pump meant climbing into the ceiling space).
I am using an esp32-S3.
I tried change to esp-idf frame work but this only made the crashing worse. Often within 5 minutes of a restart it would freeze and not come back up.
I tried removing the web server on esp-idf, which changed nothing. 

When I changed back to audrino and left the web server out it, appears to have resolved the freezing issue. It's been two days without having to manually restart it myself. 

For me it appears that the web server was causing the issue.

echavet, 2025-03-29:
Hi,
Thank you so much for your detailed report! It’s helpful to have all this information, including the logs and your observations.
The freezes you’ve experienced disappearing after disabling features like captive_portal and web_server makes sense to me. I’ve seen similar behavior when enabling too many features that can strain ESPs' resources, leading to instability. I also suspect that some ESPs, in some conditions, might be losing their WiFi connection intermittently, which could explain the uptime counter resetting—something worth keeping an eye on.

From the tests you’ve described and the logs you’ve shared, there seems to be a link to resource overload, whether it’s memory or CPU. That’s likely why reducing the options in your YAML configuration helps minimize the crashes or restarts. The overly verbose logging could also be contributing to the problem—those frequent and detailed outputs might be taxing the system further, especially if the ESP is already close to its limits.

I'm re-opening the issue but I'm quite sure the firmware is not the root cause. But I didn't spend much time on this, so If you think I'm missing something, just tell me...

phidauex, 2025-06-04:
Any updates here? Are we still seeing frequent restart issues, or has the reduction in resources largely addressed the issue? Are there any misbehaving units that we should update documentation to not recommend?

tiagoalves, 2025-06-14:
My devices have not restarted for about 45 days now which curiously coincides more or less with when I stopped heating my house. My ACs have been mostly off since then and I just started cooling my house a few days ago. No restarts since then either.

As for resource usage being a potential reason for the restarts, I do remember checking RAM usage a few months ago and I didn't see memory increasing or peaking before restarts. I didn't chart CPU usage though.

In any case, even with the restarts, this works very well for me (thank you!). Even my automations run smoothly but I guess there's always the chance that an automation runs while the devices are restarting.

Happy to help debug this further if you could provide me with instructions of what to do but it might have to wait for next winter as I might not have restarts until then.

PR #246 (2025-03-11): Constant freezes or restarts
Fixes #245
commit cdf04d5f (2025-03-11): Constant freezes or restarts
Fixes #245
closed 2025-06-17 by echavet