echavet/MitsubishiCN105ESPHome #248: Install on ESP32-C3 no longer works with ESPHome 2025.2.2

KeiranCantilina, 2025-03-12:
I've been having great success using this integration since October of 2024, but yesterday I was suddenly unable to install OTA updates to new devices because the compilation process fails with "fatal error: esphome/components/uptime/sensor/uptime_seconds_sensor.h: No such file or directory".

I've also tried compiling on Windows 10 using ESPhome on Python 3.10 via the command line, and I get the exact same issue.
Here's the log output from ESPHome Builder:

```
INFO ESPHome 2025.2.2
INFO Reading configuration /config/esphome/esphome-web-1f3948.yaml...
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.5.1
INFO Compiling app...
Processing esphome-web-1f3948 (board: lolin_c3_mini; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32.git#51.03.07)
--------------------------------------------------------------------------------
Platform Manager: Installing git+https://github.com/pioarduino/platform-espressif32.git#51.03.07
INFO Installing git+https://github.com/pioarduino/platform-espressif32.git#51.03.07
git version 2.39.5
Cloning into '/data/cache/platformio/cache/tmp/pkg-installing-vp2vhif2'...
Note: switching to 'ec17486ffa451374e108b72be07e9c8fa83338e8'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

Platform Manager: espressif32@51.3.6+sha.ec17486 has been installed!
INFO espressif32@51.3.6+sha.ec17486 has been installed!
Tool Manager: Installing https://github.com/pioarduino/esp-idf/releases/download/v5.1.5/esp-idf-v5.1.5.zip
INFO Installing https://github.com/pioarduino/esp-idf/releases/download/v5.1.5/esp-idf-v5.1.5.zip
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: framework-espidf@3.50105.0 has been installed!
INFO framework-espidf@3.50105.0 has been installed!
Tool Manager: Installing espressif/toolchain-riscv32-esp @ 12.2.0+20230208
INFO Installing espressif/toolchain-riscv32-esp @ 12.2.0+20230208
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: toolchain-riscv32-esp@12.2.0+20230208 has been installed!
INFO toolchain-riscv32-esp@12.2.0+20230208 has been installed!
Tool Manager: Installing espressif/tool-xtensa-esp-elf-gdb @ ~12.1.0
INFO Installing espressif/tool-xtensa-esp-elf-gdb @ ~12.1.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: tool-xtensa-esp-elf-gdb@12.1.0+20221002 has been installed!
INFO tool-xtensa-esp-elf-gdb@12.1.0+20221002 has been installed!
Tool Manager: Installing espressif/tool-riscv32-esp-elf-gdb @ ~12.1.0
INFO Installing espressif/tool-riscv32-esp-elf-gdb @ ~12.1.0
Downloading  [####################################]  100%          
Unpacking  [####################################]  100%          
Tool Manager: tool-riscv32-esp-elf-gdb@12.1.0+20221002 has been installed!
INFO tool-riscv32-esp-elf-gdb@12.1.0+20221002 has been installed!
Tool Manager: Installing https://github.com/pioarduino/esptool/releases/download/v4.8.1.1/esptool.zip
INFO Installing https://github.com/pioarduino/esptool/releases/download/v4.8.1.1/esptool.zip
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-esptoolpy@4.8.1 has been installed!
INFO tool-esptoolpy@4.8.1 has been installed!
Tool Manager: Installing tasmota/tool-mklittlefs @ ^3.2.0
INFO Installing tasmota/tool-mklittlefs @ ^3.2.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
[... 228 lines cut ...]
    kwh_sensor:
      name: Energy Usage
      disabled_by_default: false
    runtime_hours_sensor:
      name: Runtime Hours
      entity_category: diagnostic
      disabled_by_default: false
# Default logging level
logger:
  hardware_uart: UART1 # Uncomment on ESP8266 devices
  level: INFO

# Enable Home Assistant API
api:
  encryption:
    key: "[key here]"

# Allow Over-The-Air updates
ota:
- platform: esphome

wifi:
  ssid: [ssid here]
  password: [password here]
```

phidauex, 2025-03-13:
Try adding the uptime sensor to your yaml (below). Something changed in 2025.2.0 that prevents the uptime sensor class from loading unless the sensor is explicitly added to the YAML. I know Eric is already investigating.

```yaml
sensor:
  # Uptime sensor.
  - platform: uptime
    name: ${name} Uptime
```

Add this to your configuration, clean the build files, then try building again.

KeiranCantilina, 2025-03-13:
Thanks for the super quick response!
Your suggestion did the trick!

closed 2025-03-13 by KeiranCantilina