echavet/MitsubishiCN105ESPHome #253: Cannot Update - uptime_seconds_sensor.h: No such file or directory

el-Presi, 2025-03-14:
Hi guys,

I cannot update none of my ESPs. (Other ESPs without CN105ESPHome work well).
I get this error.

I've checked the DNS alreadym other ESP Updates work fine - Error persists for about a month now (I thought its maybe a temporary thing)

`Compiling .pioenvs/esp-buro-klimaanlage/src/esphome/components/cn105/climateControls.cpp.o
Compiling .pioenvs/esp-buro-klimaanlage/src/esphome/components/cn105/cn105.cpp.o
Compiling .pioenvs/esp-buro-klimaanlage/src/esphome/components/cn105/componentEntries.cpp.o
Compiling .pioenvs/esp-buro-klimaanlage/src/esphome/components/cn105/cycle_management.cpp.o
In file included from src/esphome.h:37,
                 from src/esphome/components/cn105/Globals.h:2,
                 from src/esphome/components/cn105/cn105.h:2,
                 from src/esphome/components/cn105/cn105.cpp:2:
src/esphome/components/cn105/uptime_connection_sensor.h:2:10: fatal error: esphome/components/uptime/sensor/uptime_seconds_sensor.h: No such file or directory
    2 | #include <esphome/components/uptime/sensor/uptime_seconds_sensor.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.`

phidauex, 2025-03-14:
Known issue that came up in esphome 2025.2.0 - see note on the front page for a simple workaround: https://github.com/echavet/MitsubishiCN105ESPHome#mitsubishi-cn105-esphome

Tadimsky, 2025-03-14:
It looks like this particular errors was fixed with this: https://github.com/echavet/MitsubishiCN105ESPHome/issues/236

But you need to not point at a tag as 1.4 does not include this fix.

closed 2025-05-20 by phidauex