echavet/MitsubishiCN105ESPHome #260: Msz-fh25ve

drRobertDev, 2025-03-26:
My Mitsubishi Msz-fh25ve have Purify system is possible integrate in extension? cause sometime it look like is off and to enable need to use remote controller.

Thanks

phidauex, 2025-04-08:
Unfortunately this is an area where we would need to get serial captures from a Kumo adapter to try to identify how the Purify function works and is controlled by the thermostat. Since the API is undocumented, there is a lot of reverse engineering to do... I doubt this will get figured out soon, but I'll mark as a feature enhancement and we'll see if someone with a similar model can do some investigating in the future.

phidauex, 2025-06-04:
Closing this as a duplicate of #282 , work on enabling the purifier can continue there, which would hopefully benefit multiple units.

closed 2025-06-04 by phidauex