echavet/MitsubishiCN105ESPHome #262: High Fan mode for PVA indoor unit mapped as 'Middle'?

joergbattermann, 2025-04-08:
Hey there,

I just gave this project a try for/with our combo of ducted PVA-A36AA7 & PUZ-A36NKA7 units and one thing I noticed is when going through the fan modes (per MHK2 / Kumo Cloud) they map to the following for this project

Selection happens via MHK2/Kumo Cloud > what MitsubishiCN105ESPHome project reports
Auto > AUTO
Low > LOW
Medium > MEDIUM
High > MIDDLE

When selecting (the available) 'HIGH' via the ESPHome/MitsubishiCN105ESPHome integration Kumo cloud indicates a 'Powerful' that you can't select via Kumo Cloud at all... but it's displaying it.

Anyway, that MHK2/Kumo Cloud High > mapping to > MIDDLE per this project is a bit odd to be frank and I was wondering if that's a small bug?

phidauex, 2025-05-06:
Mitsubishi uses a frustratingly large number of ways to describe the fan speeds, and they don't map clearly between models. The ESPHome climate component that we use also has some limitations in the number of different fan speeds available. You can use the "supports" option in the climate config to enable or disable certain fan speeds, so you could turn off any that don't have a clear analog to your unit:

```yaml
  supports:
    mode:
    - AUTO
    - COOL
    - HEAT
    - DRY
    - FAN_ONLY
    fan_mode:
    - AUTO
    - MIDDLE
    - QUIET
    - LOW
    - MEDIUM
    - HIGH
    swing_mode:
    - 'OFF'
    - VERTICAL
    - HORIZONTAL
    - BOTH
```

Unfortunately beyond that I'm not sure that a solution is possible, since the CN105 doesn't report the model information, and the climate component is limited in the names and number of fan modes available.

phidauex, 2025-06-04:
Closing this issue for now, without upstream changes to ESPHome, we don't currently have a way to implement more precise mapping of fan speeds to Mitsubishi's labeling.

closed 2025-06-04 by phidauex