echavet/MitsubishiCN105ESPHome #268: warnings in compile

ideasman69, 2025-04-13:
I've had zero issues compiling the firmware in esphom builder over the weekend, but since last night - i'm now seeing this warning while compiling:

`Compiling .pioenvs/bedroom-ac/src/esphome/components/cn105/hp_readings.cpp.o
src/esphome/components/cn105/heatpumpFunctions.cpp: In member function 'void esphome::CN105Climate::functionsArrived()':
src/esphome/components/cn105/heatpumpFunctions.cpp:48:46: warning: '%i' directive output may be truncated writing between 1 and 11 bytes into a region of size between 0 and 255 [-Wformat-truncation=]
   48 |         snprintf(states, sizeof(states), "%s %i: %i ", states, code, value);
      |                                              ^~
src/esphome/components/cn105/heatpumpFunctions.cpp:48:17: note: 'snprintf' output between 7 and 282 bytes into a destination of size 256
   48 |         snprintf(states, sizeof(states), "%s %i: %i ", states, code, value);
      |         ~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/cn105/heatpumpFunctions.cpp:48:17: warning: 'snprintf' argument 4 overlaps destination object 'states' [-Wrestrict]
src/esphome/components/cn105/heatpumpFunctions.cpp:39:10: note: destination object referenced by 'restrict'-qualified argument 1 was declared here
   39 |     char states[256];
      |          ^~~~~~`

i can see pull request [#249](https://github.com/echavet/MitsubishiCN105ESPHome/pull/249) was merged 7 hours ago.. not sure if its related

avwuff, 2025-04-15:
Yes this is probably caused by my change -- I actually didn't intend for my PR to be merged as-is, it was intended to open a discussion...

avwuff, 2025-04-15:
Fixed in #269

phidauex, 2025-05-06:
@ideasman69 - Since I believe this was fixed, I'm closing this issue for now. If you are still getting compile errors feel free to reopen and add some updated logs.

closed 2025-05-06 by phidauex