echavet/MitsubishiCN105ESPHome #272: Feature request: Implement Fahrenheit/Celsius conversion to match MA Remote Controller

bertique, 2025-04-21:
I'm wondering if you can switch the implementation of the Fahrenheit to Celsius from the calculation to a mapping table similar to https://github.com/gysmo38/mitsubishi2MQTT/pull/272 ?

The benefit would be to avoid unexpected behavior that sometimes occurs to specific rounding/display values eg. on the MA Remote Controller.

ams2990, 2025-05-21:
@dsstewa, I would be happy to submit a PR copying your code from gysmo38/mitsubishi2MQTT/pull/272, but that code is licensed under the LGPL and this project has no license (making it all rights reserved). Are you okay with that?

ams2990, 2025-05-21:
Thanks! Done in https://github.com/ams2990/MitsubishiCN105ESPHome/commit/302fe61f8b1ab55bbb4d50760a2eb40f1103b9bc. Will test it tomorrow.

phidauex, 2025-05-21:
Silly question, but, how are these helper functions actually used? I don't see them referenced anywhere else in the component - are they used by ESPHome internally?

ams2990, 2025-05-21:
No, they don't appear to be. I just rebuilt using my repo as the external_component and observed no change. Thinking about it more, this makes sense -- if we had some built-in units conversion, why would we need `lambda: return (x - 32) * (5.0/9.0);` in the config? I tried changing the functions to return bogus values, and can't see any change from doing that.

I looked more into how this could be accomplished. I think if `platform: cn105` supported `filters` and `unit_of_measurement`, we could do the ESPHome->HA conversion in a lambda in the config. It's not immediately obvious to me where the conversion from F to C would live when making temperature adjustments in HA.

ams2990, 2025-05-21:
@phidauex, digging into this more, where does the C to F translation on data sent from ESP->HA happen? I assumed ESPHome was sending C and HA was translating that into F, but my climate entity in HA doesn't have `unit_of_measurement` set, so I don't see how it could know. Perhaps this is built into the HA portion of ESPHome, rather than the on-device portion? That seems moderately unlikely given https://esphome.io/components/sensor/index.html#example-converting-celsius-to-fahrenheit.

ams2990, 2025-05-22:
Mmm, I'm pretty sure it's https://github.com/home-assistant/core/blob/dev/homeassistant/components/esphome/climate.py#L133C5-L133C55, which can't be overridden. So if we wanted to present the correct temperatures to ESPHome, we would potentially need to map the Celsius values the unit gives us to Celsius values which convert correctly to the Fahrenheit values we want to display in HA? Truly disgusting.

ams2990, 2025-05-23:
It took quite a bit of trial and error, but the above approach worked: https://github.com/ams2990/MitsubishiCN105ESPHome/commit/982fb888459afcc1cca74b63f3a58cdb3ebf6634. Not ready for review yet -- all of the places where it says `if (true)` need to be replaced with an actual configuration option. However, I'm running this on my own system and it's working very well (in conjunction with https://github.com/echavet/MitsubishiCN105ESPHome/issues/292#issuecomment-2903159227). I really appreciate how easy ESPHome makes it to run your own fork.

closed 2025-05-25 by echavet