echavet/MitsubishiCN105ESPHome #277: hvac_action or state always shows idle

fwump38, 2025-04-26:
I have this installed on an ATOM S3 Lite roughly following the steps from [here](https://github.com/echavet/MitsubishiCN105ESPHome/discussions/83)

My indoor unit is a ducted air handler: [PVFY-P36NAMU-E1](https://www.mylinkdrive.com/usa/S_Series/R410A_Systems-4/Indoor_Equipment-8/Multi_Position_Air_Handler-5/PVFY_P36NAMU_E1?product&categoryName=Multi_Position_Air_Handler-5) with Outdoor unit: [MXZ-SM36NAMHZ](https://www.mylinkdrive.com/USA/M_Series/R410A_Systems-3/Outdoor_Equipment/R410A_Outdoor/MXZ_SM36NAMHZ2_U1?product&categoryName=R410A_Outdoor) And my working YAML Config: https://gist.github.com/fwump38/25a954c2ae600c36d2982f0e7e30da91

I have the device connected to Home Assistant as well.

I'm experiencing something similar to #9  where the state according to ESP Home (as well as the `hvac_action` attribute in the `climate.heat_pump` entity in Home Assistant always shows as Idle regardless of what stage, sub mode, or auto sub mode the unit is in. I don't get any other info besides those sensors (ie: compressor frequency, power usage etc).

Since most folks using this appear to have mini splits I suspect my ducted unit reports things a little differently. I assume the same logic that was updated in #9 needs to account for my unit but I don't know C++ well enough (or at all) to know how to make a PR to fix that.

fwump38, 2025-05-14:
FWIW I figured out that my system does not send anything back from `0x06` response to get status (ie: operating status, input power, energy usage, etc)

I ended up making a fork where I hacked together a way to set operating status based on `stage != IDLE` and hvac_action is now reporting correctly.

phidauex, 2025-05-20:
@fwump38 - Have you had a chance to try the new build? There is a new option to `stage_sensor` allowing you to use it as a fallback for determining operating mode for units that don't report it the "normal" way:

```yaml
stage_sensor:
      name: stage sensor
      use_as_operating_fallback: true
```

fwump38, 2025-05-20:
Hi - I just tried it. I saw the new option but wasn't sure where to put it. ESPHome Builder doesn't seem to like that config:

```
climate.cn105: [source /config/esphome/heat-pump.yaml:261]
  platform: cn105
  id: hp
  name: Heat Pump
  icon: mdi:heat-pump
  visual: 
    min_temperature: 17
    max_temperature: 31
    temperature_step: 
      target_temperature: 1
      current_temperature: 0.1
  remote_temperature_timeout: 15min
  update_interval: 3500ms
  debounce_delay: 100ms
  stage_sensor: 
    name: Stage
    entity_category: diagnostic
    disabled_by_default: True
    
    [use_as_operating_fallback] is an invalid option for [stage_sensor]. Please check the indentation.
    use_as_operating_fallback: True
```

Should it be somewhere else?

phidauex, 2025-05-20:
Hmm, you might try specifying refresh: 0s on the external component to make sure it is grabbing the latest copy (otherwise it uses a cached version for some period of time). Otherwise yes, `use_as_operating_fallback` should be under stage_sensor. My configs are validating correctly with the new sensor value (though I haven't tried it in "true" since my units don't have the operating mode issue).

```yaml
external_components:
- source:
    url: https://github.com/echavet/MitsubishiCN105ESPHome.git
    type: git
  refresh: 0s
```

fwump38, 2025-05-20:
I was using the tag but I guess v1.4.1 is before the new option.

I just installed again without the tag and with `refresh: 0s`

It wasn't letting me save the `use_as_operating_fallback` as is so I'm wondering if the downloaded copy of the repo had to be there before ESPHome Builder will accept the new option. After installing again I can no install with `use_as_operating_fallback: true`

Will report back after testing

fwump38, 2025-05-20:
Update: #287 doesn't fix the `hvac_action` in Home Assistant. 

Tested by setting to `heat` and raising the set point way above the current temp.
`stage: low`
`hvac_action: idle`

mgulick, 2025-06-03:
See #273.

My ducted heat pump had the same problem.  The CN105 UART is 5V logic, and the ESP32-S3 is 3.3V (in fact pretty much all ESP8266 and ESP32 devices) are 3.3V logic.   Some systems seem to be more forgiving and will be able to communicate with 3.3V devices, but mine couldn't.  Bottom line, the only fix I had for this issue was to use a level shifter to get back to 5V logic.  You can do this with either a ESP-01 device and the 5V adapter boards (no soldering required) or make a custom board with a level shifter.

fwump38, 2025-06-04:
Awesome! I'll have to give this a try. Might take a bit to get around to this but appreciate the tip.

mgulick, 2025-09-25:
> [@mgulick](https://github.com/mgulick) could you share an example of where the unit does not report the 0x06 value back and one where it does report the value back?

There is an example of it **not** working in https://github.com/echavet/MitsubishiCN105ESPHome/discussions/273#discussioncomment-13109420:

```
...
[01:21:48][D][CYCLE:333]: 4a: Sending status request (0x06)
[01:21:48][D][CN105:323]: creating Info packet
[01:21:48][D][CN105:084]: writing packet...
[01:22:08][W][CN105:009]: Cycle timeout, reseting cycle...
[01:22:08][I][CYCLE:052]: 6: Cycle ended in 21.0 seconds (with timeout?: YES)
```

I don't have any logs handy with it working, as I've turned off the debug messages in the current config, and I don't really want to rebuild esphome right now since there have been some compatibility issues with the recent versions.

It does look like you're receiving the 0x06 messages correctly.  I'm not sure why the second example is different.  Perhaps nothing is printed if it hasn't changed?