echavet/MitsubishiCN105ESPHome #278: PUHZ-SHW230 not working

mynkow, 2025-04-30:
Hello, I am here to report that I was not able to connect to PUHZ-SHW230. I am not sure what the indoor unit is, but it is a FTC_6 board.

```
uart:
  id: ecodan_uart
  baud_rate: 2400
  tx_pin: GPIO1
  rx_pin: GPIO3
  rx_buffer_size: 1024
  parity: EVEN
```

I see the following error:

```
[20:11:07][I][app:100]: ESPHome version 2025.4.1 compiled on Apr 30 2025, 19:57:08
[20:11:09][W][CN105:176]: Heatpump has not replied for 811 s
[20:11:09][I][CN105:177]: We think Heatpump is not connected anymore..
[20:11:09][I][CN105:010]: setupUART() with baudrate  2400 
[20:11:11][W][CN105:176]: Heatpump has not replied for 813 s
[20:11:11][I][CN105:177]: We think Heatpump is not connected anymore..
```

It is however working properly with this one:
https://github.com/tobias-93/esphome-ecodan-heatpump

NOTE: It will be great if there is a list / screenshot of the available data from some units so people could see how it looks like live working.

echavet, 2025-05-06:
thank you for your feedback

itrankolov, 2025-05-09:
For a heatpump I suggest you try this one: 
https://github.com/gekkekoe/esphome-ecodan-hp/