echavet/MitsubishiCN105ESPHome #282: MSZ-LN and iSee + purifier

rouxton, 2025-05-09:
Hi there,

New here, thanks for this excellent project. I was able to quickly replace the native WiFi interface on my MSQ-LN35VGW unit with a Wemos D1 mini picking up dust in one of my drawer. 
BTW, the native interface is not using CN105 but another connector almost identical, but without 5V (just providing 12v). I just use a voltage regulator to get 5v as my CN105 was already connected to my Airzone remote.
That being said, I have some questions/remarks/Issues and maybe help to offer regarding two features.
First one iSee sensor. On my unit, I have a button to activate iSee modes that basically control the way the wide vane will move based on what the sensor see.
With the current implementation, iSee is a read-only binary sensor that says On whenever I use one of the iSee modes. If I switch back (using the remote) to a classic wide-wane mode, the iSee flag stays on. So it's not really useful. Also, it would be great to have a way to activate iSee wide-vane modes using ESPhome. BTW, when I'm using any of the iSee mode, I got the following error in the logs : wideVane reading caution: value 0 not found, returning value at index 0. And from what I've seen in the other topics, there's no defined value 0 hence the error. 
0 means probably "Other" and we probably have another flag to check/configure Isee + wide-vane.

Second question is about the purifier feature that my unit is providing by default. I'd love to be able to control it through HA, I turned it on using the remote controller and got only the same message as before (wideVane reading caution: value 0 not found, returning value at index 0) so that's definitely something that means other but again, where is the real flag ?

And the conclusion : for both questions/remark, what is the best way to help providing the details (in a nutshell, what is the best logging level to configure) ? How to see raw data ? 

Another last question, in the logs, I can see that : 
[14:21:59][D][CYCLE:330]: 3b: Receiving room °C response
[14:21:59][D][Decoder:258]: [Room °C: 20.500000]
[14:21:59][D][Decoder:259]: [OAT  °C: 19.000000]

What does OAT means ?

Thanks

Azelphur, 2025-05-14:
I think MSQ-LN35VGW is a typo (should be MSZ?) I have MSZ-LN60VG2V and MSZ-LN18VG2V with confirmed working 5v, and a few more I'm planning on fitting. They are all MSZ-LNXXVG2V where XX is the kw rating. I'd expect yours to also have working 5v as we have almost exactly the same model. Could well be worth checking pins with multi meter, maybe you have a defective unit, or maybe there has been some change in newer units such that they don't supply 5v any more. Mainly mention it for future readers, if newer units don't have 5v it would definitely be useful to document.

Apart from that, I too would love to get the iSee/purifier going. I have pretty bad allergies and not knowing whether the purifier is on or not is a pain. Happy to provide any information / debugging that aids in getting this going.

rouxton, 2025-05-14:
Definitely a typo, my bad. My point was not about the 5 or 12V, just a side note, sorry for the confusion (I reused the connector used by the default wifi interface which is not CN105 but something else, maybe CN110 which does not provide 5V but only 12v). There is definitely still a CN105, I'm using it for my Aidoo interface and I did not want to buy a CN105 splitter.

phrozen77, 2025-07-19:
Same remark from my side - MSZ-LN18VG2V and ..25.. units working like a charm! The original Wifi interface works off of the PROG interface next to the CN105 connector.

iSee sensor is enabled but off when i turn on the units via HA, even though wide vane is set to INDIRECT - same setting is applied when i activate it via the remote, but it then turns on iSee additionally.

How can i activate the iSee sensor properly via ESPHome?

VBen, 2025-08-12:
@rouxton  Is this enhancement realized with  #370