echavet/MitsubishiCN105ESPHome #283: Horizontal Vane resets

rplushh, 2025-05-12:
Has anyone noticed their horizontal Vane setting reset to " | " after a temperature change or a reboot? It's happened to all 4 of my units. I have a MSZ-GL12NA with a wemos d1 mini flashed.

echavet, 2025-05-13:
> Has anyone noticed their horizontal Vane setting reset to " | " after a temperature change or a reboot? It's happened to all 4 of my units. I have a MSZ-GL12NA with a wemos d1 mini flashed.

Hi @rplushh,
This might be because your units don't support the horizontal vane or maybe they are not compatible with the known protocol to manage the vane. Mine do not support horizontal vane either.
Maybe, if you want to explore deeper, you should activate more verbose logs to see what's happening.

rplushh, 2025-05-13:
My unit does support horizontal vane via the handheld remote. when i change the settings, it confirms the change in verbose

`<html><body>

05:51:59 | [I] | [EVT_SETS:013] | checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
-- | -- | -- | --
05:51:59 | [I] | [CN105:248] | sending wantedSettings..
05:51:59 | [I] | [CN105:611] | widevane setting (extra select component) changed
05:51:59 | [I] | [CYCLE:010] | Defering cycle trigger of   750  ms

</body>
</html>`

rplushh, 2025-05-13:
Also just like to add that only the Horizontal Vane setting reset to | . The Vertical Vane retains it's last position set before a temperature change.

BlairC1, 2025-06-26:
I'm seeing a similar issue with a MSZ-GE60VA. I have a scene defined for heat that has both horizontal and vertical vanes defined, when activated (first time) the horizontal vane will return to the "|" position even though that's a position I never use and at the same time it also resets the horizontal vane to auto which once again is a setting I never use. Interestingly if I activate the heat scene then within seconds activate it again then both vanes will go to the positions defined in the scene. Same applies to snapshots when I leave/return.

BlairC1, 2025-06-30:
I'm having a quick dig and I'm just guessing at this point but I think HA might be sending a swing mode off on changes and then triggering the defaults defined in climatecontrols.cpp....

void CN105Climate::controlSwing() {
    switch (this->swing_mode) {                 //setVaneSetting supports:  AUTO 1 2 3 4 5 and SWING
    case climate::CLIMATE_SWING_OFF:
        this->setVaneSetting("AUTO");
        this->setWideVaneSetting("|");
        break;
    case climate::CLIMATE_SWING_VERTICAL:
        this->setVaneSetting("SWING");
        this->setWideVaneSetting("|");
        break;
    case climate::CLIMATE_SWING_HORIZONTAL:
        this->setVaneSetting("AUTO");
        this->setWideVaneSetting("SWING");
        break;
    case climate::CLIMATE_SWING_BOTH:
        this->setVaneSetting("SWING");
        this->setWideVaneSetting("SWING");
        break;

BlairC1, 2025-06-30:
Thanks, yep will give it try once I get a chance. I have workarounds in place at the moment so haven't gotten around to it.

echavet, 2025-07-01:
Bravo!

echavet, 2025-07-01:
Anyway, we aren't much further along for all that!

because, now you probably can't turn the swing mode off! can you ?

echavet, 2026-09-15:
Quick maintainer ping: #719 (sticky `last_user_vane` / setpoint grace) has been merged since this thread was active.

@rplushh — can you still reproduce the horizontal vane resetting to `|` after a temperature change or reboot on current `main`?
- If **no** → we will close this as fixed by #719.
- If **yes** → please share a short log around the reset (Decoder vane/wideVane + any SET), and we will keep it open.