echavet/MitsubishiCN105ESPHome #292: Can't change temperature step size in HA without also changing the ESP

ams2990, 2025-05-21:
My Home Assistant uses Fahrenheit, and the fact that the unit uses Celsius is making the setup more difficult to use (partially because of #272).

If I set
```
      temperature_step:
        target_temperature: 0.5
```
(or leave it out -- this seems to be the default value), then I get 0.5 °F steps in HA and 0.5 °C steps on the ESP's web interface. Similarly, if I set `target_temperature: 1`, then I get 1 °F steps in HA and 1 °C steps on the web UI. Because 1 °F ~ 0.5 °C, I want to control the precision of these two separately -- 1 °F steps in HA and 0.5 °C steps on the ESP.

(everything I wrote above applies equally to current_temperature as target_temperature).

phidauex, 2025-05-21:
I think this is basically an ESPHome limitation - the temperature_step variable is visual, so it just impacts whatever UI you are currently using. Both ESPHome and Mitsubishi work in C natively, and in the climate component the F conversion happens in the UI, where it also just follows the visual guidance and selects the 0.5F step.

However, the step is visual only - you can still send an arbitrary value to the climate entity using actions.

I just tested a quick workaround, which is that the HACS simple-thermostat card has a `step_size` option allowing you to override the default for that entity. I set my ESP device to an 0.5 step, and the card to a 1.0 step, and I get the desired effect. Unfortunately it looks like the default thermostat card doesn't have an option like this.

As you note, the situation is made worse because of https://github.com/echavet/MitsubishiCN105ESPHome/issues/272 - Mitsubishi uses a lookup table to convert C to F, and the table doesn't round the way the real numbers do, so there are a few F temperature values that are "ambiguous", which can make it hard to determine what setpoint you are actually on. The proposed fix including the clever lookup work from the mitsubishi2MQTT project should help resolve that.

I'll close this issue for now since it is outside the scope of a custom component, but if you have a chance to work on the testing for 272, then continue the discussion there.

closed 2025-05-21 by phidauex

ams2990, 2025-05-23:
I found the fix -- it's in HA. Put this in your configuration.yaml, change the entity id with whatever is appropriate:
```
homeassistant:
  customize:
    climate.second_floor:
      target_temp_step: 1
```
Restart, and the thermostat will have the correct step size.