echavet/MitsubishiCN105ESPHome #343: Unsupported WideVane code

jlpouffier, 2025-07-07:
Hello!

I have a MSZ-AY35VGK unit
My Horizontal vane always gets reset to `←←` because of this error

```
[13:02:43][W][lookup:206]: wideVane reading caution: value 130 not found, returning value at index 0
```

Looking at the code [here](https://github.com/echavet/MitsubishiCN105ESPHome/blob/a2ec2fb651a9481889dbe7b9d783038cd6b7080b/components/cn105/Globals.h#L81), it seems that 130 (or I guess `0x82`) is not supported by the library.

I do not seem to have anything special on the horizontal swing mode on my unit! 
MelCloud is giving me values that should all be already mapped on this project:

[image]

Happy to help you debug more! I'm available!

JLo

jlpouffier, 2025-07-07:
Wait...
I'm stupid:
```
static const char* WIDEVANE_MAP[11] = { "←←", "←", "|", "→", "→→", "←→", "SWING", "INDIRECT", "DIRECT", "EVEN", "OFF" };

```

In MelCloud I see
- 5 Directions
  - `←←`
  - `←`
  - `|`
  - `→`
  - `→→`
- 2 modes
  - `SWING`
  -  ⁉️

But I also see `AUTO` ... as simple as that... Maybe 130 (or I guess `0x82`) is simply `AUTO`

jlpouffier, 2025-07-07:
Currently trying to add `AUTO` on [my fork](https://github.com/jlpouffier/MitsubishiCN105ESPHome) and It seems stable, no more warnings.
```
[15:04:08][I][CYCLE:039]: 1: Cycle start
[15:04:09][I][CYCLE:052]: 6: Cycle ended in 1.0 seconds (with timeout?:  NO)
[15:04:11][I][CYCLE:039]: 1: Cycle start
[15:04:12][I][CYCLE:052]: 6: Cycle ended in 1.0 seconds (with timeout?:  NO)
[15:04:14][I][CYCLE:039]: 1: Cycle start
[15:04:15][I][CYCLE:052]: 6: Cycle ended in 1.0 seconds (with timeout?:  NO)
[15:04:17][I][CYCLE:039]: 1: Cycle start
[15:04:18][I][CYCLE:052]: 6: Cycle ended in 1.0 seconds (with timeout?:  NO)
[15:04:20][I][CYCLE:039]: 1: Cycle start
[15:04:21][I][CYCLE:052]: 6: Cycle ended in 1.0 seconds (with timeout?:  NO)
[15:04:23][I][CYCLE:039]: 1: Cycle start
[15:04:24][I][CYCLE:052]: 6: Cycle ended in 1.0 seconds (with timeout?:  NO)
[15:04:26][I][CYCLE:039]: 1: Cycle start
[15:04:27][I][CYCLE:052]: 6: Cycle ended in 1.0 seconds (with timeout?:  NO)
[15:04:29][I][CYCLE:039]: 1: Cycle start
[15:04:30][I][CYCLE:052]: 6: Cycle ended in 1.0 seconds (with timeout?:  NO)
```

I'm going to shake my unit a bit, try to change all wide vane settings with the remote and see how it reacts on Home Assistant.

But it seems promising.

I'm going to self-assign that issue to me for now - Pretty sure I can PR that one

echavet, 2025-07-07:
Hey
Could you try with last version

jlpouffier, 2025-07-07:
Re-flashed with the latest version of this repo.
Warning still here ;) 

```
[15:26:03][I][CYCLE:039]: 1: Cycle start
[15:26:03][W][lookup:206]: wideVane reading (full byte) caution: value 130 not found, returning value at index 0
[15:26:04][I][CYCLE:052]: 6: Cycle ended in 1.0 seconds (with timeout?:  NO)
[15:26:06][I][CYCLE:039]: 1: Cycle start
[15:26:06][W][lookup:206]: wideVane reading (full byte) caution: value 130 not found, returning value at index 0
[15:26:07][I][CYCLE:052]: 6: Cycle ended in 1.0 seconds (with timeout?:  NO)
[15:26:09][I][CYCLE:039]: 1: Cycle start
[15:26:09][W][lookup:206]: wideVane reading (full byte) caution: value 130 not found, returning value at index 0
[15:26:10][I][CYCLE:052]: 6: Cycle ended in 1.0 seconds (with timeout?:  NO)
```