echavet/MitsubishiCN105ESPHome #346: Voltage levels (3.3 V vs 5 V) and potential ESP32 damage risk when connecting to CN105

davcolme, 2025-07-11:
Hi,

I'm reviewing the project and planning to use it with an **ESP32-C3 Supermini**, powered directly from the CN105 port (5 V). Everything looks great, but I have a key question about **UART logic level compatibility** and the potential risk to the ESP32.

As far as I understand, the CN105 bus operates at **5 V logic**, and in many implementations (including this project), there are **pull-up resistors to 5 V** on both TX and RX lines.

However, the ESP32-C3 uses 3.3 V I/O, and according to its datasheet, the **absolute maximum voltage** on a GPIO pin is **3.6 V**. Connecting a pin directly to a 5 V line (especially TX, where the ESP32 acts as an output) may result in overcurrent through the internal clamping diodes, potentially damaging the pin over time.

Other sources (such as the blog [nicegear.nz](https://nicegear.nz/blog/hacking-a-mitsubishi-heat-pump-air-conditioner/) and [OpenEnergyMonitor forums](https://community.openenergymonitor.org/t/monitoring-controlling-ecodan-via-cn105/24981)) mention that the CN105 bus is **open-collector**, and that **pulling up to 3.3 V** is enough for reliable communication when using a 3.3 V microcontroller.

elmoret, 2025-07-13:
ESP32s are 5v tolerant, I've used them that way for years. https://www.letscontrolit.com/forum/viewtopic.php?t=8845

davcolme, 2025-07-13:
Thank you very much for your reply and support.

Just to clarify, which connection method would you recommend?
* Connecting the CN105 port directly to the ESP32 pins?
* Should we add pull-up resistors to 3.3 V or 5 V?
* Would it be advisable to place small series resistors to help limit potential damage or signal noise?

Thanks again for your help!