echavet/MitsubishiCN105ESPHome #355: setupUART() with baudrate  2400

DIG200, 2025-07-16:
I’ve been using this integration for at least a year with my two Mitsubishi Electric units, and everything has worked perfectly until now. However, after updating to ESPHome 2025.7.0, I’m getting the following error on both units (I’m using an ESP32-S3 for each):

```
[21:15:50][I][CN105:033]: Trying to connect again...
[21:16:00][W][CN105:176]: Heatpump has not replied for 40 s
[21:16:00][I][CN105:177]: We think Heatpump is not connected anymore..
[21:16:00][I][CN105:008]: setupUART() with baudrate  2400 
[21:16:04][W][CN105:176]: Heatpump has not replied for 44 s
[21:16:04][I][CN105:177]: We think Heatpump is not connected anymore..
[21:16:04][I][CN105:008]: setupUART() with baudrate  2400 
[21:16:08][W][CN105:176]: Heatpump has not replied for 48 s
[21:16:08][I][CN105:177]: We think Heatpump is not connected anymore..
[21:16:08][I][CN105:008]: setupUART() with baudrate  2400 
```

This error repeats continuously in a loop and never recovers.

My uart configuration is as follows:

```
uart:
  id: HP_UART
  baud_rate: 2400
  tx_pin: GPIO17
  rx_pin: GPIO16
```

Everything was working perfectly before the 2025.7.0 update.

DIG200, 2025-07-16:
Thank you for your reply!

I have been using GPIO17 (TX) and GPIO16 (RX) with my ESP32-S3 board for over a year, and it has always worked fine until ESPHome 2025.7.0. I didn’t change anything in my hardware or YAML config.

notentirelysure85, 2025-07-17:
No problem. Unsure if that's the cause of your issue. I have just updated one of my units to ESPHome 2025-07 and it's working with no using pins 17 and 18. Could be worth a try if you haven't soldered the pins. Mine are directly soldered so I can't easily test with your configuration :)

DIG200, 2025-07-18:
Update 2: After switching to ESP-IDF, everything works — but only until I reboot the ESP32-S3. After a normal reboot (e.g., via Home Assistant or power cycle), the same CN105 connection issue comes back.

I have to recompile and upload the firmware again to make it work.
So it seems like something fails during the initialization of UART/CN105 after a reboot, but not after a full flash.

avanh86, 2025-07-21:
I had the exact same problem is DIG200.  
I have changed the framework to ESP-IDF and it works great again.

klim123123, 2025-07-23:
I had exactly the same problem with SEZ-KD. Changed from Arduino to ESP-IDF and it's works now.
ESPHome 2025.7.2

lits101, 2025-07-26:
I've started having this problem with my M5 Atom S3's today, completely out of nowhere. I've been running 2025.7.x for a week, then today two of my four devices stopped working with the above error. Switching from arduino to esp-idf resolved it in both cases.

drRobertDev, 2025-08-22:
i start to have same issue to all my ac (3)

INFO ESPHome 2025.8.0
INFO Reading configuration /config/ac-olimpo.yaml...

[09:10:25][I][app:200]: ESPHome version 2025.8.0 compiled on Aug 21 2025, 10:05:36
[09:10:27][W][CN105:190]: Heatpump has not replied for 2999 s
[09:10:27][I][CN105:191]: We think Heatpump is not connected anymore..
[09:10:27][I][CN105:008]: setupUART() with baudrate  2400 
[09:10:31][W][CN105:190]: Heatpump has not replied for 3003 s
[09:10:31][I][CN105:191]: We think Heatpump is not connected anymore..
[09:10:31][I][CN105:008]: setupUART() with baudrate  2400 
[09:10:35][W][CN105:190]: Heatpump has not replied for 3007 s
[09:10:35][I][CN105:191]: We think Heatpump is not connected anymore..
[09:10:35][I][CN105:008]: setupUART() with baudrate  2400 
[09:10:39][W][CN105:190]: Heatpump has not replied for 3011 s
[09:10:39][I][CN105:191]: We think Heatpump is not connected anymore..

esphome:
  name: ac-olimpo
  friendly_name: ac-olimpo

esp32:
  board: nodemcu-32s
  framework:
    type: esp-idf

uart:
  id: HP_UART
  baud_rate: 2400
  tx_pin: GPIO17
  rx_pin: GPIO16

# External component reference
external_components:
  - source: github://echavet/MitsubishiCN105ESPHome

sensor:
  - platform: uptime
    name: Uptime
	
climate:
  - platform: cn105
    id: hp
    name: "Ac Olimpo"
    icon: mdi:heat-pump
    visual:
      min_temperature: 15
      max_temperature: 32
      temperature_step:
        target_temperature: 1
        current_temperature: 0.5
    # Timeout and communication settings
    remote_temperature_timeout: 30min
    update_interval: 4s
    debounce_delay : 100ms
    # Various optional sensors, not all sensors are supported by all heatpumps
    compressor_frequency_sensor:
      name: Compressor Frequency
      entity_category: diagnostic
      disabled_by_default: false
    outside_air_temperature_sensor:
      name: Outside Air Temp
      disabled_by_default: false
    vertical_vane_select:
      name: Vertical Vane
      disabled_by_default: false
    stage_sensor:
      name: Stage
      entity_category: diagnostic
      disabled_by_default: false
    sub_mode_sensor:
      name: Sub Mode
      entity_category: diagnostic
      disabled_by_default: false
    auto_sub_mode_sensor:
      name: Auto Sub Mode
      entity_category: diagnostic
      disabled_by_default: false

logger:
#  hardware_uart: UART1 # Uncomment on ESP8266 devices
  level: INFO
  logs:
    EVT_SETS : INFO
    WIFI : INFO
    MQTT : INFO
    WRITE_SETTINGS : INFO
    SETTINGS : INFO
    STATUS : INFO
    CN105Climate: WARN
    CN105: INFO
    climate: WARN
    sensor: WARN
    chkSum : INFO
    WRITE : WARN
    READ : WARN
    Header: INFO
    Decoder : INFO
    CONTROL_WANTED_SETTINGS: INFO

jimmy-zhening-luo, 2025-08-24:
Oh it could also be some unknown retarded AC-side, non-ephemeral byte like MYSTERIOUSLY_STOP_WORKING_FOR_ESPHOME_BUT_NOT_KUMO which would survive a 2-hour power down of my entire house, I'll never know bc I've read the entire 200 page installer manual and it doesn't seem that the control board has a concept of a hard reset (the remote does, but that's only for the remote). Even if it did, I wouldn't trust it since I learned from the manual that the way to toggle certain behavior is to short pins like we are living in the year 1986 and we ordered this $10,000 system from PC Hobbyist magazine.

jimmy-zhening-luo, 2025-08-24:
Or triternatively, that the Kumo adapter and ESPHome device have subtly different behavior which causes the former to still complete the loop while the latter does not in the case that the board has indeed experienced some sort of permanent state change.

echavet, 2025-08-24:
Thanks @Aetf for feedback 

So:
•	Users are experiencing UART communication issues with MitsubishiCN105ESPHome integration after updating to ESPHome 2025.7.0 or later.
•	Initial solutions involved switching the framework from Arduino to ESP-IDF and flashing the firmware twice to update both OTA partitions.
•	@Aetf tested that reverting to an older ESPHome version (e.g., 2025.6) immediately resolves the issue, indicating it might be an ESPHome-specific problem rather than a hardware or configuration error.

xMase, 2025-08-25:
It is a known regression of esp-idf ([esphome thread](https://github.com/esphome/esphome/issues/10338#issuecomment-3209070095)).
```
esp32:
  framework:
    type: esp-idf
    version: 5.4.1
```

jimmy-zhening-luo, 2025-08-28:
First of all thanks @Aetf for cracking this. You damn near saved my life, and that's not an exaggeration. Second of all, can anybody once they have time teach me why they think this issue would only surface on one device out of 3 identical devices when that one device is manually re-flashed, and that every subsequent newly flashed device would also suffer the same failure? Given I clean-built the other two devices as well but never manually reflashed them, and they work fine.

It's important to me to understand exactly what the delta is between clean-flash and clean-build-update.

Thank you.

jimmy-zhening-luo, 2025-08-28:
> It is a known regression of esp-idf ([esphome thread](https://github.com/esphome/esphome/issues/10338#issuecomment-3209070095)).
> 
> Simple work-around:
> 
> ```
> esp32:
>   framework:
>     type: esp-idf
>     version: 5.4.1
> ```

I'm still horked after this workaround. Regressing platform to 2025.7.4 also gives me build errors. Uninstalling and reinstalling the add_on would put me back to 2025.8 which may continue to break me, not sure. Will report back. In short, I cannot verify that this is a simple workaround.

jimmy-zhening-luo, 2025-08-28:
> Can also confirm 2025.6 is working. (ESP32-S3-DevKitC) ... but strangely enough I only came across this while updating my devices 2 out of 4 successfully updated but the others got stuck lacking HP communication. Same ESP32 hardware, same indoor units.

I will try 2025.6 now. There's a lot of misinformation or partial information across various linked bugs. What's clear is that I am still broken for any new devices flashed on 2025.8, 2025.7.5, and 2025.7.4, even with 5.4.1.

jimmy-zhening-luo, 2025-08-28:
> It is a known regression of esp-idf ([esphome thread](https://github.com/esphome/esphome/issues/10338#issuecomment-3209070095)).
> 
> Simple work-around:
> 
> ```
> esp32:
>   framework:
>     type: esp-idf
>     version: 5.4.1
> ```

Yeah I am just going to go ahead and say that workaround is incorrect. I believe that reverting to 2025.6 and somehow doing a full-clean to make compilation possible may be the only actual workaround.

Aetf, 2025-08-28:
Strangely enough, reverting back to esp-idf 5.4.1 actually does work for me 🤷 

I have 2 generic ESP-WROOM-32 flashed with ESPHome 2025.8.1 with esp-idf 5.4.1, and they connect happily to cn105...

xMase, 2025-08-29:
> I can confirm that reverting to 2025.6.3, which itself pulls esp-idf 5.3.2, solved my issue. Reverting to 5.4.1 did _not_ solve my issue.

try to factory reset the esp32 an than, reflash the bin into it (in short skip the ota and install it directly).

lits101, 2025-10-19:
For those still having this issue, the recommendation so far from the esphome Discord has been to try adding this to the config:

```
esphome:
  on_boot:
    - priority: 1001
      then:
        - lambda: |-
            gpio_reset_pin((gpio_num_t)01);
            gpio_reset_pin((gpio_num_t)02);
```
I've applied this to one of my Atom S3 Lite devices, and will test for a week or so.

GioMez, 2025-10-24:
In ESPHome 2025.10.3, [PR #9248](https://github.com/esphome/esphome/pull/9248) has been merged, addressing some UART related issues.
It was originally intended to make all UART components respect rx pin `FLAG_PULLUP` and `FLAG_PULLDOWN`, but since it now calls the pin’s `setup()` function, it also resets the pins properly, so the manual `on_boot` reset workaround is no longer needed.

This has been confirmed to work by several users, although depending on your specific board, you might still need to explicitly set the `pullup` or `input` mode in your YAML configuration.

Confirmed working on my setup (ESP32-C6, ESPHome 2025.10.3, ESP-IDF 5.4.2) with the `on_boot` workaround removed.

lits101, 2025-10-25:
> In ESPHome 2025.10.3, [PR #9248](https://github.com/esphome/esphome/pull/9248) has been merged, addressing some UART related issues. It was originally intended to make all UART components respect rx [...]

I tried this on my setup, and I'm getting the "can't connect to heat pump" error.

GioMez, 2025-10-25:
> I tried this on my setup, and I'm getting the "can't connect to heat pump" error.

Could you confirm your board, ESP-IDF version, and whether `pullup` and `input` are set as in the PR example?

echavet, 2025-10-26:
Connexion issue should be fixed now with a patch in the code.
Hope it will works for all users.