echavet/MitsubishiCN105ESPHome #356: SEZ-KD60 not working

mgarciadelojo, 2025-07-17:
Hi, 

First of all thank you for this awesome project. After seeing it I decided to try it on with a spare ESP32 S3 dev kit I had laying around at home. I have a SEZ-KD60 ceiling aircon unit (probably 2010) that has a CN105 port available, so I decided to give this a go.

I tested the following code, but for some reason the pins RX and TX are not able to read/write anything to the UART:

```
substitutions:
  name: bedrooms-ac
  friendly_name: Bedrooms AC
  remote_temp_sensor: sensor.bedrooms_temperature

esphome:
  name: ${name}
  friendly_name: ${friendly_name}

# For ESP32 Devices
esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf

uart:
  id: HP_UART
  baud_rate: 2400
  tx_pin: GPIO17
  rx_pin: GPIO18

external_components:
  - source: github://echavet/MitsubishiCN105ESPHome

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${friendly_name} ESP"
    password: !secret ap_password

captive_portal:

# Enable logging
logger:
  # level: INFO
  # logs:
  #   EVT_SETS : INFO
  #   WIFI : INFO
  #   MQTT : INFO
  #   WRITE_SETTINGS : INFO
  #   SETTINGS : INFO
  #   STATUS : INFO
  #   CN105Climate: WARN
  #   CN105: INFO
  #   climate: WARN
  #   sensor: WARN
  #   chkSum : INFO
  #   WRITE : WARN
  #   READ : WARN
  #   Header: INFO
  #   Decoder : INFO
  #   CONTROL_WANTED_SETTINGS: INFO
 level: DEBUG
 logs:
   EVT_SETS : DEBUG
   WIFI : INFO
   MQTT : INFO
   WRITE_SETTINGS : DEBUG
   SETTINGS : DEBUG
   STATUS : INFO
   CN105Climate: WARN
   CN105: DEBUG
   climate: WARN
   sensor: WARN
   chkSum : INFO
   WRITE : WARN
   READ : WARN
   Header: INFO
   Decoder : DEBUG
   CONTROL_WANTED_SETTINGS: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: !secret ha_bedrooms_ac_api_encryption

ota:
  - platform: esphome
    password: !secret ota_password

# Enable Web server.
web_server:
  port: 80

# Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time

# Text sensors with general information.
text_sensor:
  - platform: version
    name: ESPHome Version
  - platform: wifi_info
    ip_address:
      name: IP
    ssid:
      name: SSID
    bssid:
      name: BSSID

# Sensors with general information.
sensor:
  - platform: uptime
    name: Uptime
  - platform: wifi_signal
    name: WiFi Signal
    update_interval: 120s
  - platform: homeassistant
    name: "Remote Temperature Sensor"
    entity_id: ${remote_temp_sensor}
    internal: false
    disabled_by_default: true
    device_class: temperature
    state_class: measurement
    unit_of_measurement: "°C"
    filters:
    # Uncomment the lambda line to convert F to C on incoming temperature
    #  - lambda: return (x - 32) * (5.0/9.0);
      - clamp: # Limits values to range accepted by Mitsubishi units
          min_value: 1
          max_value: 40
          ignore_out_of_range: true
      - throttle: 30s
    on_value:
      then:
        - logger.log:
            level: INFO
            format: "Remote temperature received from HA: %.1f C"
            args: [ 'x' ]
        - lambda: 'id(bedrooms_ac).set_remote_temperature(x);'

  # Debugging purpose sensors
  - platform: template
    name: "dg_uart_connected"
    entity_category: DIAGNOSTIC
    lambda: |-
      return (bool) id(bedrooms_ac).isUARTConnected_;
    update_interval: 30s
  - platform: template
    name: "dg_complete_cycles"
    entity_category: DIAGNOSTIC
    accuracy_decimals: 0
    lambda: |-
      return (unsigned long) id(bedrooms_ac).nbCompleteCycles_;
    update_interval: 60s
  - platform: template
    name: "dg_total_cycles"
    accuracy_decimals: 0
    entity_category: DIAGNOSTIC
    lambda: |-
      return (unsigned long) id(bedrooms_ac).nbCycles_;
    update_interval: 60s
  - platform: template
    name: "dg_nb_hp_connections"
    accuracy_decimals: 0
    entity_category: DIAGNOSTIC
    lambda: |-
      return (unsigned int) id(bedrooms_ac).nbHeatpumpConnections_;
    update_interval: 60s
  - platform: template
    name: "dg_complete_cycles_percent"
    unit_of_measurement: "%"
    accuracy_decimals: 1
    entity_category: DIAGNOSTIC
    lambda: |-
      unsigned long nbCompleteCycles = id(bedrooms_ac).nbCompleteCycles_;
      unsigned long nbCycles = id(bedrooms_ac).nbCycles_;
      if (nbCycles == 0) {
        return 0.0;
      }
      return (float) nbCompleteCycles / nbCycles * 100.0;
    update_interval: 60s

# Create a button to restart the unit from HomeAssistant. Rarely needed, but can be handy.
button:
  - platform: restart
    name: "Restart ${friendly_name}"

# Creates the sensor used to receive the remote temperature from Home Assistant
# Uses sensor selected in substitutions area at top of config
# Customize the filters to your application:
#   Uncomment the first line to convert F to C when remote temps are sent
#   If you have a fast or noisy sensor, consider some of the other filter
#   options such as throttle_average.
climate:
  - platform: cn105
    id: bedrooms_ac
    name: "${friendly_name}"
    icon: mdi:heat-pump
    visual:
      min_temperature: 18 # Adjust to your unit's min temp. SmartSet units can go to 10C for heating
      max_temperature: 28
      temperature_step:
        target_temperature: 1
        current_temperature: 0.5
    # Fahrenheit compatibility mode - uses Mitsubishi's "custom" unit conversions, set to
    # "true" for better support of Fahrenheit units in HomeAssistant
    fahrenheit_compatibility: false
    # Timeout and communication settings
    remote_temperature_timeout: 30min
    update_interval: 2s
    debounce_delay : 100ms
    # Various optional sensors, not all sensors are supported by all heatpumps
    compressor_frequency_sensor:
      name: Compressor Frequency
      entity_category: diagnostic
      disabled_by_default: true
    outside_air_temperature_sensor:
      name: Outside Air Temp
      disabled_by_default: true
    vertical_vane_select:
      name: Vertical Vane
      disabled_by_default: false
    horizontal_vane_select:
      name: Horizontal Vane
      disabled_by_default: true
    isee_sensor:
      name: ISEE Sensor
      disabled_by_default: true
    stage_sensor:
      name: Stage
      entity_category: diagnostic
      disabled_by_default: true
    sub_mode_sensor:
      name: Sub Mode
      entity_category: diagnostic
      disabled_by_default: true
    auto_sub_mode_sensor:
      name: Auto Sub Mode
      entity_category: diagnostic
      disabled_by_default: true
    input_power_sensor:
      name: Input Power
      disabled_by_default: true
    kwh_sensor:
      name: Energy Usage
      disabled_by_default: true
    runtime_hours_sensor:
      name: Runtime Hours
      entity_category: diagnostic
      disabled_by_default: true
```

I can see the Climate unit, can change the modes, temperature and so on, however, the AC does not react upon those requests. What I can see in the logs is:

```
[01:11:24][I][EVT_SETS:019]: checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
[01:11:24][W][CN105:176]: Heatpump has not replied for 497 s
[01:11:24][I][CN105:177]: We think Heatpump is not connected anymore..
[01:11:24][D][CN105:156]: reconnectUART()
[01:11:24][D][CN105:145]: disconnectUART()
[01:11:24][I][CN105:008]: setupUART() with baudrate  2400 
[01:11:24][D][CN105:123]: UART est configuré en SERIAL_8E1
[01:11:24][D][CN105:018]: Envoi du packet de connexion...
[01:11:24][D][CN105:084]: writing packet...
[01:11:24][I][EVT_SETS:019]: checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
```

I tried switching pin 18 and 17 a couple of times, with no luck. At first I thought that maybe the soldering wasn't right (although I tried this board before), and decided to try a pin loopback to see if connections are good, and it worked perfectly. Soldering is fine.

When I connect the board to the AC Unit, it does not work until I connect the ground, which I think is good and necessary. I can hear a clicking sound in the unit when it is previously working. I suspect that means it is detected. The LED lights on the board lit up immediately after connecting all the cables.

Some guidance on what I can try next would be appreciated. 

Thank you 🙌

glarizza, 2025-07-22:
Ahh yeah, here's someone with a similar issue: https://github.com/echavet/MitsubishiCN105ESPHome/issues/355#issuecomment-3089916651

mgarciadelojo, 2025-07-28:
Hi @glarizza , sorry for my slow reply on this.

I have tried again, and it doesn't seem to work after flashing the board twice. Here the logs I am getting:

```
INFO ESPHome 2025.6.3
INFO Reading configuration /config/bedrooms-ac.yaml...
INFO Detected timezone 'Europe/Madrid'
INFO Starting log output from [redacted] using esphome API
INFO Successfully resolved bedrooms-ac @ [redacted] in 0.000s
INFO Successfully connected to bedrooms-ac @ [redacted] in 0.034s
INFO Successful handshake with bedrooms-ac @ [redacted] in 0.097s
[10:53:18][I][app:137]: ESPHome version 2025.6.3 compiled on Jul 23 2025, 01:19:13
[10:53:18][C][wifi:613]: WiFi:
[10:53:18][C][wifi:434]:   Local MAC: [redacted]
[10:53:18][C][wifi:439]:   SSID: [redacted]
[10:53:18][C][wifi:442]:   IP Address: [redacted]
[10:53:18][C][wifi:446]:   BSSID: [redacted]
[10:53:18][C][wifi:446]:   Hostname: 'bedrooms-ac'
[10:53:18][C][wifi:446]:   Signal strength: -57 dB ▂▄▆█
[10:53:18][C][wifi:455]:   Channel: 8
[10:53:18][C][wifi:455]:   Subnet: 255.255.255.0
[10:53:18][C][wifi:455]:   Gateway: [redacted]
[10:53:18][C][wifi:455]:   DNS1: 1.1.1.1
[10:53:18][C][wifi:455]:   DNS2: 1.0.0.1
[10:53:18][C][logger:211]: Logger:
[10:53:18][C][logger:211]:   Max Level: DEBUG
[10:53:18][C][logger:211]:   Initial Level: DEBUG
[10:53:18][C][logger:217]:   Log Baud Rate: 115200
[10:53:18][C][logger:217]:   Hardware UART: USB_SERIAL_JTAG
[10:53:18][C][logger:224]:   Task Log Buffer Size: 768
[10:53:18][C][logger:229]:   Level for 'CN105': DEBUG
[10:53:18][C][logger:229]:   Level for 'CN105Climate': WARN
[10:53:18][C][logger:229]:   Level for 'CONTROL_WANTED_SETTINGS': DEBUG
[10:53:18][C][logger:229]:   Level for 'Decoder': DEBUG
[10:53:18][C][logger:229]:   Level for 'EVT_SETS': DEBUG
[10:53:18][C][logger:229]:   Level for 'Header': INFO
[10:53:18][C][logger:229]:   Level for 'MQTT': INFO
[10:53:18][C][logger:229]:   Level for 'READ': WARN
[10:53:18][C][logger:229]:   Level for 'SETTINGS': DEBUG
[10:53:18][C][logger:229]:   Level for 'STATUS': INFO
[10:53:18][C][logger:229]:   Level for 'WIFI': INFO
[10:53:18][C][logger:229]:   Level for 'WRITE': WARN
[10:53:18][C][logger:229]:   Level for 'WRITE_SETTINGS': DEBUG
[10:53:18][C][logger:229]:   Level for 'chkSum': INFO
[10:53:18][C][logger:229]:   Level for 'climate': WARN
[10:53:18][C][logger:229]:   Level for 'sensor': WARN
[10:53:18][C][uart.idf:159]: UART Bus 0:
[10:53:18][C][uart.idf:160]:   TX Pin: GPIO17
[10:53:18][C][uart.idf:161]:   RX Pin: GPIO18
[10:53:18][C][uart.idf:163]:   RX Buffer Size: 256
[10:53:18][C][uart.idf:165]:   Baud Rate: 2400 baud
[10:53:18][C][uart.idf:165]:   Data Bits: 8
[10:53:18][C][uart.idf:165]:   Parity: EVEN
[10:53:18][C][uart.idf:165]:   Stop bits: 1
[10:53:18][C][uptime.sensor:033]: Uptime Sensor 'Uptime'
[10:53:18][C][uptime.sensor:033]:   State Class: 'total_increasing'
[10:53:18][C][uptime.sensor:033]:   Unit of Measurement: 's'
[10:53:18][C][uptime.sensor:033]:   Accuracy Decimals: 0
[10:53:18][C][uptime.sensor:033]:   Device Class: 'duration'
[10:53:18][C][uptime.sensor:033]:   Icon: 'mdi:timer-outline'
[10:53:18][C][uptime.sensor:034]:   Type: Seconds
[10:53:18][C][template.sensor:022]: Template Sensor 'dg_uart_connected'
[10:53:18][C][template.sensor:022]:   State Class: ''
[10:53:18][C][template.sensor:022]:   Unit of Measurement: ''
[10:53:18][C][template.sensor:022]:   Accuracy Decimals: 1
[10:53:18][C][template.sensor:023]:   Update Interval: 30.0s
[10:53:18][C][template.sensor:022]: Template Sensor 'dg_complete_cycles'
[10:53:18][C][template.sensor:022]:   State Class: ''
[10:53:18][C][template.sensor:022]:   Unit of Measurement: ''
[... 337 lines cut ...]
[10:55:48][I][EVT_SETS:019]: checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
[10:55:48][I][EVT_SETS:019]: checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
[10:55:48][I][EVT_SETS:019]: checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
[10:55:48][I][EVT_SETS:019]: checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
[10:55:48][I][EVT_SETS:019]: checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
[10:55:48][I][EVT_SETS:019]: checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
[10:55:48][I][EVT_SETS:019]: checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
[10:55:49][I][EVT_SETS:019]: checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
[10:55:49][I][EVT_SETS:019]: checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
[10:55:49][I][EVT_SETS:019]: checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
[10:55:49][I][EVT_SETS:019]: checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
[10:55:49][I][EVT_SETS:019]: checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
[10:55:49][I][EVT_SETS:019]: checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
[10:55:49][I][EVT_SETS:019]: checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
[10:55:49][I][EVT_SETS:019]: checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
[... cut ...]
[10:55:49][I][EVT_SETS:019]: checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
[10:55:49][W][CN105:176]: Heatpump has not replied for 179 s
[10:55:49][I][CN105:177]: We think Heatpump is not connected anymore..
[10:55:49][D][CN105:156]: reconnectUART()
[10:55:49][D][CN105:145]: disconnectUART()
[10:55:49][I][EVT_SETS:019]: checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
```

Is there anything else that could be done? Thank you,