echavet/MitsubishiCN105ESPHome #359: strange new error in one of my head units

disruptivepatternmaterial, 2025-07-24:
```20:35:18 | [W] | [lookup:206] | wideVane reading (full byte) caution: value 131 not found, returning value at index 0```

I just rebuilt, 2025.7.3 Jul 23 2025, 10:22:5

Any idea what this might be about? The unit is MSZ-GL06NA...nothing special at all!

disruptivepatternmaterial, 2025-07-24:
Oh from everything I know there is not wideVane on this unit, so I wonder what it is returning....

echavet, 2025-07-27:
This issue is a consequence of a recent pull request I merged.
The aim was to support more wide vane options, but it seems to be more complicated.
I need to find a way to debug all this. But I don't have any wide vane on my units.
I'm thinking about adding a new feature on this firmware to support MAC-587IF-E commands via a secondary (man-in-the-middle) light firmware connected to MAC-587IF-E which would transmit packets through MQTT to the actual main firmware. 
But I don't have a MAC-587IF-E neither!! 
Perhaps it would be wise to roll back the commit that introduced the issue?

phrozen77, 2025-07-27:
> But I don't have a MAC-587IF-E neither!!

I have 2 unused 577IF-E laying around? Not sure those would be of help, they’ve been connected to PROG on my MSZ-LN##VG2V units?