echavet/MitsubishiCN105ESPHome #371: New split unit for list

usmcguy, 2025-08-12:
New user here. I just recently installed a Mitsubishi mini split for my Furnished Room Over the Garage (F.R.O.G). When I saw the price of an adapter for a smart thermostat, I knew I wasnt spending my money that way. Very excited to see there is a community here that has reversed engineered the comms protocol on the CN105 port.

I was very eager to get my system using ESPHome, as it was overcooling the room. Upon connecting a **_ESP32-C6-DevKitC-1_** board to  my indoor unit, **_MSZ-WX12NL_** (outdoor unit **_MUZ-WX12NL_**), I saw that the temperature sensor was reading 4F higher than the **_SHT-45_** temp/hum sensor I have connected to an ESP32 in the middle of the room. Following the guide on the readme page, I was able to get the system to use my SHT-45 as an external sensor for the mini split. Awesome!

Now, I am curious how I would go about controlling when the system is activated based on the room temp.  Currently, Im seeing the system kick on at 72F, for example, when the set temp is 74F. Curious how I would calibrate the temperature delta before the unit activates (deactivates?)?

Also, I enabled all the climate features/settings to see what was available for configuration. Below is the list of what is or is not configurable/provides data currently.

compressor_frequency_sensor: No
outside_air_temperature_sensor: No
vertical_vane_select: no
horizontal_vane_select: yes
isee_sensor: no
stage_sensor: no
sub_mode_sensor: no
auto_sub_mode_sensor: no
input_power_sensor: yes
kwh_sensor: yes
runtime_hours_sensor: yes

For the contributing devs reading this, I suspect the below output from the ESP32 logs might be helpful. I dont know what they mean, but perhaps its useful? I share it as the 'windVane' and 'auto mode sub mode' lines appear over and over in yellow.

[22:03:48][I][app:149]: ESPHome version 2025.7.5 compiled on Aug 11 2025, 21:58:56
[22:03:48][I][CYCLE:039]: 1: Cycle start
[22:03:48]**[W][lookup:206]: wideVane reading (full byte) caution: value 131 not found, returning value at index 0**
[22:03:49]**[W][lookup:206]: auto mode sub mode caution: value 64 not found, returning value at index 0**
[22:03:49][I][CYCLE:052]: 6: Cycle ended in 1.0 seconds (with timeout?:  NO)
[22:03:49][I][main:135]: Remote temperature received from HA: 22.5 C
[22:03:51][I][CYCLE:039]: 1: Cycle start
[22:03:51]**[W][lookup:206]: wideVane reading (full byte) caution: value 131 not found, returning value at index 0**
[22:03:52]**[W][lookup:206]: auto mode sub mode caution: value 64 not found, returning value at index 0**
[22:03:52][I][CYCLE:052]: 6: Cycle ended in 1.0 seconds (with timeout?:  NO)

keithbelter, 2025-08-14:
Filter it out by adding lookup: ERROR to your logger section

logger:  
  hardware_uart: UART1
  baud_rate: 0  # 0 Enables logging, but disables serial-port logging to free CPU and memory
  level: INFO
  logs:
    lookup: ERROR

echavet, 2026-09-15:
Closing this as informational. Thanks for reporting the working model — we track compatibility mainly via README / discussions rather than long-lived “add to list” issues.

If you hit a concrete bug on that unit (setpoint, vane, power, etc.), please open a dedicated issue with model + logs.

closed 2026-09-15 by echavet