echavet/MitsubishiCN105ESPHome #378: suddenly stopped working

jruibarroso, 2025-08-25:
Hi have 6 Mitsubishi indoor units (Ref.: PEFY-P32VMA-E2) all working with ESP32 connected on CN105. Yesterday 1 units stops talking with ESP32, already replaced ESP32, reboot indoor unit by powering off and on,  but keep with this error :

```
[19:30:39][I][app:200]: ESPHome version 2025.8.0 compiled on Aug 25 2025, 14:05:57
[19:30:49][E][CN105:032]: --> Heatpump did not reply: NOT CONNECTED <--
[19:30:49][I][CN105:033]: Trying to connect again...
[19:30:51][W][chkSum:083]: KO-> 01!=00 
[19:30:54][W][CN105:190]: Heatpump has not replied for 15 s
[19:30:54][I][CN105:191]: We think Heatpump is not connected anymore..
```
Any idea about this ? Also reduced baud rate to 1200, no success also

jruibarroso, 2025-08-25:
SOLVED , reverted EspHome to previous version 2025.07 and start working again 🥇 Just don't understant how it affeted just one 1 unit since all units are exactly same model and ESP32 boards also exatly same boards

echavet, 2025-08-26:
https://github.com/esphome/esphome/issues/10338

jimmy-zhening-luo, 2025-08-28:
> SOLVED , reverted EspHome to previous version 2025.07 and start working again 🥇 Just don't understant how it affeted just one 1 unit since all units are exactly same model and ESP32 boards also exatly same boards

@jruibarroso which version of 2025.7 worked for you?

dandashl1973, 2025-08-29:
I too had an issue.  For no reason at all, just good housekeeping - I decided to upgrade two Mits AC units to ESPHome 2025.8.0 from 2025.5.1, and it completely broke communication with my heat pump.  I got "Heat Pump Not connected".  Tried all the fixes I saw online, baud rate, buffers, disabled serial logging.  To no effect.   I am using an Atom S3 ESP32 board.  Finally had to downgrade to get them working again.

morrisdylan07, 2025-08-30:
I have 4 splits and one PAA coil. 3 of the splits and the PAA unit were connected via Atom S3 Lite boards.  All stopped working last week. No data, "heat pump not connected. They would then drop offline. My one split on an 8266 esp01_1m board works fine and has had no problems. . I switched the PAA unit to a Mahtanar custom board which also now works fine. 

None of the S3s work. Not sure which update broke them, but it seemed to require a power cycle before they went offline.  I tried new clean S3s without success. Just rolled back to IDF version: 5.3.2,  platform_version: 53.3.13 and the S3 works again. I left ESPHome at 2025.8.2, just rolled back the IDF version. It seems to be working with that change on the S3s.

usmcguy, 2025-08-30:
I am now running Home assistant 2025.8.2. ESP-IDF 5.4.2 that comes with it still doesnt work. But if I specify 5.4.1 in the device config, it works again.

esp32:
  board: esp32-c6-devkitc-1
  framework:
    type: esp-idf
    version: 5.4.1

echavet, 2025-09-09:
Could people who face this issue check their python version please ?
It looks like the 3.11 version is the minimal required. 
And Grok says it may be the root cause of the issue.

alfredocdmiranda, 2025-09-09:
@echavet I am using esphome docker image and I just checked that they use Python 3.12. Also, looking into the pyproject.toml file in esphome repository there is **requires-python = ">=3.11.0"**.

echavet, 2025-09-13:
> Unfortunately downgrading ESPHome to 2025.7.5 doesn't fix this for me. Nor does setting esp32 version 5.4.1. Is there any debugging information that would be helpful here?

Did you try to make a clean before you build again ?
We're are trying to diagnose precisely the pb. The versions logged in the building process could be a good info.

ollo69, 2025-09-13:
@thejkayway, 

Sorry for stupid question, but does it even run for you? If yes, using which version?

disruptivepatternmaterial, 2025-09-13:
It is worth noting that for me, to get this to work, I had to remove ALL the devices from my splits, ERASE THEM with webesp, like yes, erase!

then reflash with 2025.7.5

I am stunned at what a bad defect this is...it was so frustrating and cost me hours. I thought I was going crazy and perhaps burned out the control boards on my units!

So got https://adafruit.github.io/Adafruit_WebSerial_ESPTool/ or something like it, open the device/connect, erase, then close and refresh with 2025.7.5.

I only use m5 devices and they are all esp32s3 stamps, in my case.

phidauex, 2025-09-13:
> It is worth noting that for me, to get this to work, I had to remove ALL the devices from my splits, ERASE THEM with webesp, like yes, erase!

Yikes, I've been half following the reports through email, hoping a quick regression fix would appear upstream, but given that it hasn't, should I put a frontpage warning on the docs telling people not to attempt updating with 2025.8.x?

thejkayway, 2025-09-15:
> [@thejkayway](https://github.com/thejkayway),
> 
> Sorry for stupid question, but does it ever run for you? If yes, using which version?

I have two others running successfully that I hadn't updated since 2024.12.2 (where this one was previously). I'm not able to revert this one back that far though (build fails with `AttributeError: module 'esphome.components.climate' has no attribute 'climate_schema'`).

The problem is also not the mini split itself, as I can physically swap one of the other two previously-flashed esp32s and they'll work fine. The "updated" one on the other hand will not connect to the other head units. I've also tried with two other spare esp32s at versions 2025.7.5 and 2025.8.4 and I'm not able to flash any of them to successfully connect to any of the mini splits (but they had all been flashed with 2025.8.2 previously, so if there's something that persists between flashes they could be subject to that).

electronsailor, 2025-09-16:
Spent hours trying to get ESP32-S3 to work on one of my Mitsubishi mini split using 2025.8.3 then 2025.8.4
Reverting to 2025.7.5 fixed the communication issues, did not have to pre-erase the ESP32.

ironicbadger, 2025-09-21:
Adding a "me too" to this thread. The units power cycled and then I noticed the failure condition on ESPHome `2025.9.1`, but the `ESP-IDF` version pinning above did not resolve this in my case. Clean build, and power cycles etc.

pixelgrb, 2025-09-27:
Absolute new user (to this project) here.  Maybe this isn't helpful to anyone, but yesterday was my very first time installing and running this firmware.  It did not work:

```
[19:05:32][I][app:200]: ESPHome version 2025.8.1 compiled on Sep 26 2025, 19:04:53
[19:05:42][E][CN105:032]: --> Heatpump did not reply: NOT CONNECTED <--
[19:05:42][I][CN105:033]: Trying to connect again...
[19:05:52][W][CN105:190]: Heatpump has not replied for 20 s
[19:05:52][I][CN105:191]: We think Heatpump is not connected anymore..
[19:05:52][I][CN105:008]: setupUART() with baudrate  2400
[19:05:54][W][CN105:190]: Heatpump has not replied for 22 s
[19:05:54][I][CN105:191]: We think Heatpump is not connected anymore..
[19:05:54][I][CN105:008]: setupUART() with baudrate  2400
[19:05:56][W][CN105:190]: Heatpump has not replied for 24 s
[19:05:56][I][CN105:191]: We think Heatpump is not connected anymore..
...
```

This was with **ESPHome 2025.8.1**.  I did build and upload multiple times (OTA) but with only small changes to the YAML config that should not have been helpful.  I then found this thread where multiple people are suspecting ESPHome versioning as the culprit.

So I tried **ESPHome 2025.6** and it worked.  To do this, I created a new python venv, copied over my yaml files - so obviously this is a clean build - with an OTA update.  So I ran "esphome run <yaml>" .. and bang, it worked.

**Here's the interesting part.**
So that was yesterday.  Today I thought I would be helpful to everyone here, and pin down what versions of ESPHome work and don't work ... and .. **every version works now**, including the 2025.8.1 version that I started with.

But get this.  Remember above I had created a new python venv to do this testing, and I copied over YAML files etc.  I went back my system's installed ESPHome 2025.8.1 (homebrew install - which I never messed with) AND the original folder with my original YAML file (which I did not mess with either) .. and it works fine now:

```
[14:36:38][I][app:200]: ESPHome version 2025.8.1 compiled on Sep 27 2025, 14:36:12
[14:36:40][I][CYCLE:039]: 1: Cycle start
[14:36:40][I][CN105:767]: power or mode changed
[14:36:40][I][CN105:737]: fan setting changed
[14:36:40][I][SETTINGS:646]: vane setting changed
[14:36:40][I][CN105:717]: vane setting (extra select component) changed
[14:36:40][I][STATUS:130]: [received]-> [room C°: 25.0, operating: NO , compressor freq: nan Hz]
[14:36:40][I][STATUS:130]: [current]-> [room C°: 0.0, operating: NO , compressor freq: nan Hz]
[14:36:41][I][STATUS:130]: [received]-> [room C°: 25.0, operating: NO , compressor freq: 3.0 Hz]
[14:36:41][I][STATUS:130]: [current]-> [room C°: 25.0, operating: NO , compressor freq: nan Hz]
[14:36:41][I][CYCLE:052]: 6: Cycle ended in 1.1 seconds (with timeout?:  NO)
[14:36:43][I][CYCLE:039]: 1: Cycle start
[14:36:44][I][CYCLE:052]: 6: Cycle ended in 1.1 seconds (with timeout?:  NO)
[14:36:46][I][CYCLE:039]: 1: Cycle start
```

I double checked everything - I'm using separate shells, 'which esphome' (not a rehash issue) etc.  I did rebuild it today (in hindsight, I wish I had just uploaded yesterday's build).  But it is the same 'esphome' install from yesterday and the same yaml files.

Has anyone else .. after 'getting it to work on an old esphome version' .. tried .. 'going back to a esphome version that wasn't working' and see it break again?  i.e. like I tried.

To be explicit, I have been doing this to switch versions (OTA updates) in my venv:

```
(.venv) % pip install esphome==2025.9.1
(.venv) % esphome clean
(.venv) % esphome run <yaml>
```

ollo69, 2025-09-28:
> Has anyone else .. after 'getting it to work on an old esphome version' .. tried .. 'going back to a esphome version that wasn't working before' and **see it break again**? i.e. like I tried.

Did you `cold boot` your split / ESP32 device after update to 2025.8.1 version?

pixelgrb, 2025-09-28:
> Did you `cold boot` your split / ESP32 device after update to 2025.8.1 version?

@ollo69 - thanks for the question.  I think your intuition is on point.

Answer is: no

My unit is a SEZ-KD18NA4 (ducted split system in an attic space).  I need to switch off a breaker 3 floors down to power it off.  I have unwilling but obedient kids though.  :)

What I did on my first day (**when it was failing**): 
- I initially hot-swapped the ESP32 (with the MIFH1 remote).
- Only once did I bounce the breaker which cold boots both split + esp32.
- I disconnected and reconnected the ESP32 from the CN105 port about 5 times (cold booting it alone).
- I did many OTA updates on 2025.8.1 (with small changes to YAML config) with the same result .. maybe 7 times.

Then I found this thread and tried 2025.6.  This was an OTA update.  And I have only tried OTA since - at least 7 times with newer firmware, always working.

or completeness, I am using a M5Stack NanoC6 (exact same as https://github.com/echavet/MitsubishiCN105ESPHome/discussions/340).

So as @echavet mentioned, maybe there is a random initial unknown/hazardous state.  It looks like once the electronics have got things working .. they stay that way.  So your intuition about cold booting is spot on.  Maybe:
- cold booting **is required to get it to fail again** (presumably only on newer firmware)
- but cold booting **is not required to get to succeed** (presumably on older firmware)

Wonder if the same is true for when it's broken.  Maybe the unknown state is random, but it stays broken on old firmware until a cold boot?  

I'll investigate more:
- I'll try cold booting the esp32 alone a bunch of times and see if I can get it to fail again.
- likewise .. cold booting the split a bunch of times.
- once I get that to repro .. characterize getting it working again too.

... and hopefully .. bisect on an esphome version.  🤞

ollo69, 2025-09-28:
> ollo69 - thanks for the question. I think your intuition is on point.

In my case on first upgrade to 2025.8.1 with split already running with prev version everything was fine. But it stop working when I had a power outage, and after that the only way to fix is to rollback to 2025.7. So it seems that the problem come on first communication initialization using 2025.8.

michaeljb2023, 2025-09-28:
> In my case on first upgrade to 2025.8.1 with split already running with prev version everything was fine. But it stop working when I had a power outage, and after that the only way to fix is to [...]

Agree, this is exactly what happened to me.

pixelgrb, 2025-09-29:
I did a 'git bisect' of esphome and nailed it down to this revision:

**Bump ESP32 IDF version to 5.4.2 and Arduino version to 3.2.1 (#9770)**

Unfortunately, that's another can of worms.  I'm not sure if I'll be able to bisect that next layer so easily.

Note:
- Each time I saw a 'good' result, I had to unplug the esp32 so it cold boots on the next attempt.  Remember that the UART failure won't happen if doing an OTA update from working to non-working code.

```
% git bisect bad
116c91e9c5fc6d0d32191bd4e6d6e406e2bff6bf is the first bad commit
commit 116c91e9c5fc6d0d32191bd4e6d6e406e2bff6bf
Author: Jonathan Swoboda <154711427+swoboda1337@users.noreply.github.com>
Date:   Tue Jul 22 19:15:31 2025 -0400

    Bump ESP32 IDF version to 5.4.2 and Arduino version to 3.2.1 (#9770)

 .clang-tidy.hash                                 |  2 +-
 esphome/components/esp32/__init__.py             | 16 ++++++++--------
 esphome/components/ethernet/esp_eth_phy_jl1101.c |  5 +++++
 esphome/components/ethernet/ethernet_component.h |  4 ++++
 esphome/core/defines.h                           |  3 +--
 platformio.ini                                   |  8 ++++----
 script/clang-tidy                                |  2 +-
 7 files changed, 24 insertions(+), 16 deletions(-)
```

GioMez, 2025-10-20:
I suggest to have a look at this https://github.com/esphome/esphome/issues/11038#issuecomment-3374955358 with a better workaround: reset tx/rx pins shortly after boot.

Tested also after multiple cold boots and it seems to be more stable.

disruptivepatternmaterial, 2025-10-22:
same...I put in the pin resets and I used esphome:latest and all is good!

pixelgrb, 2025-10-22:
Works for me too.  Just different pins on the M5Stack NanoC6:

```
esphome:
  on_boot:
    priority: 1001
    then:
      - lambda: |-
          gpio_reset_pin((gpio_num_t)1);
          gpio_reset_pin((gpio_num_t)2);
```

swoboda1337, 2025-10-23:
I unlocked the issue. Has anyone tried 5.5.1?
```
esp32:
  framework:
    type: esp-idf
    version: 5.5.1
```

echavet, 2025-10-23:
> I unlocked the issue. Has anyone tried 5.5.1?
> 
> ```
> esp32:
>   framework:
>     type: esp-idf
>     version: 5.5.1
> ```

Yes, it fails

5.4.0 is ok but forcing pin reset from lambda at esp boot `gpio_reset_pin(<pin number>);` is the best workaround.

closed 2025-10-23 by echavet

swoboda1337, 2025-10-23:
> Yes, it fails

Could try this, I think its IDF master from Sept 29th, contains unreleased changes done after 5.5.1:
```
esp32:
  framework:
    type: esp-idf
    version: 5.5.1
    source: https://github.com/pioarduino/esp-idf/releases/download/v5.5.1.250929/esp-idf-v5.5.1.tar.xz
    platform_version: 55.03.32
```

swoboda1337, 2025-10-23:
Sounds good, 2025.10.3 should is being released now.