echavet/MitsubishiCN105ESPHome #379: Getting runtime errors using M5Stack Atom Lite

aaronouthier, 2025-08-25:
Device worked after initial setup from ESPHome Web installer, however, when I compile the actual code listed for this project and install via USB Port, I get a reset loop with flash read errors repeating down the screen. The data was written less than 3 Dozen times. I didn't realize the logs feature were the runtime logs, or I would have tried that sooner. The firmware file does compile and reports it installed successfully. It never connects to my WiFi.

Log file attached.

[logs.txt](https://github.com/user-attachments/files/21976078/logs.txt)

aaronouthier, 2025-08-26:
Thank you for the detailed report.
I'm not convinced the issue is hardware related, as I just received the unit on Saturday as brand new stock. As stated in the subject line, this is an m5stack Atom Lite, which is listed as compatible with this project. As the unit is sealed inside of a plastic housing, there is very little I can do about physical inspections.

Power supply is a stable 5V, coming from the built-in USB-C Port. I don't believe my Chromebook can run Python scripts, so I don't believe I can erase the flash completely using the steps presented. I'm not certain as to how to update the bootloader. I copied and pasted the code from this project, and don't understand enough about it to accomplish this. I am new to ESPHome.

aaronouthier, 2025-08-26:
Still not working. Seems to be a config mismatch with this particular hardware. Does anyone else using this project have an Atom Lite?

Also, does it matter that the log entries occur at about 1 second intervals, as in the resets occur about 1 second apart? I would guess this is the watchdog timeout.

I'm changing the debug level to DEBUG instead of INFO to see if that helps trace the issue.

aaronouthier, 2025-08-26:
Probably a silly question, but it wouldn't matter that I haven't yet plugged this into my Heat pump yet, right?
Setting to DEBUG logging didn't help. I tried the script above, but I had to modify it quite a bit to get it to compile.

YAML:
```
esphome:
  name: m5stack-atom-lite

esp32:
  board: m5stack-atom

api:
  encryption:
    key: "2dGoGVQeOz4sXvjrLuHAXuudVUkAWRpGsVXcjVDVO14="

ota:
  - platform: esphome
    password: "2929127c711f83338e4beb638a27b1c5"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

logger:
  level: DEBUG

script:
  - id: boot_script
    then:
      - logger.log: "Hello World from M5Stack Atom Lite!"
```

Resulting log:
```
[13:24:22][I][safe_mode:042]: Boot seems successful; resetting boot loop counter
[13:24:52][D][esp32.preferences:142]: Writing 1ets Jun  8 2016 00:22:57
[13:24:52]
[13:24:52]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[13:24:52]configsip: 188777542, SPIWP:0xee
[13:24:52]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[13:24:52]mode:DIO, clock div:2
[13:24:52]load:0x3fff0030,len:4660
[13:24:52]load:0x40078000,len:15556
[13:24:52]load:0x40080400,len:4
[13:24:52]ho 8 tail 4 room 4
[13:24:52]load:0x40080404,len:3152
[13:24:52]entry 0x400805a0
[13:24:53]E (435) esp_core_dump_flash: No c�����dump partition found!
[13:24:53]E (435) esp_core_dump_flash: No core dump partition found!
[13:24:53][I][logger:165]: Log initialized
[13:24:53][C][safe_mode:082]: There have been 0 suspected unsuccessful boot attempts
[13:24:53][D][esp32.preferences:142]: Writing 1 items: 0 cached, 1 written, 0 failed
[13:24:53][I][app:090]: Running through setup()
[13:24:53][C][component:164]: Setup preferences took 0ms
[13:24:53][C][wifi:060]: Starting
[13:24:53][C][wifi:060]:  Local MAC: 90:15:06:FD:B8:BC
[13:24:53][D][wifi:505]: Starting scan
[13:24:53][C][component:164]: Setup wifi took 113ms
[13:24:53][W][component:289]: wifi set Warning flag: scanning for networks
[13:24:56][D][wifi:574]: Found networks:
[13:24:56][I][wifi:597]: - '[redacted]' [redacted]▂▄▆█
[13:24:56][D][wifi:599]:     Channel: 11
[13:24:56][D][wifi:599]:    RSSI: -55 dB
[13:24:56][D][wifi:604]: - [redacted] [redacted]▂▄▆█
[13:24:56][D][wifi:604]: - [redacted] [redacted]▂▄▆█
[13:24:56][D][wifi:604]: - [redacted] [redacted]▂▄▆█
[13:24:56][D][wifi:604]: - [redacted] [redacted]▂▄▆█
[13:24:56][D][wifi:604]: - [redacted] [redacted]▂▄▆█
[13:24:56][D][wifi:604]: - [redacted] [redacted]▂▄▆█
[13:24:56][D][wifi:604]: - [redacted] [redacted]▂▄▆█
[... 47 lines cut ...]
[13:24:58][C][logger:252]: Logger:
[13:24:58][C][logger:252]:  Max Level: DEBUG
[13:24:58][C][logger:252]:  Initial Level: DEBUG
[13:24:58][C][logger:258]:   Log Baud Rate: 115200
[13:24:58][C][logger:258]:  Hardware UART: UART0
[13:24:59][C][logger:265]:   Task Log Buffer Size: 768
[13:24:59][C][esphome.ota:075]: Over-The-Air updates:
[13:24:59][C][esphome.ota:075]:  Address: m5stack-atom-lite.local:3232
[13:24:59][C][esphome.ota:075]:  Version: 2
[13:24:59][C][esphome.ota:082]:   Password configured
[13:24:59][C][safe_mode:018]: Safe Mode:
[13:24:59][C][safe_mode:019]:   Boot considered successful after 60 seconds
[13:24:59][C][safe_mode:019]:  Invoke after 10 boot attempts
[13:24:59][C][safe_mode:019]:  Remain for 300 seconds
[13:24:59][C][api:205]: Server:
[13:24:59][C][api:205]:  Address: m5stack-atom-lite.local:6053
[13:24:59][C][api:210]:   Noise encryption: YES
[13:24:59][C][mdns:124]: mDNS:
[13:24:59][C][mdns:124]:  Hostname: m5stack-atom-lite
[13:25:00][D][api:144]: Accept 10.10.1.222
[13:25:00][W][component:317]: api cleared Warning flag
[13:25:00][W][component:423]: api took a long time for an operation (54 ms)
[13:25:00][W][component:424]: Components should block for at most 30 ms
[13:25:00][D][api.connection:1341]: Home Assistant 2025.8.3 ([redacted]) connected
```

aaronouthier, 2025-08-26:
Is it supposed to log the last string in the YAML file, because I don't see it in the logs.

echavet, 2025-08-27:
No you don't need to connect the chip to the heatpump to flash it.

But you may need to press and hold the Reset button on the M5Stack Atom Lite while connecting the USB-C cable to enter download mode for flashing. Release the button once the tool (e.g., M5Burner or esptool) detects the device.