echavet/MitsubishiCN105ESPHome #386: Input Power reads unavailable or zero

sigmabeta, 2025-09-02:
I have four indoor units. The input power fields for these units read as follows in home assistant:

- MSZFH15NA - 0W
- MSZFH15NA - Disabled (weird, they're the same model unit)
- MSZFH09NA - 0W
- MSZFH06NA - 0W

I can supply logs or the full yaml if necessary. For now:

```
    input_power_sensor:
      name: Input Power
      disabled_by_default: false
```

They're definitely not using zero power, as confirmed by my Sense energy monitor and the fact that my house is 10 degrees fahrenheit cooler than outside.

Any ideas on what I need to configure differently, or is this just a bug?

lrehmann, 2025-09-06:
some models are hit or miss when it comes to these diagnostic sensors, though personally i've had a bit of variance in supported diagnostic sensors with the newer versions

sigmabeta, 2025-09-06:
Yeah, that makes sense. I'm willing to get my hands dirty with the cpp/py code to make some progress on this. Is there ongoing discussion / documentation about how these diagnostic sensors work somewhere?

flicker581, 2025-09-16:
Just not every model has power sensor builtin. Use external sensor.

vidarak, 2025-10-23:
These definitely depend on the model.

My new MSZ-LN35VG2W shows correct values for power and energy (it seems). 

[image]

I cannot get it to show compressor frequency (the old did), but the outdoor temp works (the old didn't).