echavet/MitsubishiCN105ESPHome #391: 4x ATOMS3 LITE's no longer controlling heatpumps with the same error

haydnlj, 2025-09-18:
Background is that I have 4x ATOMS3 LITE's that have been sucessfully controlling 4x Mitsubitshi heat pumps via the CN105 port from this project for over 9 months. See attached YAML [HVAC_HP.YAML](https://github.com/user-attachments/files/22415869/HVAC_HP.YAML)

The last time I remember controlling one of the heat pumps successfully was 10 days ago (noted in Home Assistant logbook). At that point ESPHome Builder was running version 2025.8.4. 2 days ago I updated the ESPHome Builder 2025.9.0, and have noticed that none of the heatpumps are controllable - the ATOMS3's webservers are reporting the same error repeating;

`07:35:22[W][CN105:190]Heatpump has not replied for 2556 s
07:35:22[I][CN105:191]We think Heatpump is not connected anymore..
07:35:22[I][CN105:008]setupUART() with baudrate 2400`

I downgraded ESPHome Builder to 2025.8.4, did a "Clean Build Files" and "Install", but getting the same errors.

I last installed firmware manually 16 days ago to address a breaking change on "rmt_channel" no longer being needed - I therefore know it was working with upstream versions at that time. Unsure if updating ESPHome Builder, or a Home Assistant restart forced a new install since then.

Claude and Gemini are both suggesting to try locking the versions of either or both of the board "esp32-s3-devkitc-1" or this external component "MitsubishiCN105ESPHome". I have tried a both and each individually, but not getting any difference outcome.

Would appreciate some guidance!

haydnlj, 2025-09-19:
Thanks to @aaronouthier [here](https://github.com/echavet/MitsubishiCN105ESPHome/issues/380#issuecomment-3293673519)

Changing my board from **esp32-s3-devkitc-1** to **m5stack-atoms3**
seemed to have worked (still on 2025.7.5 for now)

xeoMe015, 2025-09-19:
Had some issues few days ago which were resolved by downgrading to ESPHome 2025.7

Wanted to try the above solution by changing my board type so stopped the ESPHome 2025.7 and installed the latest ESPHome 2025.9 and did a clean build.
However mine was instantly working without changing the board type, but maybe because i have the variant specified.

My configuration:

[Working ESPHome 2025.9.txt](https://github.com/user-attachments/files/22423850/Working.ESPHome.2025.9.txt)

pixelgrb, 2025-09-30:
@xeoMe015 - I think you hit the same problem I and some others hit.  e.g.
- https://github.com/echavet/MitsubishiCN105ESPHome/issues/378
- https://github.com/echavet/MitsubishiCN105ESPHome/issues/396

I narrowed it down to a [single code change in ESP-IDF v5.4.1](https://github.com/espressif/esp-idf/issues/14787#issuecomment-3353828630).  The current solutions are to:
- revert to ESPHome 2025.7.5 or below.  This fixes ESP-IDF to v5.3.2.
- bump ESP-IDF to v5.4.0 in the YAML.  Something like this:

```
esp32:
  board: esp32-s3-devkitc-1  
  framework:
    type: esp-idf
    version: 5.4.0
  variant: esp32s3
  flash_size: 8MB
```

haydnlj, 2025-09-30:
Thanks for the reply, I ended up going with both suggestions based on the reading. Figured I could happily sit on ESPHome 2025.7.5 for a few months if needed. Good to know ESP-IDF version lock is a workable alternative, and will look to update ESPHome to current release now.

pixelgrb, 2025-10-24:
FYI - this issue has been found and resolved.  There are a few ways to fix (other than reverting to an old ESPHome / ESP-IDF).  I list them here: https://github.com/echavet/MitsubishiCN105ESPHome/issues/396#issuecomment-3445264578

closed 2025-10-25 by haydnlj