echavet/MitsubishiCN105ESPHome #395: Heatpump turns off after 2–3 minutes on PEAD Ducted Unit

sherryl4george, 2025-09-29:
I have a PEAD-A18AA7 ducted unit connected via an Airzone CN105 splitter with an MHK2 thermostat. I also have an Atom S3 (with a bi-directional level shifter, ref: #273) wired in. The ESP32 successfully detects the unit, but when I set the mode to COOL, the heatpump

 only runs for about 2–4 minutes (sometimes even less) before shutting itself off.

Debugging steps attempted (no success):
1. Removed MHK2 schedules and enabled setting 134 (Central Unit in control) to confirm the MHK2 isn’t sending an OFF command.
2. Completely removed the MHK2 and its controller from the bus — same behavior.
3. Turned HP ON to COOL directly from the MHK2 instead of Home Assistant — same behavior.

Logs from last test run:

``` 
[17:17:51.140][D][control:044]: espHome control() interface method called...
[17:17:51.140][D][control:051]: Mode change asked
[17:17:51.140][I][control:246]: changing mode to COOL
[17:17:51.140][D][EVT_SETS:082]: clim.control() -> User changed something...
[17:17:51.141][D][EVT_SETS:062]: [control (wantedSettings)]-> [power: ON, target °C: -1.0, mode: COOL, fan: -, vane: -, wvane: -, hasChanged ? -> YES, hasBeenSent ? ->  NO]
[17:17:51.432][D][READ:172]: FC 62 01 30 10 06 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 57 
[17:17:51.432][D][CYCLE:453]: 4b: Receiving status response
[17:17:51.435][D][Decoder:341]: [0x06 is status]
[17:17:51.437][D][CYCLE:457]: 5a: Sending power request (0x09)
[17:17:51.437][D][CN105:390]: creating Info packet
[17:17:51.439][D][CN105:084]: writing packet...
[17:17:51.440][D][WRITE:172]: FC 42 01 30 10 09 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 74 
[17:17:51.801][D][READ:172]: FC 62 01 30 10 09 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 54 
[17:17:51.802][D][CYCLE:472]: 5b: Receiving Power/Standby response
[17:17:51.805][D][Decoder:152]: [0x09 is sub modes]
[17:17:51.806][D][Decoder:159]: [Stage : IDLE]
[17:17:51.808][D][Decoder:160]: [Sub Mode  : NORMAL]
[17:17:51.813][D][Decoder:161]: [Auto Mode Sub Mode  : AUTO_OFF]
[17:17:51.814][I][CYCLE:052]: 6: Cycle ended in 1.5 seconds (with timeout?:  NO)
[17:17:51.816][I][EVT_SETS:019]: checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
[17:17:51.816][I][CN105:317]: sending wantedSettings..
[17:17:51.818][D][EVT_SETS:062]: [wantedSettings]-> [power: ON, target °C: -1.0, mode: COOL, fan: -, vane: -, wvane: -, hasChanged ? -> YES, hasBeenSent ? -> YES]
[17:17:51.823][D][CN105:186]: building packet for writing...
[17:17:51.823][D][CN105:189]: power -> ON
[17:17:51.824][D][CN105:195]: heatpump mode -> COOL
[17:17:51.824][D][CN105:084]: writing packet...
[17:17:51.825][D][WRITE:172]: FC 41 01 30 10 01 03 00 01 03 00 00 00 00 00 00 00 00 00 00 00 76 
[17:17:51.828][D][WRITE_SETTINGS:172]: FC 41 01 30 10 01 03 00 01 03 00 00 00 00 00 00 00 00 00 00 00 76 
[17:17:51.829][I][CN105:767]: power or mode changed
[17:17:51.834][D][CN105:318]: Setting action to 4 (effective_operating: false, use_stage_fallback: no, current_stage: IDLE)
[17:17:51.835][D][CN105:396]: Climate mode is: 2
[17:17:51.835][D][CN105:397]: Climate action is: 4
[17:17:51.835][I][CYCLE:008]: Defering cycle trigger of   1500  ms
[17:17:52.173][D][READ:172]: FC 61 01 30 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 5E 
[17:17:52.173][D][Update-ACK:172]: FC 61 01 30 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 5E 
[17:17:52.175][D][ACK:521]: Last heatpump data update successful!
[17:17:55.342][D][CN105:375]: sending a request for settings packet (0x02)
== heatpump Runs in COOL ==
[17:21:41.772][I][CYCLE:039]: 1: Cycle start
[17:21:41.772][D][CYCLE:378]: 2a: Sending settings request (0x02)
[17:21:41.772][D][CN105:390]: creating Info packet
[17:21:41.772][D][CN105:084]: writing packet...
[17:21:41.773][D][WRITE:172]: FC 42 01 30 10 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7B 
[17:21:42.108][D][READ:172]: FC 62 01 30 10 02 00 00 01 03 0B 05 01 00 00 00 A8 00 00 00 00 9E 
[17:21:42.109][D][CYCLE:416]: 2b: Receiving settings response
[17:21:42.113][D][Decoder:209]: [0x02 is settings]
[17:21:42.114][D][Decoder:216]: [Power : ON]
[17:21:42.114][D][Decoder:217]: [iSee  : 0]
[17:21:42.118][D][Decoder:218]: [Mode  : COOL]
[17:21:42.119][D][Decoder:232]: [Temp °C: 20.000000]
[17:21:42.119][D][Decoder:235]: [Fan: 3]
[17:21:42.121][D][Decoder:238]: [Vane: ↑↑]
[17:21:42.122][D][Decoder:246]: widevane is not supported
[17:21:42.123][D][SETTINGS:105]: [current]-> [power: ON, target °C: 20.0, mode: COOL, fan: 3, vane: ↑↑, wvane: -]
[17:21:42.125][D][SETTINGS:105]: [received]-> [power: ON, target °C: 20.0, mode: COOL, fan: 3, vane: ↑↑, wvane: -]
[17:21:42.127][D][EVT_SETS:062]: [wanted]-> [power: -, target °C: -1.0, mode: -, fan: -, vane: -, wvane: -, hasChanged ? ->  NO, hasBeenSent ? ->  NO]
[17:21:42.135][D][SETTINGS:089]: [climate]-> [mode: COOL, target °C: 20.0, fan: MIDDLE, swing: OFF]
[17:21:42.135][D][CYCLE:419]: 3a: Sending room °C request (0x03)
[... 61 lines cut ...]
[17:21:45.999][D][CN105:390]: creating Info packet
[17:21:45.999][D][CN105:084]: writing packet...
[17:21:45.999][D][WRITE:172]: FC 42 01 30 10 06 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 77 
[17:21:46.340][D][READ:172]: FC 62 01 30 10 06 00 00 12 00 00 00 00 00 00 00 00 00 00 00 00 45 
[17:21:46.341][D][CYCLE:453]: 4b: Receiving status response
[17:21:46.343][D][Decoder:341]: [0x06 is status]
[17:21:46.346][I][STATUS:130]: [received]-> [room C°: 30.6, operating: NO , compressor freq: 18.0 Hz]
[17:21:46.348][I][STATUS:130]: [current]-> [room C°: 30.6, operating: YES, compressor freq: 17.0 Hz]
[17:21:46.349][D][CN105:396]: Climate mode is: 0
[17:21:46.354][D][CN105:397]: Climate action is: 0
[17:21:46.354][D][CYCLE:457]: 5a: Sending power request (0x09)
[17:21:46.355][D][CN105:390]: creating Info packet
[17:21:46.361][D][CN105:084]: writing packet...
[17:21:46.361][D][WRITE:172]: FC 42 01 30 10 09 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 74 
[17:21:46.705][D][READ:172]: FC 62 01 30 10 09 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 54 
[17:21:46.707][D][CYCLE:472]: 5b: Receiving Power/Standby response
[17:21:46.708][D][Decoder:152]: [0x09 is sub modes]
[17:21:46.708][D][Decoder:159]: [Stage : IDLE]
[17:21:46.710][D][Decoder:160]: [Sub Mode  : NORMAL]
[17:21:46.712][D][Decoder:161]: [Auto Mode Sub Mode  : AUTO_OFF]
[17:21:46.714][D][text_sensor:069]: 'AC Second Floor Stage': Sending state 'IDLE'
[17:21:46.719][I][CYCLE:052]: 6: Cycle ended in 1.5 seconds (with timeout?:  NO)
```

Any pointers on how I can debug this further?

maxwell5555, 2025-10-16:
> Hi,
> 
> Are you using multisplit setups?
> 
> Thanks, Eric

If I understand the question correctly, yes. I have an outdoor unit connected to two separate air handlers through a branch box. 
Outdoor unit: MXZSM48NAMHZ
Indoor wall mounted air handler: MSZEF09NAS
Indoor ducted air handler: SVZKP36NA

The msz unit has a d1 mini that's worked perfectly for the last 2 years. I've only just tried getting the svz unit connected to an esp. Until now it's used kumo.

echavet, 2026-09-15:
Closing: this issue is already labeled `wontfix`. The PEAD + Airzone splitter + MHK2 setup remains outside what we can reliably fix in this firmware (competing masters on CN105).

If someone later finds a clean CN105-only repro without the splitter/thermostat, please open a new issue with logs.

closed 2026-09-15 by echavet