echavet/MitsubishiCN105ESPHome #399: HEAT_COOL Spread (not Auto) not working

BenHolmes, 2025-09-30:
HEAT_COOL Spread (not Auto) not working
How HEAT_COOL should work

[image]

Pictured below is currently how this mode is not responding / providing the necessary controls for use.
[image]

[image]

Perhaps this is just a config issue on my part? but it seems that Auto is functioning fine, although that is very inefficient and does provide that ability to define heating and cooling values. #39 

```
substitutions:
  name: mitsubishi-heat-pump
  friendly_name: Mitsubishi Heat Pump

esphome:
  name: ${name}
  friendly_name: ${friendly_name}

esp8266:
  board: d1_mini

uart:
  id: HP_UART
  baud_rate: 2400
  tx_pin: 1
  rx_pin: 3
  
# External component reference
external_components:
  - source: github://echavet/MitsubishiCN105ESPHome

# Default logging level
logger:
  hardware_uart: UART1 # Uncomment on ESP8266 devices
  level: INFO

# Enable Home Assistant API
api:
  encryption:
    key: "{secret}"
  services:
    - service: set_remote_temperature
      variables:
        temperature: float
      then:
        # Select between the C version and the F version
        # Uncomment just ONE of the below lines. The top receives the temperature value in C,
        # the bottom receives the value in F, converting to C here.
        - lambda: "id(mhp).set_remote_temperature(temperature);"
    #        - lambda: 'id(hp).set_remote_temperature((temperature - 32.0) * (5.0 / 9.0));'
    - service: use_internal_temperature
      then:
        - lambda: "id(mhp).set_remote_temperature(0);"
  # services:
  #   - service: set_remote_temperature
  #     variables:
  #       temperature: float
  #     then:
  #       - lambda: 'id(mhp).set_remote_temperature(temperature);'

  #   - service: use_internal_temperature
  #     then:
  #       - lambda: 'id(mhp).set_remote_temperature(0);'
sensor:
  - platform: uptime
    name: Uptime
    
ota:
  platform: esphome
  password: "{apikey}"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Mitsubishi-Heat-Pump"
    password: "{pswrd}"

captive_portal:

web_server:
  port: 80

time:
  - platform: homeassistant
    id: homeassistant_time

text_sensor:
  # Expose ESPHome version as sensor.
  - platform: version
    name: ESPHome Version
  # Expose WiFi information as sensors.
  - platform: wifi_info
    ip_address:
      name: IP
    ssid:
      name: SSID
    bssid:
      name: BSSID

# Climate entity configuration
climate:
  - platform: cn105
    id: mhp
    name: "Mitsubishi Heat Pump" # update interval can be adjusted after a first run and logs monitoring
    icon: mdi:heat-pump
    supports:
      mode: [HEAT_COOL, AUTO, COOL, HEAT, FAN_ONLY]
      fan_mode: [AUTO, LOW, MEDIUM, HIGH]
      swing_mode:
    visual:
      min_temperature: 15
      max_temperature: 31
      temperature_step:
        target_temperature: 1
        current_temperature: 0.5
    # Timeout and communication settings
    remote_temperature_timeout: 30min
    update_interval: 4s
    debounce_delay : 100ms
    # Various optional sensors, not all sensors are supported by all heatpumps
    compressor_frequency_sensor:
      name: Compressor Frequency
      entity_category: diagnostic
      disabled_by_default: true
    outside_air_temperature_sensor:
      name: Outside Air Temp
      disabled_by_default: true
    vertical_vane_select:
      name: Vertical Vane
      disabled_by_default: false
    horizontal_vane_select:
      name: Horizontal Vane
      disabled_by_default: true
    isee_sensor:
      name: ISEE Sensor
      disabled_by_default: true
    stage_sensor:
      name: Stage
      entity_category: diagnostic
      disabled_by_default: true
    sub_mode_sensor:
      name: Sub Mode
      entity_category: diagnostic
      disabled_by_default: true
    auto_sub_mode_sensor:
      name: Auto Sub Mode
      entity_category: diagnostic
      disabled_by_default: true
    input_power_sensor:
      name: Input Power
      disabled_by_default: true
    kwh_sensor:
      name: Energy Usage
      disabled_by_default: true
    runtime_hours_sensor:
      name: Runtime Hours
      entity_category: diagnostic
      disabled_by_default: true
```

disruptivepatternmaterial, 2025-10-02:
Just use the YAML I shared, dual set point works perfectly....I have been doing it for months now.

If you use the cn105 in this repo this will work...just make sure you fine tune it for the exact capabilities your setup has (number of fan modes, etc...)

As I have written for years, the units DO NOT supper dual set point. It is implemented in the controller...I'm sorry you were surprised by this :( .

The deadband settings are right there in my code, they are in 2 places right now, so that is up to you if you want to externalize that or simplify it more.

I HATE things on my wall, I dont even want light switches...they are needed by building code however. Because of this I have sensors built into my house all over the place sending data to a central database which then takes averages and points the vents and circulation systems on and off.

The system looks three hours ahead for weather and pre-adjusts, it adjusts when I am more than 50 miles away and when I wake up and when I go to bed.

So I have no remotes, because I just let the system control the temperature per the setup and the headbands and hysteresis works perfectly.

```
platform: thermostat
name: Thermostat
id: meta_climate
icon: mdi:heat-pump
visual:
  min_temperature: 17
  max_temperature: 27
  temperature_step:
    target_temperature: 0.05
    current_temperature: 0.05
sensor: room_temperature
humidity_sensor: room_humidity
# Hysteresis settings for better control
heat_deadband: 1.0
heat_overrun: 1.0
cool_deadband: 1.0
cool_overrun: 1.0
# Timing controls to prevent short cycling
min_heating_off_time: 180s
min_heating_run_time: 420s
min_cooling_off_time: 300s
min_cooling_run_time: 600s
min_fanning_off_time: 60s
min_fanning_run_time: 600s
min_idle_time: 180s
min_fan_mode_switching_time: 120s
startup_delay: true
#on_boot_restore_from: memory
on_boot_restore_from: default_preset
default_preset: Home
preset:
  - name: Home in Summer
    default_target_temperature_low: 21.11
    default_target_temperature_high: 23.33
    mode: COOL
    fan_mode: "AUTO"
    swing_mode: "BOTH"
  - name: Home in Winter
    default_target_temperature_low: 22.22
    default_target_temperature_high: 25.56
    mode: HEAT
    fan_mode: "AUTO"
    swing_mode: "BOTH"
  - name: Home
    default_target_temperature_low: 21.11
    default_target_temperature_high: 23.89
    mode: HEAT_COOL
    fan_mode: "AUTO"
    swing_mode: "BOTH"
  - name: Sleep
    default_target_temperature_low: 18.33
    default_target_temperature_high: 22.22
    mode: HEAT_COOL
    fan_mode: "AUTO"
    swing_mode: "BOTH"
[... 263 lines cut ...]

swing_off_action:
  then:
    - climate.control:
        id: splitunit
        swing_mode: "OFF"

swing_vertical_action:
  then:
    - climate.control:
        id: splitunit
        swing_mode: "VERTICAL"

swing_horizontal_action:
  then:
    - climate.control:
        id: splitunit
        swing_mode: "HORIZONTAL"

swing_both_action:
  then:
    - climate.control:
        id: splitunit
        swing_mode: "BOTH"
```

disruptivepatternmaterial, 2025-10-02:
I opened up a #400 PR for all my files and a readme.

BenHolmes, 2025-10-03:
This is all really great amazing info! The Dual set point meta_climate is exactly what I am looking for!

I was successfully able to get and older config from this thread to work but I don't actually have HEAT_COOL mode available as an option. and the dry function does not work

[image]

However when I attempt to use the latest suggested config  resolves but does not reconnect to esphome api.

The config that works-ish

```
substitutions:
  name: mitsubishi-heat-pump
  friendly_name: Mitsubishi Heat Pump
  remote_humid_sensor: sensor.living_area_humidity

esphome:
  name: ${name}
  friendly_name: ${friendly_name}

esp8266:
  board: d1_mini

uart:
  id: HP_UART
  baud_rate: 2400
  tx_pin: 1
  rx_pin: 3
  
# External component reference
external_components:
  - source: github://echavet/MitsubishiCN105ESPHome

# Default logging level
logger:
  hardware_uart: UART1 # Uncomment on ESP8266 devices
  level: INFO

# Enable Home Assistant API
api:
  encryption:
    key: "{scrt}"
  services:
    - service: set_remote_temperature
      variables:
        temperature: float
      then:
        # Select between the C version and the F version
        # Uncomment just ONE of the below lines. The top receives the temperature value in C,
        # the bottom receives the value in F, converting to C here.
        - lambda: "id(mhp).set_remote_temperature(temperature);"
    #        - lambda: 'id(hp).set_remote_temperature((temperature - 32.0) * (5.0 / 9.0));'
    - service: use_internal_temperature
      then:
        - lambda: "id(mhp).set_remote_temperature(0);"
  # services:
  #   - service: set_remote_temperature
  #     variables:
  #       temperature: float
  #     then:
  #       - lambda: 'id(mhp).set_remote_temperature(temperature);'

  #   - service: use_internal_temperature
  #     then:
  #       - lambda: 'id(mhp).set_remote_temperature(0);'
sensor:
  - platform: uptime
    name: Uptime
  - platform: template
    id: hp_current_temperature
[... 769 lines cut ...]

    fan_mode_low_action:
      then:
        - climate.control:
            id: mhp
            fan_mode: "LOW"

    fan_mode_medium_action:
      then:
        - climate.control:
            id: mhp
            fan_mode: "MEDIUM"

    fan_mode_high_action:
      then:
        - climate.control:
            id: mhp
            fan_mode: "HIGH"
    fan_mode_quiet_action:
      then:
        - climate.control:
            id: mhp
            fan_mode: "QUIET"

```

BenHolmes, 2025-10-03:
I think this would be excellent to PR to the README, or at least a reference to a pinned discussion or something in an effort to provide clarity on dual set point capabilities that as a search around with the right term I see pop's up often.